weimarnetz/weimarnetz #59: /etc/config/network enthaelt fehler nach apply_profil.code

L337hium, 2015-02-17:
`apply_profil.code` hat brechnungsfehler fuer '/etc/config/network'
"""
config 'interface' 'wlan'    
#       option 'ifname' 'wlan0'  
        option 'proto'  'static'  
        option 'ipaddr'  '...1'
        option 'netmask' '255.255.255.255'
                                               

config 'interface' 'wlanadhoc'
#       option 'ifname' 'wlan0'
        option 'proto'  'static'
        option 'ipaddr'  '...0'
        option 'netmask' '255.255.255.255'
"""

bittorf, 2015-02-18:
das kommt von ipv4address_plus_x() wenn argument1 leer ist.
ich denke drueber nach.

glycoknob, 2015-02-18:
Müsste damit zusammenhängen: 

     _ipsystem ffweimar 199

läuft, aber 
      _ipsystem ffweimar 451  

bzw. alles > 256 geht nicht nicht: arithmetic error 

Ich hab gestern die exakte Stelle nicht gefunden, aber apply_profile nutzt einfach nur _ipsystem.

Vielleicht ist durch die v6 Änderungen was kaputt gegangen.

glycoknob, 2015-02-18:
```
root@hvf5-salon:~ set -x; _ipsystem ffweimar 431
+ _ipsystem ffweimar 431
+ _ipsystem_ffweimar 431             
+ local node_number=431
+ local city=63
+ local network=10
+ local ipv6_prefix=2001:bf7:193
+ local range_start=2
+ local range_end=1020
+ local range_first_random=970
+ local wifi_mask=8
+ local wifi_leasetime=30m
+ local wifi_leasenum=29
+ local lan_leasenum=13
+ local oct1 oct2 oct3 oct4 oldIFS s n random node_number_hex node_hex
+ [ -z 431 ]
+ [ 431 = random ]
+ _sanitizer do 431 ip4 check
+ _sanitizer_do 431 ip4 check           
+ local INPUT=431
+ local WORK=431
+ local CHEAT=check
+ local KEYWORD OPTION
+ [ -n ip4 ]
+ KEYWORD=ip4
+ OPTION=
+ [ ip4 = ip4 ]
+ _sanitizer_ip4 431 
+ local args=431 
+ local string=
+ local oldifs= 

+ IFS=.
+ set 431
+ IFS= 

+ continue
+ WORK=
+ shift
+ [ -n check ]
+ KEYWORD=check
+ OPTION=
+ [ check = check ]
+ [ dummy = 431 ]
+ return 1
+ [ 431 -ge 2 -a 431 -le 1020 ]
+ s=0
+ n=431
+ random=false
+ printf %X 431
+ node_number_hex=1AF
+ node_hex=0:1AF
+ [ 431 -ge 970 ]
+ [ 431 -gt 765 ]
+ [ 431 -gt 510 ]
+ [ 431 -gt 255 ]
-ash: arithmetic syntax error
```

glycoknob, 2015-02-18:
Fehler hab ich eingebaut.. ich teste nochmal und mach einen Pull-Request fertig

PR #61 (2015-02-19): fix my broken quoting in _ipsystem and cleanup _ipsystem
Bernd bekommt ein Bier :) 

fixes #59 

Problem war, dass arithmetische Ausdrücke `$(( a + b))` anders funktionieren und weder `$` noch Quotes brauchen... Quotes machen es bei ash kaputt...
closed 2015-02-19 by bittorf