weimarnetz/weimarnetz #60: Probleme mit serieller Konsole / apply_profile.code

glycoknob, 2015-02-18:
Beim Treffen gestern beobachtet - vielleicht kann @L337hium  noch mehr Details dazu schreiben... 

Auf einem 1043ND läuft `apply_profile.code` durch wenn man über SSH eingeloggt ist... ist man über die serielle Konsole verbunden, dann erscheinen Log-Messages vielfach (mindestens 3x) und es gibt eine Schleife bzw. bleibt der Router!? oder die serielle Konsole hängen... @L337hium meinte auch, dass es eine Kernel-Panic gab.. 

Problem scheint gibt es mit oder ohne den Redirect auf  `/dev/console` in `apply_profile.code` zu geben... 

So richtig klar ist nicht, woher das kommt und warum das passiert..

`procd` kümmert sich um das logging und müsste  auch per default auf  `/dev/console` schreiben... ist nur unklar woher da die schleife kommt... im Prinzip sollte es auch egal sein wie viele Tools auf /dev/console schreiben.. eine Schleife sollte unmöglich passieren... 

zum Reproduzieren https://github.com/weimarnetz/weimarnetz/commit/3fc6aeea0d46886238a26c0d310c904840526e37  nutzen und factory image flashen wenn man über die serielle Konsole verbunden ist... 

Können wir vielleicht bei einem der nächsten Treffen nochmal in Ruhe debuggen.

bittorf, 2015-02-19:
beim naechsten treffen bin ich wieder dabei...

glycoknob, 2015-02-19:
Ich vermute da stimmt beim Router was nicht oder es ist ein Bug außerhalb von kalua... hab es jetzt über die Pseudo-Serieller Konsole und Virtualbox getestet und da läuft auch alles sauber durch, auch mit Logging nach /dev/console.. ich pack das Logging mal wieder rein... ist jetzt x86 und Virtualbox und muss nichts bedeuten, aber am Script liegt es vermutlich nicht..

bittorf, 2015-03-03:
ok, bernd ist kaputt

closed 2015-03-03 by bittorf