weimarnetz/weimarnetz #65: make CONFIG_DOWNLOAD_FOLDER of use in build script

L337hium, 2015-02-22:
Do you see a way to implement the OpenWRT config variable `CONFIG_DOWNLOAD_FOLDER` in the build scripts? It could be set like `REPOURL` or `REPONAME` with `bash`s function `export`.

CONFIG_DOWNLOAD_FOLDER helps to prevent download errors while building with `make`.

glycoknob, 2015-02-22:
In mybuild.sh gibt es https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/mybuild.sh#L814 `set_openwrt_build.config` .. da könnte man das einbauen 

Das hier nimmt die Umgebungsvariable `$CONFIG_DOWNLOAD_FOLDER` und packt sie in `.config`: 

Wenn man das am ende von `set_openwrt_build.config` einbaut, dann kannst du mit der Umgebungsvariable den Download-Folder angeben.

    sed -i 's|^CONFIG_DOWNLOAD_FOLDER.*|CONFIG_DOWNLOAD_FOLDER='"$CONFIG_DOWNLOAD_FOLDER"'|g' .config

die | sind notwendig, weil sed sonst die / in der Variablen verwenden will.... 

Hab es getestet, verändert den Download-Folder bei mir, wenn das ich das vor `make` ausführe.

L337hium, 2015-02-22:
Is their a way to pass a global shell variable to `openwrt-config/config_standard.txt`?
Like be adding a new line
```
CONFIG_DOWNLOAD_FOLDER="$DOWNLOAD_FOLDER"
```
to that template.

It would be good to test for existince while executing `set_build` in https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/mybuild.sh#L153 or to be more specific in https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/mybuild.sh#L153, otherwise `CONFIG_DOWNLOAD_FOLDER=""` will be set as it would be by default.

L337hium, 2015-02-22:
https://github.com/weimarnetz/weimarnetz/issues/65#issuecomment-75441228
If have not had considert the excaping slash desaster from the beginning. But what if
```
export DOWNLOAD_FOLDER="\/home\/${USER}\/src\/openwrt_download_folder\/"
```

glycoknob, 2015-02-22:
@L337hium: Könnte auch gehen, will man aber nicht, oder? 

Ansonsten könnte würde es wohl auch klappen, dass an die config_standard.txt anzuhängen, es gibt eine Funktion, die die Optionen mit der .config merged... hab es nicht getestest.. 

`echo "CONFIG_DOWNLOAD_FOLDER="/home/foo/dl" >> openwrt-config/config_standard.txt` 

könnte auch klappen...

bittorf, 2015-02-22:
es ist alles viel einfacher:
wenn ihr openwrt ge-clone-t habt, macht ihr:

{{{
mkdir ../openwrt_download
ln -s ../openwrt_download 'openwrt/dl'
}}}

alles was openwrt also runterlaedt (naemlich nach 'dl'), landet in eurem directory. das koennt ihr auch schoen aufheben und immer wieder verwenden...

L337hium, 2015-02-22:
Ja ist ein workaround. Dachte nur weils menuconfig selber unterstuetzt kann mans da mit reinnehmen.

bittorf, 2015-02-23:
meiner meinung nach quark, weil das eine sehr individuelle entscheidung ist. ein symbolischer link der ausserdem des openwrt-directorys liegt wird dem wohl am meisten gerecht...ich schliesse mal.

L337hium, 2015-02-25:
```
git clone $REPOURL
mkdir myrelease; cd !$
mkdir -p release/openwrt
ln -s -T /an/obsolute/path/to/an/other/openwrt/dl /path/to/myrelease/release/openwrt/dl
DO="../$REPONAME/openwrt-build/build_release.sh"
$DO "HARDWARE.$HARDWARE" ffweimar_standard patch:901-minstrel-try-all-rates.patch patch:luci-remove-freifunk-firewall.patch patch:openwrt-remove-ppp-firewall-from-dep    s.patch patch:openwrt-remove-wpad-mini-from-deps.patch ffweimar_luci_standard hostapd vtunnoZlibnoSSL i18n_german https owm shrink tc use_bb1407
```

This fails shortly after executing `$DO`. 
`$REPOURL` will be cloned, and `make defconfig` just started. Not sure if `make defconfig` fails, or the taks next after that. It looks like have something to do with path handling. I saw paths like `..//bin/` which looks not as intendet... Have makefiles some special symlink handling?

bittorf, 2015-02-25:
das stimmt doch vorn und hinten nicht. schau dir nochmal die befehle an - bitte copy & paste und nicht einfach irgendwas reinschreiben, sonst kann man das nicht nachvollziehen. cd !$ sag ich nur...

L337hium, 2015-02-25:
ich hab nich irgendwas kopierpastert -.- egal hier ist das ende vom log. ich hab auch nen `script`-mitschnitt...

```
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: [START] invoking: 'tc' from 'HARDWARE.ar71xx
logger: set_build() using 'weimarnetz/openwrt-config/.config'
logger: apply symbol: CONFIG_PACKAGE_kmod-sched-core=y
logger: apply symbol: CONFIG_PACKAGE_tc=y
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: [READY] invoking: 'tc' from 'HARDWARE.ar71xx
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: [START] invoking: 'use_bb1407' from 'HARDWARE.ar71xx
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: [START] executing 'make defconfig  '
make: *** No rule to make target `defconfig'.  Stop.
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: [READY] executing 'make defconfig  '
sed: can't read package/base-files/files/etc/init.d/apply_profile.code: No such file or directory
sed: can't read package/base-files/files/etc/init.d/apply_profile.code: No such file or directory
sed: can't read package/base-files/files/etc/init.d/apply_profile.code: No such file or directory
logger: unknown filenames, search in 'bin//...' and read 'openwrt-config/HowTo_add_new_HARDWARE.md'
executing: 'make -j1 V=s'
make: *** No targets specified and no makefile found.  Stop.
../weimarnetz/openwrt-build/build_release.sh: 101: ../weimarnetz/openwrt-build/build_release.sh: ./scripts/getver.sh: not found
cat: package/base-files/files/etc/variables_fff+: No such file or directory
../weimarnetz/openwrt-build/build_release.sh: 103: ../weimarnetz/openwrt-build/build_release.sh: cannot create bin/revisions.json: Directory nonexistent
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: please removing everything via 'rm -fR release' if you are ready
logger: Wed Feb 25 21:36:23 CET 2015: [/home/bastian/bernd/WEIMARNETZ/test/myrelease/release/openwrt]: ../weimarnetz/openwrt-build/build_release.sh: # buildstring:  HARDWARE.ar71xx ffweimar_standard patch:901-minstrel-try-all-rates.patch patch:luci-remove-freifunk-firewall.patch patch:openwrt-remove-ppp-firewall-from-deps.patch patch:openwrt-remove-wpad-mini-from-deps.patch ffweimar_luci_standard hostapd vtunnoZlibnoSSL i18n_german https owm shrink tc use_bb1407
```

andibraeu, 2015-02-26:
Am Mittwoch, den 25.02.2015, 12:44 -0800 schrieb L337hium:
> ich hab nich irgendwas kopierpastert -.- egal hier ist das ende vom
> log. ich hab auch nen script-mitschnitt...

liegt das alles auf einer Partition? oder ist dein dl-Ordner woanders?
Damit hatte ich mal Probleme...

L337hium, 2015-02-26:
ja, auch releativ nah beieinander. hab die sym links auch absolute angegeben. ich kann das auch nochmal mit nem hardlink testen. die frage bleibt warum das bricht, und ob man die config option nicht doch setzen und verfuegbar machen kann.

glycoknob, 2015-02-26:
Das hier https://github.com/weimarnetz/weimarnetz/issues/65#issuecomment-76024461  kann nicht funktionieren... 

Und zwar: 

    ln -s -T /an/obsolute/path/to/an/other/openwrt/dl /path/to/myrelease/release/openwrt/dl
    DO="../$REPONAME/openwrt-build/build_release.sh"

Du setzt den Link schon vor dem Clonen des OpenWRT-Repositories... d.h. entweder geht das clonen schief oder der Link ist kaputt... funktionieren dürfte das jedenfalls nicht. 

Der Link muss innerhalb des geclonten Repository sein...

L337hium, 2015-02-26:
deswegen hab ich ja vorher alle ordner mit `mkdr -o release/openwrt` angelegt und darein den link erstellt. das ist also ein henne-ei-problem. wenn ich das script erst einmal durchlaufen lassen, dann hab ich ja schon wieder alles (hoffentlich) runtergeladen.

bei einem normalen openwrt hab ich die config option mal getestet und das ging ohne probleme.

glycoknob, 2015-02-26:
Das kann nicht sein... 

    [~]$ mkdir foo                                                                                       
    [~]$ touch foo/test                                                                                  
    [~]$ mkdir foo/openwrt                                                                               
    [~]$ git cloen git://git.openwrt.org/openwrt.git foo                                                 
    [~]$ git clone git://git.openwrt.org/openwrt.git foo                                                 
    fatal: Zielpfad 'foo' existiert bereits und ist kein leeres Verzeichnis.

Bist du sicher das das clone klappt? In dem log scheitert auch `make defconfig`.. das deutet darauf hin, dass OpenWRT gar nicht da ist... 

zudem:

    [~]$ mkdir -o foobar                                                                                 
   mkdir: Ungültige Option -- o

L337hium, 2015-02-26:
-o hatte ich schon korrigiert, dein cloen ist auch falsch geschrieben... alles pillepalle.

das das clone scheitert hatte ich auch schon vermutet aber nicht getestet. das clone/update klappt nur wenns sich bei dem ordner bereits um das git-repo handelt?

ja, also gibts vorschlaege wie man zum ziel kommt?

glycoknob, 2015-02-26:
Ich verstehe nicht so richtig wo das Problem ist.. gibt doch mehr als genug Möglichkeiten: 

- die Buildscripte patchen und die sed-Zeile einbauen... 
- nach dem git-clone im Buildscript den link setzen... 
- Das einbauen der Option in der `config_standard.txt` Datei sollte auch laufen... 

!?!?

glycoknob, 2015-02-26:
Wegem https://github.com/weimarnetz/weimarnetz/issues/65#issuecomment-76222280 

hab einen kleinen Patch gebaut, der in dem Fall abbricht: 
https://github.com/weimarnetz/weimarnetz/issues/65#issuecomment-76222280

Muss das noch testen, wenn das so klappt würd ich einen Pull-Request basteln... ich hatte auch schon Probleme weil etwas schief gegangen ist und ich dachte alles läuft...

closed 2015-06-06 by bittorf