weimarnetz/weimarnetz #66: webcaptive portal error

L337hium, 2015-02-24:
```
/usr/lib/lua/luci/dispatcher.lua:328: Access Violation
The page at 'freifunk/captive/' has no parent node so the access to this location has been denied.
This is a software bug, please report this message at http://luci.subsignal.org/trac/newticket
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:328: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>
```

andibraeu, 2015-02-24:
Am Dienstag, den 24.02.2015, 13:12 -0800 schrieb L337hium:
> /usr/lib/lua/luci/dispatcher.lua:328: Access Violation
> The page at 'freifunk/captive/' has no parent node so the access to this location has been denied.
> This is a software bug, please report this [...]

auf welcher Umgebung?

L337hium, 2015-02-24:
tl-wr841v8.2 mit 'akutellem' kalua

glycoknob, 2015-02-25:
@L337hium kannst du noch die genaue Build-Zeile angeben? Oder ist es die aus dem Wiki? 

    $DO "HARDWARE.ar71xx" ffweimar_standard patch:901-minstrel-try-all-rates.patch patch:luci-remove-freifunk-firewall.patch patch:openwrt-remove-ppp-firewall-from-deps.patch patch:openwrt-remove-wpad-mini-from-deps.patch ffweimar_luci_standard hostapd vtunnoZlibnoSSL i18n_german https owm shrink tc use_bb1407

!? ...wenn ja, dann versuch ich es mal die Tage zu reproduzieren...

glycoknob, 2015-02-25:
...hängt auch mit #54 zusammen

bittorf, 2015-03-03:
auf welchem router kann man das live sehen? es schein ein theme-loading-fehler zu sein...

L337hium, 2015-03-06:
hab grad nochmal den aktuellen master gebaut und bekam via seriell folgendes zu lesen

```
daemon.err uhttpd[2268]: /usr/lib/lua/luci/dispatcher.lua:328: Access Violation                                                                                           
daemon.err uhttpd[2268]: The page at 'freifunk/status/' has no parent node so the access to this location has been denied.                                                
daemon.err uhttpd[2268]: This is a software bug, please report this message at http://luci.subsignal.org/trac/newticket                                                   
daemon.err uhttpd[2268]: stack traceback:                                                                                                                                 
daemon.err uhttpd[2268]:    [C]: in function 'assert'                                                                                                                     
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:328: in function 'dispatch'                                                                                  
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>                                                      
daemon.err uhttpd[2268]: /usr/lib/lua/luci/dispatcher.lua:328: Access Violation                                                                                           
daemon.err uhttpd[2268]: The page at 'freifunk/status/' has no parent node so the access to this location has been denied.                                                
daemon.err uhttpd[2268]: This is a software bug, please report this message at http://luci.subsignal.org/trac/newticket                                                   
daemon.err uhttpd[2268]: stack traceback:                                                                                                                                 
daemon.err uhttpd[2268]:    [C]: in function 'assert'                                                                                                                     
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:328: in function 'dispatch'                                                                                  
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>                                                      
daemon.debug fff+ /usr/bin/scheduler wget_do(): max 15s, 25 bytes, wget  -qO -  'http://127.0.0.1:2006/all'                                                               
daemon.debug fff+ /usr/bin/scheduler wget_do(): [OK] no error message                                                                                                     
daemon.debug fff+ /usr/bin/scheduler olsr_restarts_are_suspicious_often(): percent from border: 0% restarts: 0 border: 30                                                 
daemon.info fff+ /usr/bin/scheduler incoming_wifi_activity(): [ERR] timediff > 60 sec = 69                                                                                
daemon.info fff+ /usr/bin/scheduler incoming_wifi_activity(): [OK] first activity seen on dev wlan0 (413 packets) - marking                                               
daemon.debug fff+ /usr/bin/scheduler incoming_wifi_activity(): [OK] framecounter for dev wlan0: old + diff = new : 413 + 455 = 868 (during 69 sec)                        
daemon.debug fff+ /bin/sh watch_switch_linkstatus(): [OK] nothing changed                                                                                                 
daemon.err uhttpd[2268]: /usr/lib/lua/luci/dispatcher.lua:328: Access Violation                                                                                           
daemon.err uhttpd[2268]: The page at 'freifunk/status/' has no parent node so the access to this location has been denied.                                                
daemon.err uhttpd[2268]: This is a software bug, please report this message at http://luci.subsignal.org/trac/newticket                                                   
daemon.err uhttpd[2268]: stack traceback:                                                                                                                                 
daemon.err uhttpd[2268]:    [C]: in function 'assert'                                                                                                                     
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:328: in function 'dispatch'                                                                                  
daemon.err uhttpd[2268]:    /usr/lib/lua/luci/dispatcher.lua:195: in function </usr/lib/lua/luci/dispatcher.lua:194>
```

Darueber hinaus (k.A. ob zusammenhaengend) werd ich als LAN-Client gesplasht.

L337hium, 2015-03-06:
Mhm das mit dem gesplashten LAN liesz sich nicht reproduzieren...

glycoknob, 2015-03-06:
Hi, könntest du die Paketliste vom Router posten? 

Könnte sein, dass Abhängigkeiten für LuCI nicht installiert sind - die Splash-Seite registriert sich so: 

https://github.com/weimarnetz/openwrt-packages/blob/master/utils/luci-weimar-splash/luasrc/controller/weimarnetz/splash.lua

zum Vergleich luci-mod-freifunk: https://github.com/openwrt/luci/blob/master/modules/luci-mod-freifunk/luasrc/controller/freifunk/freifunk.lua

andibraeu, 2015-03-08:
```freifunk/status/``` hat nichts mit dem weimar-splash zu tun, glaub ich. Geht das restliche Webinterface?

L337hium, 2015-03-08:
nein, das webinterface ist nicht erreichbar.

andibraeu, 2015-03-08:
gut, dann ist das kein luci-weimar-splash-problem... zunächst mal muss LuCI itself laufen.

was da auf master grad passiert weiß ich nicht, vielleicht haste grad ne Kombination erwischt, die nicht geht

bittorf, 2015-03-11:
soll ich das zumachen?

glycoknob, 2015-03-11:
Ich glaub der Bug tritt noch auf - bin noch nicht zum testen gekommen, aber Bernd hat einfach nur die normale Firmware verwended. Ich hatte das Problem teilweise auch... ich denke sollte noch offen bleiben.

bittorf, 2015-03-11:
bernd: scripte doch mal ein ding, was bei alles routern probiert die seite aufzurufen und wenn die fehlermeldung drinne ist, knotennummer ausspucken

glycoknob, 2015-03-12:
Grad mal getestet... Knoten 195 -> geht nicht (von mir verbastelt) folgende luci-Pakete sind installiert: 

```
opkg list  | grep ^luci 
luci-base - 0.12+git-15.006.29251-d876593-1
luci-lib-httpclient - 0.12+git-15.006.29251-d876593-1
luci-lib-json - 0.12+git-15.006.29251-d876593-1
luci-lib-luaneightbl - 0.12+git-15.006.29251-d876593-1
luci-lib-nixio - 0.12+git-15.006.29251-d876593-1
luci-theme-bootstrap - 0.12+git-15.006.29251-d876593-1
luci-weimar-splash - 0.2.5
```

Läuft (soweit) - Knoten 197 - Frish geflashtes aktuelles git von kalua... 1:1 nach Anleitung gebaut.. folgende Paketliste: 

```
~ opkg list | grep ^luci
luci-app-olsr - 0.12+git-15.037.36195-f1e2a26-1
luci-app-olsr-services - 0.12+git-15.037.36195-f1e2a26-1
luci-app-owm - 0.5.3
luci-app-owm-cmd - 0.5.3
luci-base - 0.12+git-15.037.36195-f1e2a26-1
luci-i18n-english - 0.12+git-15.037.36195-f1e2a26-1
luci-i18n-german - 0.12+git-15.037.36195-f1e2a26-1
luci-lib-httpclient - 0.12+git-15.037.36195-f1e2a26-1
luci-lib-json - 0.12+git-15.037.36195-f1e2a26-1
luci-lib-luaneightbl - 0.12+git-15.037.36195-f1e2a26-1
luci-lib-nixio - 0.12+git-15.037.36195-f1e2a26-1
luci-mod-admin-full - 0.12+git-15.037.36195-f1e2a26-1
luci-mod-freifunk - 0.12+git-15.037.36195-f1e2a26-1
luci-theme-bootstrap - 0.12+git-15.037.36195-f1e2a26-1
luci-weimar-splash - 0.2.5
```

Also fehlt luci-mod-freifunk... 

Hab mal versucht ein Script zu schrieben... wir brauchen noch ein Feld für die IP in der API: 

benötigt bash/zsh und [jq](http://stedolan.github.io/jq/)

```
cat > /tmp/node2ip <<EOF
node2ip()
{
n=$1
s=0
if [ "$n" -gt 765 ]; then 
 n=$(( n - 765 ))
 s=192
elif [ "$n" -gt 510 ]; then
 n=$(( n - 510 ))
 s=128
elif [ "$n" -gt 255 ]; then
 n=$(( n - 255 ))
 s=64
fi

echo 10.63.$n.$(( s + 33 ))
}
EOF
```

Ist fast ein Einzeiler ;) 
```

. /tmp/node2ip; while read i; do ip=$( node2ip $i ); echo "$i\t$ip\t$(curl -m 5 -s -I http://$ip/cgi-bin/luci/freifunk/captive | head -n1)"; done <<< $(curl -s http://reg.weimarnetz.de/GET/ffweimar/knoten | jq '.result.knoten[].number' | sort -n )

```
Ergebnis hier: https://gist.github.com/glycoknob/6d7e8ac19516db1e9927

Übrigens alle die mit 307 Antworten laufen nicht mit Windows Phone bzw. IE inklusive 11... siehe: https://github.com/weimarnetz/openwrt-packages/issues/7

Also sieht gut aus, scheint kein größeres Problem zu sein

glycoknob, 2015-03-12:
Ach und LAN wird gesplashed bei mir in der aktuellen Firmware

bittorf, 2015-03-12:
ok, das problem das LAN gesplashed ist, loese ich gerade...

bittorf, 2015-03-12:
ok, LAN ist nun wieder ungesplased: a270342c698f840dc45f5da1180887c6965b9e31

bittorf, 2015-03-12:
ich denke schon, das dies dringend gefixt gehoert:

URL="https://gist.githubusercontent.com/glycoknob/6d7e8ac19516db1e9927/raw/d9930756d4e9e1a02d42db0a76096a82a1dd7b9c/captive-test"

bastian@X301:~/software/weimarnetz$ wget -qO - "$URL" | fgrep "HTTP/" | fgrep -v "200 OK"
7       10.63.7.33      HTTP/1.1 307 Weimarnetz Captive Portal
18      10.63.18.33     HTTP/1.1 307 Weimarnetz Captive Portal
20      10.63.20.33     HTTP/1.1 307 Weimarnetz Captive Portal
34      10.63.34.33     HTTP/1.1 404 Not Found
43      10.63.43.33     HTTP/1.1 307 Weimarnetz Captive Portal
44      10.63.44.33     HTTP/1.1 307 Weimarnetz Captive Portal
51      10.63.51.33     HTTP/1.1 404 Not Found
52      10.63.52.33     HTTP/1.1 404 Not Found
53      10.63.53.33     HTTP/1.1 404 Not Found
75      10.63.75.33     HTTP/1.1 307 Weimarnetz Captive Portal
117     10.63.117.33    HTTP/1.1 307 Weimarnetz Captive Portal
119     10.63.119.33    HTTP/1.1 307 Weimarnetz Captive Portal
139     10.63.139.33    HTTP/1.1 307 Weimarnetz Captive Portal
141     10.63.141.33    HTTP/1.1 307 Weimarnetz Captive Portal
143     10.63.143.33    HTTP/1.1 404 Not Found
148     10.63.148.33    HTTP/1.1 307 Weimarnetz Captive Portal
149     10.63.149.33    HTTP/1.1 307 Weimarnetz Captive Portal
150     10.63.150.33    HTTP/1.1 307 Weimarnetz Captive Portal
151     10.63.151.33    HTTP/1.1 307 Weimarnetz Captive Portal
152     10.63.152.33    HTTP/1.1 307 Weimarnetz Captive Portal
153     10.63.153.33    HTTP/1.1 307 Weimarnetz Captive Portal
170     10.63.170.33    HTTP/1.1 307 Weimarnetz Captive Portal
173     10.63.173.33    HTTP/1.1 307 Weimarnetz Captive Portal
175     10.63.175.33    HTTP/1.1 307 Weimarnetz Captive Portal
182     10.63.182.33    HTTP/1.1 307 Weimarnetz Captive Portal
183     10.63.183.33    HTTP/1.1 307 Weimarnetz Captive Portal
184     10.63.184.33    HTTP/1.1 307 Weimarnetz Captive Portal
185     10.63.185.33    HTTP/1.1 307 Weimarnetz Captive Portal
186     10.63.186.33    HTTP/1.1 307 Weimarnetz Captive Portal
187     10.63.187.33    HTTP/1.1 307 Weimarnetz Captive Portal
190     10.63.190.33    HTTP/1.1 307 Weimarnetz Captive Portal
194     10.63.194.33    HTTP/1.0 501 Not Implemented
195     10.63.195.33    HTTP/1.1 500 Internal Server Error
201     10.63.201.33    HTTP/1.1 307 Weimarnetz Captive Portal
203     10.63.203.33    HTTP/1.1 307 Weimarnetz Captive Portal
256     10.63.1.97      HTTP/1.1 307 Weimarnetz Captive Portal
270     10.63.15.97     HTTP/1.1 307 Weimarnetz Captive Portal
272     10.63.17.97     HTTP/1.1 307 Weimarnetz Captive Portal
276     10.63.21.97     HTTP/1.1 401 Authorization Required
277     10.63.22.97     HTTP/1.1 401 Authorization Required
287     10.63.32.97     HTTP/1.0 501 Not Implemented
547     10.63.37.161    HTTP/1.0 501 Not Implemented
599     10.63.89.161    HTTP/1.0 501 Not Implemented
861     10.63.96.225    HTTP/1.1 401 Authorization Required
862     10.63.97.225    HTTP/1.1 401 Authorization Required
911     10.63.146.225   HTTP/1.1 307 Weimarnetz Captive Portal

bittorf, 2015-03-12:
wenn bei einem frisch installierten - 1:1 nach Anleitung gebautem router 'luci-mod-freifunk' drinne ist und der fehler also nicht da, koennen wir das ticket schliessen, oder? (das beheben des 307-fehlers ist auch schon gemacht, muss nur netzweit eingespielt werden...)

glycoknob, 2015-03-12:
ich denke schon... es gibt noch den Bug, dass luci-weimar-splash (und das kalua)-Paket keien Abhängigkeiten definiert hat, dann passiert das auch nicht beim testen, für die Firmware ist das aber estmal egal.

glycoknob, 2015-03-12:
Danke für den LAN-Fix.. das ich hab ich wohl übersehen

closed 2015-03-12 by bittorf

L337hium, 2015-03-12:
Ich hab die router grad leider nicht hier, aber ist das packet denn wirklich mit drin? Also mindestens die letzten 3 revisionen die ich gebaut hatte, waren in der hinsicht kaputt. Also keine splash und eben fehlerhaftes lan-splashed.

bittorf, 2015-03-12:
ed: schau doch mal noch. glycoknopf schreibt doch bei 1:1 anleitung geht...bitte mal machen

glycoknob, 2015-03-12:
@L337hium: Hab ich mich auch gefragt, aber das ist in [ffweimar_luci_standard](https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-config/config_ffweimar_luci_standard.txt) seit gut 2 Jahren drinnen. 

Achso ich hab es mit Barrier Breaker getestet.. kann gut sein, dass Trunk kaputt ist. Aber ich dachte das Problem tritt unter BB auf?

L337hium, 2015-03-12:
@glycoknob @bittorf das ist es ja. Habe immer mit der liste aus README.md gebaut. Und bis auf wenige ausnahmen immer bb_1407, das ich ein use_trunk gebaut habe, ist schon einige wochen her. Wie gesagt wenn ich heut abend im maschinenraum bin, guck ich mir die packetlisten nochmal an, bau den aktuellen master und teste nochmal.