weimarnetz/weimarnetz #68: /etc/init.d aufräumen/anpassen für Paket-Build

glycoknob, 2015-03-02:
Sorry, grad nur Tickets, keine Zeit für Patches... 

Wir haben ein Problem: 

http://wiki.openwrt.org/doc/devel/packages#packaging_a_service

> Packaging a service

> If you want to install a service, (something that should start/stop at boot time, that has a /etc/init.d/blah script), you should make sure that the init.d script can be run on the host. At image build time, all init.d scripts found are run on the host, looking for the START=20/STOP=99 lines.

> This is what installs the symlinks in /etc/rc.d, so they are only created when you rebuild the entire image. If you want the symlinks to be created when a package is installed, such as via opkg, you should add a postinstall script which runs...

Wir müssten mal die Scripte aufdröseln und verschieben oder alles mit dem init.d System von OpenWRT kompatibel machen. Momentan rufen Scripte in /etc/init.d andere Scripte in /etc/init.d auf... 

Das sieht dann so aus: 

zum Glück ist kein `rm ~` in den init-Scripten... 

```
make[3]: Leaving directory `/home/mt/openwrt'
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S02check_interface_correctness_fff+: line 2: /tmp/loader: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S02check_interface_correctness_fff+: line 4: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S02check_interface_correctness_fff+: line 5: _ssh: command not found
/etc/init.d/S02check_interface_correctness_fff+ does not have a START or STOP value
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S51crond_fff+: line 2: /tmp/loader: No such file or directory
Usage: /home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/rc.common (stop | start [nano|minimal|standard] | restart [...] | status)
/etc/init.d/S51crond_fff+ does not have a START or STOP value
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 2: /tmp/loader: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 4: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 28: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 30: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 32: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 33: _watch: command not found                                                                         
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 35: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 41: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 43: _watch: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 44: _net: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 53: cron.add_collected_userdata_into_db: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 55: _math: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 55: _log: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 56: _netfilter: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 60: /usr/sbin/cron.reverse_ssh_tunnel: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 62: _firmware: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 71: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 72: _net: command not found
iptables v1.4.21: can't initialize iptables table `mangle': Permission denied (you must be root)
Perhaps iptables or your kernel needs to be upgraded.
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 85: /etc/init.d/netfilter: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 86: /etc/init.d/netfilter: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 91: _system: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S69watch_olsrd_fff+: line 91: [: : integer expression expected
/etc/init.d/S69watch_olsrd_fff+ does not have a START or STOP value
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/rc.common: line 97: .: /home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/apply_profile.regulatory.bin: cannot execute binary file
/etc/init.d/apply_profile.regulatory.bin does not have a START or STOP value
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S72filter_fff+: line 2: /tmp/loader: No such file or directory
mt: /home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/rc.common: [ERR] user = mt: seems we are in cross compiling stage?
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 2: /tmp/loader: No such file or directory
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 140: _system: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 140: [: -lt: unary operator expected
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 147: _file: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 135: _log: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 26: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 51: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 52: uci: command not found
Device "nonexistent_loopback" does not exist.
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 55: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 76: _sanitizer: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 26: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 51: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 52: uci: command not found
[... 20 lines cut ...]
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 95: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 95: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 95: uci: command not found
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/etc/init.d/S41build_static_netparam: line 95: uci: command not found
[... cut ...]
/home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/usr/lib/opkg/info/luci2.postinst-pkg: line 5: uci: command not found
rm -f /home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/usr/lib/opkg/info/*.postinst*
rm -f /home/mt/openwrt/build_dir/target-x86_64_uClibc-0.9.33.2/root-x86_64/usr/lib/opkg/info/*.prerm*
make[2]: Leaving directory `/home/mt/openwrt'
make[2]: Entering directory `/home/mt/openwrt'
make[3]: Entering directory `/home/mt/openwrt/target/linux'
make[4]: Entering directory `/home/mt/openwrt/target/linux/x86_64'
```

bittorf, 2015-03-03:
ich kuemmere mich drum, also alles schon in boo() und start() umbauen

glycoknob, 2016-01-09:
Läuft jetzt bei mir, habe nochmal auf master gerebased und noch ein Problem gefixt. Von mir aus könnt ihr commiten.

closed 2016-01-09 by bittorf