weimarnetz/weimarnetz #71: Vorschlag: /etc/apply_profile.d

glycoknob, 2015-03-08:
Hallo, 

Ich habe damit schon angefangen: https://github.com/glycoknob/weimarnetz/tree/master/kalua/etc/apply_profile.d (läuft noch nicht)  

Die Dateien würden alle in einer Subshell gesourced werden, ganz ähnlich der `boot` file aus dem `init.d` Verzeichnis:

        cd /etc/apply_profile.code.d || return 0
        files="$(ls)"
        [ -z "$files" ] && return 0
        for file in $files; do
                ( . "./$(basename $file)" ) 
        done

Hätte den Vorteil die Konfiguration aufsplitten zu können und jede Datei wäre relativ unabhängig von den anderen. Variablen müssten vorher global gesetzt sein oder man zwingt sich direkt uci in der Konfiguration zu nutzen.. Würde Änderungen einfacher machen, da man relativ sicher sein kann, dass z.B. Änderungen beim VPN oder OLSR keine Auswirkungen auf anderes haben.. ist jetzt auch schon so, aber apply_profile.code ist doch recht groß mit Funktionen über mehrere Seiten hinweg sowie ähnlich riesigen switch-case Konstrukten... 

Weiterer Vorteil wäre, dass man z.B. die OLSR-Konfig direkt aus der fertigen Netzwerk und Wireless Konfiguration generieren könnte. 

Die Scripte hier würden sich dafür anbieten: http://wiki.openwrt.org/doc/devel/config-scripting - könnte man ggf. auch zur Vereinfachung nutzen. 

Wenn das so läuft, dann könnte man es sogar noch weiter aufsplitten z.B. wenn die Wireless-Schnittstellen konfiguriert sind, eine extra Datei für SSIDs vergabe erstellen etc.pp 

Weiterer Vorteil wäre, dass wenn wir dahin wollen möglichst modulare Pakete zu haben, diese einfach ihre Konfigurationsscripte nach `/etc/apply_profile.d` packen können.. z.B. luci-weimar-splash oder vielleicht mal ein paar Sachen aus dem FreifunkAdvanced-Feed und gleich mitkonfiguriert werden.

andibraeu, 2015-03-08:
ja, das klingt super!

L337hium, 2015-03-08:
Wenn etwas failed in der shell, gibt man einen exit code groeszer 0 zurueck. 0 wird nur geworfen, wenn keine fehler auftreten.

glycoknob, 2015-03-08:
@L337hium: Yo..hab das aus dem boot-script kopiert dort geht es um die `uci-defaults` wenn es den Ordner nicht gibt, dann ist das auch okay... für `apply_profile.code` wäre 1 besser. Danke für den Hinweis.

bittorf, 2015-03-12:
ich finde die idee erstmal gut. auf jeden fall aufraeumen. ein problem ist, das manche functionen globale variablen erzeugen, die von anderen spaeter gebraucht werden. ich denke ich werde das in dem rahmen: radio0(adhoc+ap+ap) + radio1(adhoc+ap+ap) + radioN(adhoc+ap+ap) und dhcp-geht machen...(dazu muss ich aber erstmal rausfinden wie man es konfiguriert. der usecase, das man 3 radio's hat ist nicht so abwegig: ein 3600er mit USB-WLAN-stick (bäääääm!)

bittorf, 2015-06-07:
ich schliesse das mal, weil es kein echtes issue ist, sondern eher mal was für ne lange winternacht...

closed 2015-06-07 by bittorf