weimarnetz/weimarnetz #73: rebuilding kalua does not always work...

glycoknob, 2015-03-08:
Entweder ich habe Einbildungen oder es gibt ein Problem... 

Schritte: 

per SSH eingeloggt.. 

ich editiere z.B. in `/etc/kalua/vpn` herum... speichere die Datei

dann `/etc/kalua_init` 
und `. /tmp/loader` 

dann z.b. `_vpn build_vars` 

Was soll passieren: Mein neuer Code wird ausgeführt 
Was passiert: Der alte Code wird ausgeführt 

Reboot hilft... 

Ich dachte das updaten klappt so?

glycoknob, 2015-03-10:
Mehr Details: 

`/etc/kalua_init` läuft durch, neue Files landen in `/tmp/kalua` und `CHECKSUMS` wird aktualisiert.. 

Das Problem ist hier: 

in `/tmp/loader` (wird von `kalua_init` da hingeschrieben):
    case "$LODEV" in lo) return 0;; esac

```
root@hvf5-salon:~ /etc/kalua_init
/etc/kalua_init: line 99: _file_discusage: not found
Mar 10 01:17:29 ' [up 34250s 6549504 bytes free] TERM:xterm-256color /etc/kalua_init [daemon.info] gen_loader() [OK] done '/tmp/loader' (in 1.18sec, using  bytes in /tmp/kalua) - you MUST reinclude the new loader with '. /tmp/loader' if this is an interactive shell
root@hvf5-salon:~ echo $LODEV
lo
```

Also werden die neuen Files nicht inkludiert.... das leeren von `$LODEV` scheint auch nicht zu klappen... 

Ansonsten hab ich die Variable noch hier gefunden: 

```
/tmp/loader:case "$LODEV" in lo) return 0;; esac
/tmp/loader:LODEV=lo;LOMTU=65536;LOADR=127.0.0.1;LOMSK=255.0.0.0;LOBRC=127.255.255.255;LONET=127.0.0.0;LOPRE=8;LOOLSR=;WIFIDEV=wlan0;WIFIMTU=1500;WIFIADR=10.63.195.1;WIFIMSK=255.0.0.0;WIFIBRC=10.255.255.255;WIFINET=10.0.0.0;WIFIPRE=8;WIFIOLSR=;LANDEV=eth1;LANMTU=1500;LANADR=10.63.195.33;LANMSK=255.255.255.240;LANBRC=10.63.195.47;LANNET=10.63.195.32;LANPRE=28;LANOLSR=;WANDEV=eth0;WANMTU=1500;WANADR=192.168.2.118;WANMSK=255.255.255.0;WANBRC=192.168.2.255;WANNET=192.168.2.0;WANPRE=24;WANOLSR=;WIFI_DEVS='wlan0 wlan0-1 ';
/tmp/NETPARAM:LODEV=lo
```

glycoknob, 2015-03-10:
Workaround: 

    /etc/kalua_init && LODEV= . /tmp/loader

glycoknob, 2015-03-11:
Feature, kein Bug :)

closed 2015-03-11 by glycoknob

bittorf, 2015-03-11:
ich habs mal in der logausgabe gefixt: dba0a0fa9c8cfd6bb1eda53da5e9c8d398529b82