weimarnetz/weimarnetz #75: _net ping_getlatency Änderungen/Verbesserung für VPN

glycoknob, 2015-03-09:
Hallo, 

das VPN-Script ist jetzt so gebaut, dass ein Aufruf von _net ping_getlatency größer als 0 sein muss. 0 Bedeutet der Server ist nicht erreichbar... 

Wenn es Okay ist, dann würd ich das hier auch noch als Pull-Request senden: 
https://github.com/glycoknob/weimarnetz/commit/2e7ad7f513f1a88e3c467321a45ec22f2de034f1 

- Es wird 3x gepingt.. ist vielleicht sicherer falls mal ein Ping nicht durchkommt. 
- Die letzte Zeile wird bis zum = gelöscht, dann das 2. Element (avg) genommen als Zeit.. ist unter Ubuntu und Busybox getestet...  das wird dann mit der Funktion int() auf den nächsten Integer abgerundet... int(0.999) = 0 - daher das +0.999 
- Wenn es eine Antwort von Ping gibt dann ist die Latenz immer größer als 0, also mindestens 1 
- Bei Fehlern dürfte nur 0 zurück gekommen, da undefined oder 0 +0.999 immer noch 0 ist 

Hintergrund ist, dass bei mir im LAN _net ping_getlatency auch 0 zurückgibt wenn die Verbindung gut aus.. also z.B. ein Ping von 0.3ms .. das könnte unter Umständen das VPN Script davon abhalten zu laufen z.B. auf Servern oder so...

bittorf, 2015-03-11:
hab es mal noch besser gefixt def229e005f36e82b8220429511a7159b9519921 - im LAN wird jetzt wirklich '0' zurueckgegeben, und <leer> wenn es fehlschlaegt. ausserdem ohne AWK 8-)

closed 2015-03-11 by bittorf

glycoknob, 2015-03-11:
Top!