weimarnetz/weimarnetz #76: Feature: Sicherstellen, dass Netz nur über VPN läuft oder keine Internetfreigabe...

glycoknob, 2015-03-20:
Bin mir grad nicht sicher, aber nach lesen vom VPN-Code sieht es so aus, dass _wenn es einen WAN-Gateway_ gibt Internet darüber läuft - völlig unabhängig vom VPN Status... 

Das Script setzt die Routen und ändern den Default-Gateway auf das VPN, wenn die Verbindung erfolgreich ist... https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/kalua/vpn#L172 

Daraus müsste folgen: 

- Bis das VPN startet geht Internet _nicht_ über das VPN raus.. wenn das init.d Script in der aktuellen Firmware nicht greift, dann kann das schon ne weile dauern bis cron das VPN erneut starten will... 

- Gibt es Probleme mit dem VPN z.b. alle Server offline, nötige Ports sind geblockt, etc.pp läuft Internet über den normalen Anschluss... 

- bricht vtun ab, dann wird die Default-Route auch abgrissen und bis zum nächsten cron Check läuft Internet über den normalen Anschluss... 

Angesichts der Versprechen und den Abmahmungen müssen wir das irgendwie sauber fixen... per Firewall Internet-Zugang blocken und im VPN-Script freischalten bzw. wieder abschalten? Irgendwie das Routing so laufen lassen, dass Pakete nur beim vtun/tap Interface ankommen? 

Internet sollte für DHCP-Clients nur über das VPN rausgehen und sonst immer ins leere gehen. Der Router selbst muss natürlich ins Netz...

L337hium, 2015-03-20:
Das VPN dient nur **INDIREKT** dem Stoererhaftungsfoo! **PRIMAER** ist das fuer die Vernetzung der Funk-Inseln da... 

Das hat schon seinen Grund warum das "Netz" gehen soll auch wenn da eine VPN-Stoerung vorliegt. Es kommt auch immer mal wieder vor das bei IN-Berlin ein Switch falsch konfigueriert ist oder sonst ein Problem auftritt und dann der primaere VPN-Server nicht erreichbar ist. Dat war nich so cool, wenn dann auf keinem Router (gerade die alten) das (Inter)net nicht ginge.

bittorf, 2015-03-21:
ich denke auch das es sehr wichtig ist, sich nicht vom VPN abhaengig zu machen. Evtl. koennten man einen (dauerhaften) logeintrag machen, wenn VPN nicht aktiv/wieder aktiv ist.

glycoknob, 2015-03-21:
Von der Homepage: 

> *Wie ist das rechtlich, wenn ich meinen Internetanschluss anderen zur Verfügung stelle?*
Um Internet-Einspeisende vor möglichen Abmahnungen zu schützen und von der sogenannten Störerhaftung zu entlasten betreibt der Weimarnetz e.V. einen VPN-Server in Berlin über den alle WLAN-Teilnehmerinnen getunnelt werden. Dadurch können Nutzerinnen des Weimarnetz nicht direkt einem Internetanschluss zugeordnet werden.

Nun... VPN und Störerhaftung sind die üblichen Fragen, wenn man jemand fragt ob er einen Router aufstellen will... soweit ich weiß haben wir stehts bejaht, dass da nicht passieren kann weil VPN,  es ist so auf der Homepage dokumentiert und _nirgends_ steht etwas davon, dass es halt manchmal nicht funktioniert und man doch ne Abmahnung oder schlimmeres bekommen kann..

L337hium, 2015-03-21:
On 03/21/2015 05:28 PM, glycoknob wrote:
> Von der Homepage:
>
>     /Wie ist das rechtlich, wenn ich meinen Internetanschluss anderen
>     zur Verfügung stelle?/
>     Um Internet-Einspeisende vor möglichen Abmahnungen zu schützen und
>     von der [...]

Dann mach nen vorschlag fuer ne config-option. Aber ich verspreche dir,
dass geheule ist viel groeszer wenn andauernd 'das internet nicht geht'.

Diese ganze abmahnparanoia nervt! Als wenn die leute den ganzen tags nix
besseres zu tun haetten als schlechte blockbaster zu saugen. Ausserdem
hat die industrie so viel aengste geschuert, dass die user eh nur noch
auf streaming- und one-click-hostern unterwegs sind.

Entschuldige meinen angepickten ton, aber ich heute vorhin exakt die
gleiche disskussion mit den erfurtern, wie es sein koenne, dass der
router dann trotzdem internet hat und ueberhaupt da keine krypto auf dem
tunnel liegt m(

Das argument mit 'na es soll eigentlich nur mesh-wolken vernetzen'
greift schon gar nicht mehr, da sich scheinbar alle funker nur noch als
/service-zur-verfuegung-steller/ betrachten.

Das ganze vpn-gedoens wo alles auf einem gateway rauskracht macht am
ende auch nur probleme. Die politik soll sich lieber drum kuemmern dass
die anwaelte sich neue jobs suchen und die richter sollen mal ne
weiterbildung besuchen, in wie fern ne ip adresse 'aussagekraeftig' ist.

Also back-to-topic: Lieber internet als gar kein internet. Wenn du ne
schmackige formulierung fuer die webseite findest die diesen komplexen
sachverhalt verstaendlich erklaert OHNE dass dir die unbedarften user
wegrennen, dann koennen wir den ja einbauen, aber ich mag nicht dran
glauben dass es eine solche formulierung gibt, dafuer haben die wie
gesagt alle viel zu viel angst.

glycoknob, 2015-03-21:
Das Problem ist die Störerhaftung... sonst wären VPN-Tunnel nicht notwendig und alles wäre schön... 

Ich finde es zumindest zweifelhaft zu kommunizieren, dass so ein Freifunk-Router die Störerhaftung explizit umgeht um dann, wenn die Leute mit den Schreiben der Kanzlei zum Treffen kommen zu sagen: Naja... äh manchmal auch nicht... 

Krypto auf dem Tunnel halte ich auch für sinnvoll, sichert dir halt deine Datenintegrität auf dem Transportweg ab... das das wegen der Hardware nicht klappt ist auch okay, aber man will das schon haben...

Was machen wir denn außer einen Service zur Verfügung zu stellen? Im Unterschied zur Telekom ist es halt ein unkommerzieller, gemeinschaftlich organisierter Service... 

Die Politik wird das Ticket hier nicht lesen und sich denken: Yoah, der Bernd hat ja recht, machen wir das mal so... 

Ich persönlich denke halt, dass immer und nur VPN ein guter Kompromiss ist um Nutzer vor der Störerhaftung zu schützen. Ich würde das persönlich gerne so haben, weil ich keinen Bock habe auf Abmahmungen und Hausdurchschungen im Worst-Case weil der Staat meint eine IP-Adresse ist mit einer Person verknüpft... 

Klar kannst du das als Angst auslegen, was du ja schon getan hast, aber man kann es auch als Naivität ggü den Repressionsorganen bezeichnen, der man sich nicht umbedingt wissensichtlich hingeben muss... User anzulügen geht gar nicht... ein paar redudante VPN-Server aufzustellen ist für mich das kleinere Problem. 

/bin heute auch angepickt... wie auch immer, können ja nochmal beim Treffen drüber sprechen und uns auf irgendwas einigen...

andibraeu, 2015-03-21:
Mein Senf dazu:

Solange wir aus den Abmahnungen noch mit der Argumentation "Wir machen Freifunk, also geht wieder weg" rauskommen brauchen wir auch an der jetzigen Standardeinstellung nichts ändern. Hat ja bei 2 Participants hier auch schon geklappt, und da war VPN auch "aus". Ich halte es auch eher für kontraproduktiv, wenn wir bei kaputten VPN-Server kein Internet mehr einspeisen können.

Zum Thema Krypto auf dem Tunnel: Das verstehe ich immer noch nicht, warum wir den kürzesten Teil auf der Strecke zum Ziel verschlüsseln sollten, wo es spätestens ab dem VPN-Server unverschlüsselt weitergeht. Das verbrennt doch nur unnötig Resourcen und bringt keinen Schutz. Im Meshnetzwerk stellste dann im Zweifel einen eigenen Knoten auf und lauschst mit. Ist also nur Schlangenöl

L337hium, 2015-03-22:
auch wenn ich gern mit andi streite ;)

in beiden dingen: full ack!

aber die erfurter halten uns auch fuer wahnsinnig, wie sich vorhin bei
der diskussionsrunde #2 herausstellt. ich fasse sowas aber mal als
kompliment auf, dass wir da pragmatischer an die sache ran gehen.

zum punkt: vpn kann auch mal ausfallen. wenn ich leuten beim treffen nen
router geflasht habe, habe ich auch immer darauf hingewiesen, dass es
einfach immer mal zu technischen stoerungen kommen kann. seis nun wegen
nem software bug, oder weil das netz beim housing-provider kaputt st,
oder schlicht die kiste down ist. also ja im default soll der traffic
wegen der missaere ueber den vpn gehen, aber das ist kein garant! und ja
harte krypte ist sinnfrei und keys erst an admins schicken ist noch sinn
freier, aber da sind 'wir' scheinbar alleine mit der auffassung...

ja wir sollten beim treffen vielleicht nochmal ein stuendchen
diskussionszeit einraeumen. viellecht aber erst nachdem wir weiter an
der firmware rumgemacht haben ;)

bittorf, 2015-03-22:
ich denke eine config-option könnte trotzdem nicht schaden. quasi: VPN kaputt = internet-ankündigung rausnehmen. das VPN ist ja quasi so selten kaputt und wir haben ja ausfallsicherheit, das dies wohl nicht so schlimm ist und es nimmt den leuten den wind aus dem segel, die sagen: da gib es ja eine lücke...

egon0, 2015-03-26:
@L337hium: ihr seid nicht allein und nicht irre. Und ich bin immer wieder froh wenn ich von einzelnen Freifunkern lese die die Sache ähnlich sehen wie ich. So ein Killswitch ist aber an sich OK, mit entsprechender Kommunikation. Unser Netz ist ja gänzlich anders strukturiert, aber wenn das ExitVPN ausfällt Routen wir auch bewusst über 2 oder 3 "ungesicherte" Anschlüsse -  seit 4 oder 5 Jahren Abmahnungs-frei.

bittorf, 2015-03-27:
hab gekackt! 4a9cacb46fa6ce256e3aa7c593ac27db5b62728d

glycoknob, 2015-03-27:
Sehr cool! Danke!

bittorf, 2015-03-28:
ich werde am dienstag mal zeigen, wie man das schoen in einer UML testet...

0x6d54, 2015-09-05:
Feature: VPN aktiv, aber Internet geht direkt raus.

glycoknob, 2016-08-12:
Gibt es als Knopf in der Firmware seit einer Weile. [x] Done.

closed 2016-08-12 by glycoknob