weimarnetz/weimarnetz #77: Speicherplatz sparen

L337hium, 2015-03-24:
folgende aenderungen im 'menuconfig' sparen bis zu 200KB (fuer 4MB Router)

Utilities --> discs --> swap-utils = n

Base System --> busybox --> Linux System Utils --> swponoff = y

Kernel modules --> Wireless Drivers --> kmod-mac80211 --> Enable 802.11s mesh support = n 

Kernel modules --> USB Support --> kmod-usb2 / kmod-usb-core / kmod-usb-ohci = n

Global build settings --> Crash logging / Enable support for printk / Enable process core dump support / Enable printk timestamps = n

Base System --> busybox --> Linux System Utils --> dmesg

Target Images --> squashfs --> (1024) Block size (in KiB)

Optional `opkg` rausoperieren

L337hium, 2015-03-24:
Also fuer einen **TP-Link 841ND v8** wir sind von 36?? KB, auf 34?? KB und jetzt auf **3268 KB** runter!

3840    bin/ar71xx/openwrt-r44620-ar71xx-generic-tl-wr841n-v8-squashfs-factory.bin
3268    bin/ar71xx/openwrt-r44620-ar71xx-generic-tl-wr841n-v8-squashfs-sysupgrade.bin

L337hium, 2015-03-25:
Die alte Firmware mit r37711 hat 3539 KB.

andibraeu, 2015-03-25:
> Optional opkg rausoperieren
> 

opkg sollte nicht raus, ist schließlich das Standardpaketmanagement für
OpenWrt. Und unsere Softwareerweiterungen kommen ja inzwischen aus als
Paket.
Da würden wir uns also Updates ziemlich verbauen

storchi, 2015-03-25:
bittorf meinte, man könnte nen alias auf opkg scripten, dass dann ansagt wo man sich mit wget das opkg binary ziehen kann und unter /bla/xxx ablegt.

bittorf, 2015-03-25:
ja, bin gerade am basteln:
# build: noOPKG ->
cd /tmp
  wget http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base/opkg_9c97d5ecd795709c8584e972bfdf3aee3a5b846d-7_ar71xx.ipk
  tar xzf opkg_*
  rm ./debian-binary ./control.tar.gz ./opkg_*
  tar xvzf data.tar.gz ./bin/opkg
  rm data.tar.gz
  /tmp/bin/opkg install mein_packet
  rm /tmp/bin/opkg

das spart nochmal 50k

bittorf, 2015-03-25:
wireless-tools = 'iwconfig' aus 'base-system' kann auch weg. spart nochmal ***50k***

bittorf, 2015-03-25:
und 'px5g' braucht man auch nicht (initialer zertifikats-generator fuer https) - den brauchen nur ganz frische knoten, die noch keins haben (wird bei sysupgrade uebernommen)

bittorf, 2015-03-25:
und JFFS2 kann gehackt werden, das man noch 2 bloecke mehr frei hat (128k)

glycoknob, 2015-03-25:
weiß nicht ob es sich für 50kb lohnt opkg rauszuoperieren... opkg für Updates ist schon ganz nett.

bittorf, 2015-03-25:
aber 'opkg' geht doch weiterhin - es wird halt nur "kurz, mal eben" aus dem internet runtergeladen und in die ram-disk kopiert...

bittorf, 2015-03-25:
falls es noch fragen gibt 8-)
bastian@gcc20:~$ id bastian
uid=1337(bastian) gid=1337(bastian) groups=1337(bastian)

bittorf, 2015-03-25:
utilities 'iwinfo' und 'px5g' also auch weg...

bittorf, 2015-03-25:
px5g muesste also auch wenn es gebraucht wird aus dem netz geladen -> in die ramdisk -> cert machen und bin wegwerfen...

bittorf, 2015-03-25:
image ist jetzt 3140k und da ist luci UND tc schon drinne...

bittorf, 2015-03-25:
mit dem 'opkg', sieht jetzt so aus:

root@EG-labor-AP:~ $OPKG info kernel
Mar 25 11:57:22 ' [up 27706s 7430144 bytes free] TERM:screen -ash [daemon.info] opkg_raminstaller() fetching filename from http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base
Connecting to downloads.openwrt.org (78.24.191.177:80)
Connecting to downloads.openwrt.org (78.24.191.177:80)
-                    100% |**********************************************************************************************|   132k  0:00:00 ETA
Mar 25 11:57:24 ' [up 27708s 7393280 bytes free] TERM:screen -ash [daemon.info] opkg_raminstaller() fetching package from http://downloads.openwrt.org/barrier_breaker/14.07/ar71xx/generic/packages/base/opkg_9c97d5ecd795709c8584e972bfdf3aee3a5b846d-7_ar71xx.ipk
Connecting to downloads.openwrt.org (78.24.191.177:80)
opkg.ipk             100% |**********************************************************************************************| 51512   0:00:00 ETA
./bin/opkg
Package: kernel
Version: 3.18.9-1-628cca0666bd03e08db24f81b10bb80c
Depends: libc
Status: install hold installed
Architecture: ar71xx
Installed-Time: 1427091548

root@EG-labor-AP:~ $OPKG info kernel
Package: kernel
Version: 3.18.9-1-628cca0666bd03e08db24f81b10bb80c
Depends: libc
Status: install hold installed
Architecture: ar71xx
Installed-Time: 1427091548

bittorf, 2015-06-07:
alles eingepflegt, bei bedarf kann man nochmal in der liste schmöckern

andibraeu, 2015-06-07:
es fehlt noch das Script, um px5g temporär zu laden.

bittorf, 2015-06-08:
ich habe probiert px5g rauszunehmen, aber das braucht nur 9k - ist also den aufwand nicht wert. bei 50k fuer opkg schon eher...

bittorf, 2015-06-08:
wer list hat kann den autodownload fuer opkg machen:
https://github.com/bittorf/kalua/blob/master/openwrt-addons/etc/kalua/software#L464

zusaetzlich muss im loader noch gesetzt sein:
OPKG="_software opkg_raminstaller"
und alle aufrufe von opkg muessen ersetzt werden durch $OPKG
(wobei das im normalfall OPKG=/bin/opkg ist....

closed 2015-06-08 by bittorf