weimarnetz/weimarnetz #79: SSID's fürs WLAN 8-)

bittorf, 2015-04-21:
Vorschlag:
1 x adhoc "weimarnetz"
1 x AP "weimarnetz (Standort)"
1 x AP "weimarnetz.de"
1 x AP "weimar.freifunk.net"
1 x AP "noch frei"

es wird fuer adhoc und fuer AP jeweils eine IP gebraucht,
das heisst fuer dual-radio-router, braucht man
3 IP's: 1 x adhoc - 2ghz, 1 x adhoc 5ghz und 1 x AP
(nur fuer roaming braucht man NAT)

bittorf, 2015-06-07:
wir sollten sparsam mit zusaetzlich SSID's umgehen, jedes virtuelle Interface kostet performance...

andibraeu, 2015-06-07:
ich denk, du hattest schon eine fertige Lösung?
Die Idee, eine feste AP-SSID zu haben fand ich gut

wir sollten das mal weiter im Hinterkopf behalten

bittorf, 2015-06-08:
achso. ich bin weiterhin dafuer
1 x adhoc
1 x AP 'weimarnetz.de'
1 x AP 'weimarnetz.de (TanteEmmaLaden)'

das geht aber nur per macvlan/roaming und das wird jetzt nicht mehr in die stabile gefummelt.

glycoknob, 2016-01-09:
Vorschlag: 

https://wiki.freifunk.net/Berlin:Firmware#Was_hat_es_mit_den_SSIDs_.22intern-chXX.berlin.freifunk.net.22_auf_sich.3F

adhoc: intern-ch05.weimar.freifunk.net
AP: wie bisher - bzw. Roaming

bittorf, 2016-01-09:
ich wuerde ungern die adhoc-SSID ähnlich zu AP machen, das verwirrt nur.
wie waere es mit: "intern-kanal5.freifunk"?

glycoknob, 2016-01-09:
Hab ich doch vorgeschlagen? Ob intern-ch05.weimar.freifunk.net oder intern-kanal5.freifunk is mir wuppe.

bittorf, 2016-01-10:
na ich finde wenn
AP weimar.freifunk.net
dann sollte adhoc anders sein und nicht
ADHOC = intern-ch05.weimar.freifunk.net
sondern
ADHOC = intern-kanal5.freifunk

schwierig. ch = channel oder einfach nur "5"?

glycoknob, 2016-01-10:
macht sinn. intern-kanal5.freifunk klingt cool :)

kaese77, 2016-01-10:
Ich will auch:

"ch5.ff" oder "ch5[.adhoc].ff"

Reicht!

VG, René

bittorf, 2016-01-10:
ja, weniger ist mehr - denn es *ist* ja intern/nicht zu nutzung.

FreifunkUFO, 2016-01-11:
das "intern" klingt ggf. aber so, dass das nur für Vereinsmitglieder sei :-o 
mit "i" vs. "w" wird jenes Netz dann auf den meisten Endgeräten (mit alphabetischer Sortierung) noch vor den eigentlichen vAPs angezeigt!?

bittorf, 2016-01-11:
@FreifunkUFO guter Einwand mit der Alphabetischen Sortierung.
was kann man statt "intern" noch nehmen? Es soll suggeriert werden das
man sich damit nicht versucht zu verbinden, aber eine technische Notwendigkeit besteht das anzuzeigen.  eventuell sowas wie:

x-mesh.ch5.ff

andibraeu, 2016-01-11:
warum muss es überhaupt lesbar sein?

Ich schlage ```zmrtjgvrje``` vor

FreifunkUFO, 2016-01-11:
(Leider nur?) teilweise ist es aber auch sinnvoll, wenn an der SSID erkannt werden kann, dass es sich um Freifunk handelt. Da wäre "ff" unpraktisch.
O.g. Zielgruppe "man" sind ja auch nur die Leute mit Smartphones. ("nicht versuchen zu verbinden"), für andere Freifunk-Router ist ja eigentlich dieses "intern" Netz zu verwenden, um sich mit dem Freifunk zu verbinden.. auch hier scheint der Name "intern" verwirrend.

Viel Glück noch bei der SSID-Findung, siehe Erfurt ist die Diskussion darüber auch geeignet, alle technischen Hintergründe zu vernachlässigen und sich darüber zu zerstreiten. :-/

andibraeu, 2016-01-11:
eigentlich wollten wir die adhoc-ssid verstecken, das geht aber technisch nicht, ohne auch die AP-SSId zu verstecken. den routern ist es egal, wie die SSID heißt, hauptsache die BSSID ist gleich. Der Name ist also relativ egal, er sollte nur nicht zum Verbinden animieren

storchi, 2016-01-11:
FreifunkBackbone ...?

bittorf, 2016-01-11:
ja, backbone finde ich auch gut. damit es nach unten sortier wird, sollte noch ein 'x' davor.

x-backbone.ch5.freifunk

glycoknob, 2016-01-11:
+1 für Backbone. 

@FreifunkUFO: Auf WLAN Ad-Hoc ist der DHCP bei uns schon seit ewigkeiten kaputt, Clients bekommen da keine Adresse - momentan ist die SSID "weimarnetz" und irritiert sicherlich viele.

kaese77, 2016-01-12:
> ja, backbone finde ich auch gut. damit es nach unten sortier wird, sollte noch ein 'x' davor.
> 
> x-backbone.ch5.freifunk

"xbckb.ch5.freifunk"

René

storchi, 2016-01-12:
man zeige mir ein gerät, dass die wlans nach alfabet sortiert
cs

FreifunkUFO, 2016-01-13:
@andibraeu zu ad-hoc-ssid verstecken: Habt ihr schon versucht, die ad-hoc-SSID "leer" zu setzen? (siehe freifunk leipzig chat)
@storchi dann komm mal vorbei, ich habe das zb beim Mac und auf Android-Smartphones (wahrscheinlich unterschiedlich je nach verwendeter App)
@glycoknob In Leipzig geht ad-hoc auf DHCP auch schon lange nicht mehr (allerdings haben wir das gezielt abgeschaltet). Nutzer sind lernfähig, zumal ad-hoc im Endgerät anders angezeigt wird. Eigentlich bräuchte man also gar nichts ändern! Habt ihr wirklich viele verwirrte Leute und welche SSID schlagen diese Leute denn vor?
@kaese77 Das "ch" könnte ja auch noch weg. das "xbckb" ist so kryptisch wie @andibraeu es sich gewünscht hat.

kaese77, 2016-11-02:
bzgl. #240 :

Beschlußvorlage für 08.11.2016:

1 x 'xxx[.]5.freifunk'  || 'x5.freifunk'  # ( erstes sieht unanständig und für manche vllt. einladend aus ;-) )
1 x AP 'weimarnetz.de'
1 x AP 'weimarnetz.de (TanteEmmaLaden)'

glycoknob, 2017-01-24:
Aktuell im Code: 

adhoc: ff.intern.<knotennr>.ch<channel> 
ap: weimar.freifunk.net

mtippmann, 2018-08-14:
alles konfigurierbar im profile jetzt...

closed 2018-08-14 by mtippmann