weimarnetz/weimarnetz #80: Menu und Splash auf Mobilen Geräten vertragen sich nicht...

glycoknob, 2015-04-26:
[image]

L337hium, 2015-04-26:
das geraet in der breitbildansicht verwenden hilft *raeusper* -- na wenn wer bootstrap beherscht koennte man das mal fixen, abernaja...

andibraeu, 2015-04-27:
*raeusper* http://getbootstrap.com/2.3.2/

glycoknob, 2015-04-27:
So ein Hamburger ☰ wäre cool für das Menü auf mobilen Geräten.. das geht wohl erst ab Bootstrap3.x..

Hab grad hier entdeckt: https://developer.chrome.com/devtools/docs/device-mode damit müsste man das schnell fixen können... bootstrap lernen ist nicht viel aufwand, ist nur css und html.

glycoknob, 2015-04-27:
Hat sich erledigt.. ist schon im aktuellen luci-gefixt: https://github.com/openwrt/luci/pull/333 

Man muss aber den Cache löschen damit es neu angezeigt wird, kann man vielleicht über Header lösen.

closed 2015-04-27 by glycoknob