weimarnetz/weimarnetz #99: Routertests

andibraeu, 2015-06-16:
# Test Szenarien 
## Image
- [ ] Imagegröße ok für 4MB Geräte

## Ursprungszustand Router
- [ ] fabrikneu mit tp-link firmware
- [ ] weimarnetz Firmware mit Beibehaltung der Einstellungen
- [ ] weimarnetz Firmware mit neu Schreiben der Einstellungen

## Getestete Hardware
- [ ] 841er
- [ ] 3600er
- [ ] Nanostation Loco M2
- [ ] Nanostation Loco M5

## Registrator
- [ ] bei vorhandenem und gültigem SSH-Key bestätigt der Registrator die Knotennummer
- [ ] stimmt der SSH-Key nicht überein (=Nummer schon vergeben) bekommt der Client die nächste freie Nummer und apply_profile schreibt die Config mit der neuen Knotennummer neu
- [ ] nach 30 Tagen ohne Rückmeldung wird eine Knotennummer wieder freigegeben und kann neu vergeben werden

## Splash-Page
Spash-Page fuer WLAN-Clients auf verschiedenen Client Platformen testen (mit und ohne Javascript testen)
- [ ] Android 2.x
- [ ] Android 4.x
- [ ] iOS
- [ ] Windows Phone
- [ ] Win 7.8, ...
- [ ] Max OS X (Safari, Chrome, FF, ...)
- [ ] Firefox unter Linux ...
- [ ] LAN-Clients sollten nicht gesplasht werden.

## VPN
- [ ] startet VPN nach dem das WAN Interface in den Link UP Status wechselt. (muss fuer jede Hardware extra getestet werden) - ohne Frage
- [ ] läuft olsr am vpn interface
- [ ] Werden WLAN-Clients ueber das VPN geroutet?
- [ ] Werden LAN-Clients nicht ueber das VPN geroutet?
- [ ] Wird das VPN beendet, wenn WAN zu down wechselt?
- [ ] Wechselt die VPN-Verbindung im laufenden Berieb, wenn der VPN-Server ausfaellt/nicht mehr erreichbar ist, auf einen anderen VPN-Server.
- [ ] werden endgeraet >1hop auch durch tunnel geleitet

## OLSR
- [ ] wlan nachbarn vorhanden
- [ ] lan nachbarn vorhanden
- [ ] tc priorisiert OLSR-Pakete (wie kann ein Test aussehen? Reicht es, den Testdownload zwischen zwei Routern anzuwerfen?
    Infos gibt auch tc -s -d qdisc show <devname>
    ein guter Test ist nur auf einem schwachen link moeglich. also z.b. DNT <=> Musikschule Ottmar Gerster

## WLAN
- [ ] adhoc interface
- [ ] ap interface
- [ ] ssids

## Fragen und Infos

* was ist mit dem registrator problem beim neu flashen des gleichen routers?
 * siehe Registratortests
 * nach 30 Tagen neu vergeben? da würde ich eher 2 jahre vorschlagen, da es sonst zu häufig zu knotennummer wechseln kommt dann lieber eine deregistrierung funktion erfinden
 * Ich würd das einfach mal beobachten, ob das nach 30 Tagen passiert. Kommt ja nur vor, wenn sich ein Router 30 Tage lang gar nicht meldet
* wollen wir offiziell nur die meshkit firmware unterstützen?
 * cs sagt ja
 * bastian sagt auch ja (solange wie es keine weboberfläche für build.sh gibt)
 * bin ich auch dafür, alles andere läuft unter experimentell. Mehr Quellen als Meshkit können wir nicht stemmen
* wollen oder müssen wir mehr Hardware unterstützen? z.B. 4300er oder 1043er?
 * cs sagt nein -> eigendrösel
 * mehr müssen wir nicht intensiv testen, das sollten wir auch kommunizieren. Aber jeder darf auf eigene Gefahr auch im Meshkit sich eine Firmware für andere Geräte erstellen lassen -- Andi 2015-06-10 10:51 UTC+2

* Build-tests:
 * ../build.sh --openwrt r45938 --hardware 'TP-LINK TL-WR841N/ND v8' --usecase 'freifunk-4mb,weimarnetz'  ==> 2.949.124 Bytes
 * ../build.sh --openwrt r45938 --hardware 'TP-LINK TL-WR1043ND'       --usecase 'freifunk,weimarnetz'          ==> 4.915.204 Bytes
  * (nur zwei Beispiele, es kann nicht allgemeingültig sein, weil für jedes Modell etwas andere Zahlen durch andere Innereien/Bauteile sind)

* sysupgrade /tmp/fw : wie soll das neue kalua packet installiert werden?
 * hat nix mit Testszenrarien zu tun
 * kalua selbst ist ja schon immer mit im Image integriert
 * nichtsdestotrotz sollte das kalua-Paket als openWRT-Paket wie andere Pakete auch gebaut und installiert werden (machen wir im Weimarnetz-Repo aktuell so). Die Firmware ist so konfiguriert, dass sie package repositories auf http://www.weimarnetz.de/freifunk/firmware/packages/ sucht. Dort können wir dann auch neue Versionen von Kalua oder der Splashpage bereitstellen. Ein Update erfolgt dann wie in OpenWrt üblich mit opkg update && opkg install kalua -- Andi 2015-06-10 10:58 UTC+2

andibraeu, 2015-06-16:
=> Kopiervorlage

## Test Einzelrouter
* Name: 
* Modell: 
* Knotennummer: 
* verwendetes Image:
* Zustand vor Update: 
* Tests:
- [ ] WLAN:
- [ ] OLSR: 
- [ ] VPN: 
- [ ] Splash Page:
- [ ] Registrator: 

## Anmerkungen/Auffälligkeiten

andibraeu, 2015-06-16:
## Test tp-dev
* Name: tp-dev
* Modell: 1043nd
* Knotennummer: 23
* verwendetes Image: von meshkit.ja.ishalt.so, cc1505_rc1
* Zustand vor Update: vorher Weimarnetz, Update mit Erhaltung aller Einstellungen
* Tests:
- [x] WLAN: 1x AdHoc Weimarnetz, 1x AP weimarnetz Treffen Dienstags M18
- [x] OLSR: hat WLAN-Nachbarn, Priorisierung getestet: zero-download mit anderem Router, der auch tc verwendet: ETX verändert sich marginal von 1,3 auf 1,4, zero-download mit Router ohne tc: ETX geht auf 8
- [x] VPN: nicht relevant, hängt nicht am LAN
- [x] Splash Page: Splash für WLAN-Clients funktioniert auf: iOS, Chromium, Firefox, LAN-Clients nicht angeschlossen
- [x] Registrator: Update bestehender Knotennummer hat funktioniert

## Anmerkungen/Auffälligkeiten
keine

andibraeu, 2015-06-16:
## Test SchubertSpielplatz
* Name: SchubertSpielplatz
* Modell: Bullet M2
* Standort: Schubert31
* Knotennummer: 17
* verwendetes Image: von meshkit.ja.ishalt.so, cc1505_rc1
* Zustand vor Update: vorher Weimarnetz, Update mit Erhaltung aller Einstellungen
* Tests:
- [x] WLAN: 1x AdHoc Weimarnetz, 1x AP weimarnetz Frei wie in Freiheit
- [x] OLSR: hat WLAN-Nachbarn, LAN-Kopplung mit Knoten 73
- [ ] VPN: nicht relevant, hängt nicht am Internet
- [x] Splash Page: Splash für WLAN-Clients funktioniert auf: iOS, Chromium, Firefox, LAN-Clients nicht angeschlossen
- [x] Registrator: Update bestehender Knotennummer hat funktioniert

## Anmerkungen/Auffälligkeiten
bei einem Updateversuch war die falsche IP für die WLAN-Devices gesetzt, war aber nicht reproduzierbar und

andibraeu, 2015-06-16:
## Test Cafe S140
* Name: s140
* Modell:  841ND v8.2
* Standort: M18, S140
* Knotennummer: 154
* verwendetes Image: von meshkit.ja.ishalt.so, cc1505_rc1
* Zustand vor Update: sysupgrade mit Erhalt der Einstellungen nicht mehr möglich
* Tests:
- [x] WLAN: 1xAP Weimarnetz - S140, 1xAdHoc weimarnetz
- [x] OLSR: hat WLAN-Nachbarn, Internetzugriff funktioniert
- [ ] VPN: 
- [x] Splash Page: Android, iOS getestet und funktioniert
- [x] Registrator: alte Knotennummer behalten nach Rückkopieren von ssh-pubkey (/etc/dropbear/*)

## Anmerkungen/Auffälligkeiten
* WAN-config musste angepasst werden, um den Router an dem Anschluss betreiben zu könne. Feste IP

glycoknob, 2015-06-18:
## Test TP-Link 841NDv8.2 

* Name: netbox 
* Modell: 841ND v8.2
* Standort: Bei mt zuhause ;)  
* Knotennummer: 188 
* verwendetes Image: von meshkit.ja.ishalt.so, cc1505_rc1  - sysupgrade ohne n
* Zustand vor Update: letzte stabile meshkit.weimarnetz.de Version 
* Tests: 
[-] OLSR: hat WLAN-Nachbarn, Internetzugriff funktioniert *NICHT*

## Anmerkungen: 
Sehe OLSR nachbarn, kann nur einen von beiden pingen... wenn das läuft, dann teste ich noch weitere Probleme

```
root@netbox:~ n
Table: Links
Local IP        Remote IP       Hyst.   LQ      NLQ     Cost
10.63.188.1     10.63.71.195    0.00    1.000   1.000   1.000
10.63.188.1     10.63.33.129    0.00    0.765   1.000   1.308

Table: Neighbors
IP address      SYM     MPR     MPRS    Will.   2 Hop Neighbors
10.63.71.225    YES     YES     YES     7       2
10.63.33.161    YES     YES     YES     7       1
```

Die 10.63.71.195 ist der MAC1200R mit kalua.. I'll investigate! 

```
root@netbox:~ ping 10.63.33.129
PING 10.63.33.129 (10.63.33.129): 56 data bytes
64 bytes from 10.63.33.129: seq=0 ttl=64 time=2.628 ms
64 bytes from 10.63.33.129: seq=1 ttl=64 time=2.497 ms
64 bytes from 10.63.33.129: seq=2 ttl=64 time=2.363 ms
64 bytes from 10.63.33.129: seq=3 ttl=64 time=1.555 ms
64 bytes from 10.63.33.129: seq=4 ttl=64 time=2.470 ms
64 bytes from 10.63.33.129: seq=5 ttl=64 time=1.870 ms
64 bytes from 10.63.33.129: seq=6 ttl=64 time=1.983 ms
^C
--- 10.63.33.129 ping statistics ---
7 packets transmitted, 7 packets received, 0% packet loss
round-trip min/avg/max = 1.555/2.195/2.628 ms
root@netbox:~ ping 10.63.71.195
PING 10.63.71.195 (10.63.71.195): 56 data bytes
^C
--- 10.63.71.195 ping statistics ---
25 packets transmitted, 0 packets received, 100% packet loss
root@netbox:~ 
```

storchi, 2015-06-23:
## Test 3600er
* Name: cs-test
* Modell: 3600
* Standort: cranach47
* Knotennummer: 122
* verwendetes Image: von meshkit.ja.ishalt.so, cc1505_rc1 - factory
* Zustand vor Update: WN Zwischenversion 42742 als Knoten 112
* Tests:
- [ ] WLAN: AP im 5ghz wird nicht konfiguriert - soll das so?
- [ ] OLSR: adhoc ok, lan ok 
- [ ] VPN: spinnt voll, debuggen!
- [ ] Splash Page: im chrome keine umleitung ohne internet, aufruf 8.8.8.8 geht -> dns ding
- [ ] Registrator: ok, image kam mit neuer knotennummer

## Anmerkungen/Auffälligkeiten

closed 2016-01-17 by andibraeu