weimarnetz/weimarnetz #100: vpn: check vtun availabilty

andibraeu, 2015-06-23:
client tries to connect to a non-existing vtun server, if server is pingable, but no vtun is runnig

bittorf, 2015-06-23:
nc -w 3 -z 4.vpn.weimarnetz.de && echo 'ja geht'

glycoknob, 2015-06-23:
```
[ $(echo "test" | nc -w 3 3.v.weimarnetz.de 5001 | grep -c VTUN) -gt 0 ] \
   && echo "läuft" ||  echo "kaputt"
```

bei der Gelegenheit könnte man auch einbauen, die Liste der Server durchzugehen wenn der mit dem schnellsten Ping nicht läuft... dafür müsste das `persist yes` auf der `vtund.conf` weg. 

Beim heutigen Disconnect-Bug würde obiger Check auch nicht helfen, aber wenn alle Server durchprobiert würden, dann ist die Chance höher das einer läuft... 

Mal sehen

commit 07cac443 (2015-06-24): kalua: net: tcp_port_reachable() initial implementation - used in vpn-script. fixes #100
closed 2015-06-24 by bittorf