weimarnetz/weimarnetz #103: /etc/kalua/links holt immer 404 URLs aller 5 Sekunden

glycoknob, 2015-07-21:
```
Tue Jul 21 13:19:40 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:19:45 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:19:51 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:19:57 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:20:03 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:20:10 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:20:15 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
Tue Jul 21 13:20:22 2015 daemon.info fff+ /bin/sh wget_do(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware//packages/images/stable/info.txt'
```

Ist wohl die Update funktionalität... aber aller 5 Sekunden ist etwas oft.

bittorf, 2015-07-28:
wir sollten zumindest eine platzhalter dorthin tun, wo drin steht: "# kein update" - das kann aber nur der admin vom server...

andibraeu, 2015-07-28:
warum holt der das so oft ab? reicht doch, wenn er einmal am tag guckt und sich schlafen legt, wenn er nichts findet (404=nicht gefunden)

storchi, 2015-07-28:
sollten wir nicht.
bei der letzten ähnlichen aktion sind die broadcom geräte stehen geblieben

storchi, 2015-07-28:
mein 3600er macht das nicht. wie kann ich das reproduzieren?
cs

andibraeu, 2015-07-28:
was steht in ```/etc/config/system```? Gibts da Einträge für den Updateserver?

bittorf, 2015-07-30:
es passiert wenn alle 6 stunden das mini-script-firmware-update durchlaeuft und man auf beta oder testing gestellt hat:

for METHOD in stable beta testing; do echo "$METHOD"; wget -qSO - http://intercity-vpn.de/firmware/packages/images/$METHOD/info.txt | grep HTTP;done

die broadcom-geraete sind deswegen stehen geblieben *weil* sie nie aktualisiert wurden und die festplatte voll war. der fehler war seit *3* jahren behoben und keine hatte es eingespielt.

bittorf, 2015-07-30:
man kann es einfach reproduzieren mit:
_firmware update_pmu
oder
_firmware update_pmu beta

andibraeu, 2015-07-31:
wenn ich das jetzt richtig interpretiere läuft die for-Schleife für alle modi. Die Frage ist aber, warum probiert er es dann so ausdauernd und häufig. Das sollten wir unterbinden.

bittorf, 2015-07-31:
er probiert deswegen so haeufig, weil 404 als fehler behandelt wird und auch bei schlechten verbindungen ein update klappen soll. ich habe mal den wget-code aktualisiert:
9535f3dd6fcf6fb3d15c20c0cccaeb6c3c2bfadf

closed 2015-07-31 by bittorf