weimarnetz/weimarnetz #107: VPN: Tests werden beim Start 2mal durchlaufen

andibraeu, 2015-09-01:
außerdem: Umstellen auf Zufall

storchi, 2015-09-02:
Moin, hab mal nachgedacht.
Die Möglichkeit einfach so, ohne DNS Einträge einen neuen VPN Server an den Start zu bringen klingt schon irgendwie cool. Das wäre die Lösung mit zwei hart kodierten VPN Servern, wovon mindestens einer immer gehn muss, damit ein VPN client dem Netz beitreten kann und dann übers olsr Dienstplugin andere VPN Server angekündigt bekommt. Nachteil: VPN muss mind. zwei mal nacheinander gestartet werden. Krasser Nachteil: top Abschnorchel Möglichkeit. Für mich überwiegen die Nachteile.
cs

glycoknob, 2015-09-02:
Wäre viel Arbeit... würd das auch erstmal hinten anstellen - wir müssten die Ausgabe vom olsrd-Nameservice-Plugin erstmal auswerten und dann dort auch eventuellen Bogus rausfiltern... wegen Missbrauch/Abschnorcheln: Auf allen Routern ist `usign` - wir können also unsere VPNs mit einen private-key den wir kennen signieren... so eine Signatur ist kurz genug, dass der auch über den OLSR-Nameservice mitgeliefert werden kann... Problem gelöst.

storchi, 2015-09-02:
Haben wir noch das Problem mit dem doppelten VPN Neustart. Oder man lässt den neuen Router erstmal bei einem der zwei festen VPN Servern (betrifft ja nur neue Router), merkt sich alle anderen VPN Server und benutzt die beim nächtlichen Neustart. Und einmal am Tag wird die VPN Serverliste aktualisiert?
Probleme?
cs

glycoknob, 2015-09-02:
Ich hatte überlegt einfach eine VPN-Liste anzulegen z.B. in uci /etc/config/ffweimar-vpn oder so... und der VPN-Server nutzt dann einfach nur diese Liste beim Neustart/Reboot - dann halt zufällig oder so... die Liste kann man ja beim booten oder aller paar Stunden aktualisieren... dann wäre beides getrennt

storchi, 2015-09-02:
klingt doch vom konzept erstmal ganz gut. wollen wir alle vpn server durch eine signatur schützen? (auch die 2 festen), die sind ja schon durch die DNS Konfiguration "geschützt". Ich sage ja, macht wahrscheinlich auch die umsetzung einfacher, wenn einfach jede vpn verbindung den signatur test macht.
cs

glycoknob, 2015-09-02:
Müsste man sich überlegen was man signieren will... ich hätte jetzt einfach
gedacht die eine Zeile für das Nameservice-Plugin wo dann so etwas
drinsteht wie vpn://test.vpn:5000|nolzo|testvpn  und dann halt eine zeile
mit testvpn:<hash> .... ich übeleg mir was..

Das aktuelle kalula mit OpenWRT-Trunk kann sich auch grad nicht zum
VPN-Verbinden obwohl die Server laufen... muss das erstmal debuggen.

andibraeu, 2015-09-05:
beim Stoppen wurden die Tests durchgeführt. Deswegen führte es bei einem Restart dazu, dass die Tests 2 mal durchlaufen wurden

closed 2015-09-05 by andibraeu