weimarnetz/weimarnetz #113: etx Wechsel - Termin?

glycoknob, 2015-09-05:
1.1.2016 für den Wechsel des ETX-Verfahrens wird wohl ziemlich kritisch.... später?

bittorf, 2015-09-05:
quark, das schaffen wir locker!

andibraeu, 2015-09-05:
ich bin auch für eine Verschiebung nach hinten. Wir haben das bisher nicht getestet. Vor allem 1.1. ist schon ziemlich doof, da ist Weihnachten und Kongress. In den Wochen davor bleibt da erfahrungsgemäß nicht viel Zeit

bittorf, 2015-09-06:
jetzt flennt mal nicht rum, das sind noch 4 Monate!

glycoknob, 2015-09-06:
Dann machen wir 5 oder 6 draus und am Wochenende des Wechsels, können wir zur Not von Hand Router updaten oder flicken... am 1.1. und in der Woche darauf wird das niemand machen wollen (können). Oder?

bittorf, 2015-09-06:
ok, also auf ein bis zwei wochen kommt wirklich nicht drauf an, dafuer.

bittorf, 2015-11-17:
die frage ist nur, wie man das elegant und klar im Code ausdrückt. Ich denke es sollte einfach so bleiben...

andibraeu, 2015-11-17:
Neujahr ist und bleibt ein sehr ungünstiger Zeitpunkt.

Und wegen elegant und klar: Unix-Timestamp geht doch auch und lässt sich gut vergleichen, muss eben noch ein Kommentar dazu

storchi, 2015-11-18:
Der Termin fuer die etx metrik umstellung wurde vom 1.1.16 um ca. 2 wochen nach hinten verschoben.

1) muessen die router, die vom 1.1. wissen umgestellt werden
2) brauchen wir eine Liste von allen Routern
3) brauchen wir eine Liste der Router, auf die wir nicht drauf kommen
4) brauchen wir eine Liste der Router, die die neue metrik nicht können (alte broadcom geräte)

bittorf, 2015-11-18:
zu 1)
die werden sowieso noch umgestellt, denn es *muessen* sowieso alle geraete angefasst werden.

bittorf, 2015-11-18:
pullrequest fuer umschaltdatum wurde angepasst:
82aa3b8e6e81ae4bc9ca37709b3cd38b16792065

andibraeu, 2015-11-23:
siehe auch #160

closed 2015-11-23 by andibraeu