weimarnetz/weimarnetz #115: Zugriff auf WAN-Subnetz möglich

glycoknob, 2015-09-09:
Moin, 

Ich hab grad Kaffee vor meinen Haus getrunken und festgestellt, dass ich über Freifunk problemlos auf alles in meinen `192.168.2.0/24` Subnetz zugreifen kann... 

Das ist IMHO übel.... so kann jeder der einen Knoten aufstellt z.B. auf private Freigaben zugreifen, viele Leute teilen auch Sachen zuhause, die sie sonst nicht teilen. Ich kann z.B. über meinen Drucker drucken. 

Die Funktioniert ist schon in `luci-mod-freifunk` drinne - siehe http://doc.meshkit.freifunk.net/daily/html/usage/firewall.html `restrict_local` 

Wie fixen wir das?

bittorf, 2015-09-09:
im prinzip ist unser NAT zu gut.

momentan (in _netfilter_masquerade_start)
iptables -t nat -I POSTROUTING -o $WANDEV -j MASQUERADE
iptables -t nat -I POSTROUTING -o $WANDEV ! -d $WANNET/$WANPRE -j MASQUERADE

es wird dann also nur maskiert, wenn die zieladresse nicht im WANNET liegt.
das hat zur folge das die pakete zwar hinkommen, aber es keine rueckroute
gibt. deswegen klappts nicht.
kannst du bitte mal die zweite zeile bei dir eintragen und schauen obs hilft?

glycoknob, 2015-09-09:
Ich teste gleich mal. Danke!

storchi, 2015-09-10:
andere sehen sowas als feature...

bittorf, 2015-09-10:
wir koennten zu mindest ne uci-variable dafuer einfuehren und per default geht es halt...aber erstmal muss jemand meine zeile testen.

glycoknob, 2015-09-10:
Ich teste das gleich mal zusammen mit den VPN-Änderungen... ein wenig Geduld.

glycoknob, 2015-09-12:
Sieht gut aus! Komme noch ins Netz, aber Zugriff das lokale Netz klappt nicht mehr! Ich bastel morgen mal einen Pull-Request mit Option dafür... ob Default oder nicht können wir ja ausdiskutieren.

andibraeu, 2015-12-14:
@glycoknob  gibt es hier noch was zu tun?

glycoknob, 2015-12-14:
@andibraeu: Ja, muss nochmal testen.

glycoknob, 2015-12-21:
Also per Default kommt man an alles was so am WAN-Netz hängt... 

wenn man `restrict_local=1` in `/etc/config/system` setzt, dann ist das WAN tabu. 

Frage wäre, was man als Default haben will - ich denke restrict_local='1' als Default ist nicht so schlecht. Es gibt ziemlich viele Fritzboxen, die per Default offen sind ohne Passwort, irgendwelche Windows oder Apple-Shares mit privaten Kram... wäre blöd da Leute blosszustellen, wenn sie Freifunk wollen.

glycoknob, 2015-12-22:
Also ich wäre für: 

Default - angeknipst 
Knopf in den Settings 

Geht das klar? Dann bastel ich ein Ziehbegehren.

andibraeu, 2015-12-22:
ja, Default sollte an sein, wir sagen ja jedem, dass das eigene LAN nicht erreichbar ist.

in welche Section muss das in /etc/config/system?

Das muss dann in applyprofile.code angepasst und auf der Konfigseite ergänzt werden

glycoknob, 2015-12-22:
Wie wird das denn hier gehandhabt? https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/kalua/netfilter#L802 

Ist es schlimm, wenn da noch die [0] in der section steht?

bittorf, 2015-12-22:
nein, die [0] ist auch kompatibel mit neu.

andibraeu, 2015-12-22:
system hab ich so gelassen, weil das evtl. von openwrt so geschrieben wird

andibraeu, 2015-12-23:
das ist auf der Settingspage nun auch einstellbar: https://github.com/weimarnetz/openwrt-packages/commit/9715f111caae6d346e52387244981154190747e3

closed 2015-12-23 by andibraeu