weimarnetz/weimarnetz #119: Kompatibilität mit kalua-Firmware für Roaming-Netze

glycoknob, 2015-09-13:
Hallo, kalua macht ja Roaming mit macvlan im Netz 192.168.0.0/16 - offensichtlich kommt unsere Firmware nicht damit klar und die Daten landen nicht beim VPN-Server. 

Jedenfalls Router mit kalua und Roaming kommen nicht über Weimarnetz-Firmware Router ins Netz. Zumindest nicht über den VPN. 

Das müsste eigentlich klappen, aber mit `ip rule` hat es nicht geklappt.. vermutlich weil 192.168.0.0/16 auch vom WAN genutzt wird und man da noch ein Device benötigt für `ip rule` das geht wohl mit `iif` und `oif`. 

Hier nur damit es nicht vergessen wird, müssen wir mal schauen. Wäre ja blöd, wenn das nicht klappen würde.

bittorf, 2015-09-14:
ich will die IP's fuers roaming auch noch aendern, das war ja nur ein erster test (der leider sonst sehr gut laeuft...)

glycoknob, 2015-09-14:
Also es funktioniert auch jetzt schon mit dem Roaming IPs - nur gehen die nicht über das VPN raus... 

Das funktioniert jetzt aber auch mit https://github.com/glycoknob/weimarnetz/commit/23249aa4416a08e21faa325137f47b1d2376fef9 - ohne Probleme soweit (bis auf, dass wohl einer der VPN-Server noch nicht aktualisiert ist - siehe Mail auf der Liste dazu).

bittorf, 2015-09-15:
danke

glycoknob, 2015-09-17:
Gibt noch einen neuen Adressbereich der niemand stört und inhaltlich auch halbwegs passt: 100.64.0.0/10 - https://tools.ietf.org/html/rfc6598 sowie: https://en.wikipedia.org/wiki/Carrier-grade_NAT - ist ja letzlich unser Setup, wenn man beim VPN NAT macht... würde sich halt nicht mit 192.168.0.0/16 beißen.

bittorf, 2015-09-17:
interessante idee.

glycoknob, 2016-08-12:
geht bei #36 weiter.

closed 2016-08-12 by glycoknob