weimarnetz/weimarnetz #121: harmonize variables between kalua and luci-weimar-splash

andibraeu, 2015-09-22:


bittorf, 2015-09-23:
puh, du meinst uci? bitte mal nur die variablen benennen, z.b.
weimarnetz$ git grep 'uci -q get' | sed -n 's/^.*-q get \(.*\) .*/\1/p' | cut -d' ' -f1 | sort | uniq | grep -v wireless.radio | grep -v wifi-iface|grep -v network\. | grep -v ^dhcp\.

firewall.@adblock[0].enabled
meshwizard.community.name
meshwizard.contact.mail
meshwizard.contact.name
meshwizard.contact.nickname
meshwizard.contact.note
meshwizard.contact.phone
meshwizard.system.conloglevel
meshwizard.system.latitude
meshwizard.system.location
meshwizard.system.longitude
olsrd.@Hna4[0].netaddr
olsrd.@Hna4[0].netmask
olsrd.@Interface[$i].interface
olsrd.@meta[0].hnaslave
olsrd.@olsrd[0].disabled
'olsrd.@olsrd[0].LinkQualityAlgorithm'
opkg.@packages[0].essential
opkg.@packages[0].private
-q
system.@community[0].splash
system.@ethers[0].$MODE
system.@ethers[0].wifi
system.@fwupdate[0].mode
system.@fwupdate[0].url
system.fwupdate.mode
system.@monitoring[0].serverip
system.@monitoring[0].url
system.@netfilter[0].lan_masquerading
system.@profile[0].ipsystem
system.@profile[0].name
system.@profile[0].nodenumber
system.@rdate[0].server
system.@system[0].hostname
system.@system[0].latitude
system.@system[0].latitude),$(uci
system.@system[0].longitude
system.@system[0].noswinstall
system.@system[0].timezone
system.@system[0].zram_size_mb
system.@vpn[0].active
system.@vpn[0].anonym
system.@vpn[0].domain
system.@vpn[0].enable
system.@vpn[0].force
system.@vpn[0].ipaddr
system.@vpn[0].jsonpath
system.@vpn[0].prefix
system.@vpn[0].proto
system.vpn.port
system.@weblogin[0].auth_credentials
system.@weblogin[0].authserver
system.@weblogin[0].default_speed_down
system.@weblogin[0].default_speed_up
system.@weblogin[0].enabled
system.@weblogin[0].force_lan_reachable
system.@weblogin[0].mac_unshaped
system.weblogin.email2owm
system.weblogin.enabled
uhttpd.main.max_requests

glycoknob, 2015-09-23:
Es geht um das Konfigurationsgui: https://github.com/weimarnetz/openwrt-packages/commit/11ca3ec1673d7477586f3a2c56bed8331107b532

Konkret wären das 

    system.@system[0].noswinstall='1'

Das ist glaub ich nur im Weimarnetz - noswinstall ist ein blöder Name... negation ist auch nicht so schön (IMHO). 

system.@system[0].restrict_local='1'

Das habe ich eingebaut - kein Zugriff aus WAN - wird in _netfilter start verwendet 

system.weblogin.enabled='0'

Der Schalter für die Splash-Page

system.profile.nodenumber='188'

Gibt eine Option zur Änderung der Knotennummer - die führt auch nochmal appy_profile.code aus... hier müsste mal mal testen ob das wirklich klappt 

system.fwupdate=fwupdate
system.fwupdate.url='http://weimarnetz.de/freifunk/firmware/'
system.fwupdate.mode='stable'

Firmware Update vs. noswinstall... hier kann man den Branch auswählen. 

system.vpn.enable='1'
system.vpn.force_block_wifi='1'
system.vpn.force_block_lan='1'

VPN-Regeln - die werden in _vpn block gesetzt - VPN über LAN soll nicht ins Menü.

andibraeu, 2015-09-23:
es ging mir gar nicht darum, über alle variablen zu diskutieren, sondern nur darum, wo sie sich zwischen der luci-weimar-splash-Eimstellungsseite und Kalua unterscheiden:

das wären:
* system.vpn.force_block_wifi='1' (kalua) vs. system.vpn.disableinternet (luci)
* restrict_local (kalua) - gibts noch nicht auf der Einstellungsseite
* system.vpn.force_block_lan fliegt eh wieder raus

der Rest ist eher unstrittig.

Bei der Knotennummer guckt die apply_profile.code übrigens in dieser Reihenfolge nach der Knotennummer:

1. Argument bei Aufruf
1. meshwizard.community.nodenumber (diese datei wird im meshkit erzeugt)
1. system.profile.nodenumber

dann schreibt sie die Nummer in system.profile

bittorf, 2015-09-24:
zu 'system.@system[0].noswinstall' - es heisst auch bei mir so. doofer name, gebe ich dir recht. wollen wir das umbenennen oder einfach so lassen?

bittorf, 2015-09-24:
```
system.@system[0].restrict_local='1'
```
habe ich uebernommen

bittorf, 2015-09-24:
```
system.@weblogin[0].enabled
```
habe ich auch so

bittorf, 2015-09-24:
```
system.@profile[0].nodenumber
```
habe ich auch so

bittorf, 2015-09-24:
```
system.@fwupdate[0].url
system.@fwupdate[0].mode
```
habe ich auch so.
mode kann bei mir sein: 0|stable|beta|testing wobei <leer> = stable ist
(ist beim weimarnetz genauso)

andibraeu, 2015-09-24:
es geht nicht um Kalua und Weimarnetz :-) sondern nur um die Variablennamen, die wir auf der Einstellungsseite benutzen und wie sie in unserer Firmware heißen

bittorf, 2015-09-24:
die 3 vpn-sachen muss ich bei mir noch einbauen.
(bin da aber noch unsicher, weil ich das VPN fuer verschiedene sachen benutze)

andibraeu, 2015-09-24:
system.vpn.enable='1' würd ich umbenennen in system.vpn.mode

das beschreibt es nämlich besser.

dann gibts die 3 modi:
1. aus
1. an inkl. internet
1. an exkl. internet

bittorf, 2015-09-24:
das halte ich fuer eine gute idee. welche modi kommen in frage?

andibraeu, 2015-09-25:
na die :-)

1. aus
1. an inkl. internet
1. an exkl. internet

oder was meinst du?

bittorf, 2015-09-25:
ne klar, aber wie willst du die genau benennen?
meiner meinung nach ist es aber vielleicht besser den modus nicht auf 1 sache zu beschraenken, sondern evtl. sogar 2 vpn-sachen zu starten. ich moechte z.b. nicht nur das innercity-vpn zuenden, sondern gleichzeitig auch noch ne VPN-verbindung zu 2 rechnern halten. hmmm...

andibraeu, 2015-09-25:
achso :-)

aus => vpn_off
an inkl. internet => vpn_full
an exkl. internet => vpn_cloud_only (dann haben wir auch cloud im Quelltext stehen :) )

aus Simplizitätsgründen würde ich das erstmal bei einer VPN-Verbindung belassen. Das ist der Standardfall, wer mehr will muss selbst basteln. Zumindest im Weimarnetz sehe ich keinen Bedarf dafür

bittorf, 2015-09-27:
ich meine folgendes waere besser:
mode = off
mode = on
mode = innercity

andibraeu, 2015-09-28:
ja, klingt gut. @glycoknob ? :-)

glycoknob, 2015-09-28:
Ja, bin die letzten Tage nicht dazu gekommen. Ich bastel mal was

andibraeu, 2015-11-19:
Variablen werden gesetzt und genutzt

closed 2015-11-19 by andibraeu