weimarnetz/weimarnetz #122: rene will x86 (cs ist dagegen)

andibraeu, 2015-09-22:
ganz ehrlich

glycoknob, 2015-10-08:
x86 ist noch nicht wirklich Unterstützt... UML ist auch noch nicht so richtig getestet.. würd das noch offen lassen

kaese77, 2015-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Danke Martin,

nur weil sich storchi x86 nicht zulegen will, heisst das noch lange nicht, dass es nicht benötigt wird.

Soll er doch weiter mit seinen Taschenrechnern arbeiten. Mädchen! ;-)

René

> Reopened #122 <https://github.com/weimarnetz/weimarnetz/issues/122>.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWFnciAAoJEJuS/5F7Le+B+zgIANccVHh8RJNuK7QJx7gmPCOj
EP0WNuL63hhQI+EMz8jym7D5l0PK37S/SEWQIjVI71oU3lVfnMrcdRKFyYXajZeM
NnUcBsuO5SH78Kmg3Gdel6esCblphbchmv73ljb208rGBoYw3v4ZEf6P7Na/AOf1
VpqLZkD6vUmG59VS593EZCS1sYdARluYBtM3HNerI4QkWWHalgTVcrGL2+iPqnuH
qdIdFIuhqIt+14K7DMo6+iSUiBpoIoXvIxp0Y5YqVfHCdnJZQfoFuY7EFwJmzm3f
Aa283Wp/afHuA+JtPVwiIULnvLSA7uv/Fu0ii6fBVQZQxA0UMU3zILcyUWqUR0c=
=pdZr
-----END PGP SIGNATURE-----

glycoknob, 2015-10-08:
Wenn ich mich richtig erinnere war die Idee ein möglichst komplettes Image zu bauen - mit allen WLAN-Treibern, müssten wir mal testen ob das auch klappt.

kaese77, 2015-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Platzprobleme habe ich nicht. Ich denke in UML wird auch mehr als 4MB Platz sein. Also von mir aus gern alles rein.

René

> Wenn ich mich richtig erinnere war die Idee ein möglichst komplettes Image zu bauen - mit allen WLAN-Treibern, müssten wir mal testen ob
> das auch klappt.

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWFnsKAAoJEJuS/5F7Le+B6c4IALXeUemUB5b4naPy+kh7hcG8
6cEHO4bl2pi7EVaoCbBP9kySx9IJZ6/Q9/5jK2/7GlejHNnir6TiIY5ODvB9WQip
cQki7SmyJ1aCFuy665p8n0fgZhKC3YykA55R72kc95RoGQTHfzgrFLCbJaqAzvXl
v08rKvGRn8e181Q4M4SuhR3gK1XiyLijlrbSUWpm28IWMdrGYifZBGZSHDO1YJxo
guydaFJ3carcxgblWwD51lkzyJnI4HjP55HP+ncKglHDJKnqHibO1Wd4nuGVblaq
sg4BvhV8CFP85kLZmhl8pZYa+XF/FCrURV6qH5zqiHL6IfNKvSvHOcAr74EFur4=
=v3OB
-----END PGP SIGNATURE-----

andibraeu, 2015-10-08:
ich dachte, das ging schon...

storchi, 2015-10-08:
X86 unterstützung ist fakultativ. Wenn das wer will soll ers machen und nicht die wenigen leute von den wichtigen sachen abhalten. 
Ich fordere Entschädigung!

kaese77, 2015-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

> X86 unterstützung ist fakultativ. Wenn das wer will soll ers machen und nicht die wenigen leute von den wichtigen sachen abhalten. Ich
> fordere Entschädigung!

Du hast doch jar nischt jemacht! Aber ´n Bier kannste gerne ham!

René
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWFootAAoJEJuS/5F7Le+BmjAH/0EOlIlJTU+g2SLDZt2ATqxI
ZZKXunf5gip9j6l6kkG8Q4tgL6nv5k64r5XgCvB18GxRlMQytMXtw6tImoWO6lHG
BC8drUcsWFs54cJ4VIIYbrACKSfuBOsyfSAjErgWTlhXxOp/BJ9C1e+jSkjKadiP
l8cHkuO2tFurZUAVOEI7LCpgm6oshPCE9NrEnBMHPRRP6bSdJ7C9P6AfeyAr6P2V
VdMRqd7OX5IOPtJxnC26dGxWTgBfS33eVTI1OfooBDJ7ds4yfSgpmeeYFGimCMdC
zOj0tOGtMStNVhuxdmXkbG+sX/6kyIGxXZ6QR2nz//TojroZxvjwpqukLqPj8iI=
=caHt
-----END PGP SIGNATURE-----

storchi, 2015-10-08:
Doch. Ich musste zuhören und mich aufregen und hier schreiben. Vom armen tippi mal abgesehn...

kaese77, 2015-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Zum Zuhören hat Dich keiner gezwungen. Und zu tun haste eh nix, sonst könntest Du hier nicht den ganzen Tag schreiben.

Auch Du wirst eines Tages von x86 profitieren. Du weisst es nur noch nicht! ;-)

Dein Gejammer ist rein hypothetisch. Geh doch zu Pegida, da kannste jetzt schon Angst vor Dingen haben, die vllt. eintreten werden. Du
Nörglerererer.

> Doch. Ich musste zuhören und mich aufregen und hier schreiben. Vom armen tippi mal abgesehn...

VG, René
-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWFpncAAoJEJuS/5F7Le+BvIMIAKPjqIbHSVQlEfTPcV1oA9f/
vEWBhc3Gxb0Owfd4WMCN/HGaASqygBTTxViW06kI7zUaO2t/HPNn859d3A+6Qu0Q
yGEJLaDtrncHJ2SdVwQx5jKBlvgI2iawbW4mkSU8xm5E4gZ9G77VzSmu+LnUWndI
KMgzy8eS5pRMG1xfG2zgred04SOTtuKudjmJvQvZsa83XuZvFO6zrQTvR2lgpGZS
boHRe5ikoivRxR54wSakfsSv6SCbbBDfhqq8dei3yEMFhJ8npLHGBbFUI0tIyL5n
0RHQAXcsZhT6du4X0/GakmJMkiSvXXf+WD5/Tz5kMHrePV114bxWGvae3mMRhls=
=5smf
-----END PGP SIGNATURE-----

glycoknob, 2017-01-19:
Klappt jetzt - siehe https://gitlab.bau-ha.us/weimarnetz/firmware/

closed 2017-01-19 by glycoknob