weimarnetz/weimarnetz #123: test new random function on heart and kidneys

andibraeu, 2015-09-22:


bittorf, 2015-09-23:
# viele integer von 0...1 erzeugen und nachschauen wie oft was vorkommt:
root@box:~ I=10000; while let I-=1; do _math random_integer 0 1; done >/tmp/RANDOM
root@box:~ grep -c 0 /tmp/RANDOM
5055
root@box:~ grep -c 1 /tmp/RANDOM
4944

# wieviele integer erzeugen von 0...500 und schauen wir gut das so verteilt ist;
bastian@X301:~$ I=100000; while let I-=1; do _math random_integer 0 500; done >/tmp/RANDOM^C
bastian@X301:~$ wc -l /tmp/RANDOM
99999 /tmp/RANDOM
bastian@X301:~$ grep -c ^0$ /tmp/RANDOM
203
bastian@X301:~$ grep -c ^1$ /tmp/RANDOM
234
bastian@X301:~$ grep -c ^2$ /tmp/RANDOM
203
bastian@X301:~$ grep -c ^3$ /tmp/RANDOM
197
bastian@X301:~$ grep -c ^256$ /tmp/RANDOM
193
bastian@X301:~$ grep -c ^500$ /tmp/RANDOM
189

# sieht in meinen augen gut aus.

glycoknob, 2015-09-23:
Yo, die alte war doch auch okay? Nur `_list` shuffle war ein wenig unfair (was vermutlich am Algorithmus gelegen hat). ...letzlich ist es für paar VPN-Server auch egal.

glycoknob, 2015-09-23:
alles gut. 

```
root@MartinsPlastebomber:~ list=" a b c d"
root@MartinsPlastebomber:~ rm /tmp/list 
root@MartinsPlastebomber:~ for i in $(seq 1 1000); do _list random_element "$list" >> /tmp/list; done 
 root@MartinsPlastebomber:~  sort /tmp/list | uniq -c
    225 a
    269 b
    240 c
    266 d
root@MartinsPlastebomber:~ 
```

closed 2015-09-23 by glycoknob