weimarnetz/weimarnetz #125: Chaos Calmer Release Party planen

andibraeu, 2015-09-22:


andibraeu, 2015-09-22:
@storchi Schnaps koofn

1 1/2 oz Plymouth® gin
1/4 oz triple sec
1 1/2 oz orange juice
3/4 oz lime juice
1 tsp grenadine syrup

bittorf, 2015-09-23:
29. September oder 6. Oktober?

andibraeu, 2015-09-23:
wenn die firmware fertig ist

andibraeu, 2015-12-07:
5.1.2016 oder 12.1.2016?

storchi, 2015-12-07:
Am 07.12.2015 um 12:20 schrieb Andi Bräu:
> 5.1.2016 oder 12.1.2016?

ich kann immer

storchi, 2015-12-07:
Am 22.09.2015 um 23:22 schrieb Andi Bräu:
> @storchi <https://github.com/storchi> Schnaps koofn
>
> 1 1/2 oz Plymouth® gin
> 1/4 oz triple sec
> 1 1/2 oz orange juice
> 3/4 oz lime juice
> 1 tsp grenadine syrup

check

bittorf, 2015-12-07:
ich schlage Dienstag den 12. Januar vor...

andibraeu, 2015-12-23:
* S140 gebucht
* Grill gebucht
* Thomas macht c64-Musik
* Barschichten
* Einladungen verschickt

andibraeu, 2015-12-23:
TODOs:

* Getränke für Cocktail im Cafe bestellen (Klaus)
* Grillgut kaufen
* Spendenbox

andibraeu, 2016-01-17:
die is fertig. Abrechnung muss noch gemacht werden

closed 2016-01-17 by andibraeu

glycoknob, 2016-01-17:
War Super! Danke euch nochmal für die Orga! Ich war ja nur zum trinken da ;)