weimarnetz/weimarnetz #126: VPN Tunnel lagt gewaltig - aber nicht immer

storchi, 2015-10-08:
Beobachtung: bein browsen immer mal wieder (ca. jeder 20. Seitenaufruf) wartet man bis zu 30s bis die Seite da ist. Sieht mit chrome Entwicklertools so aus:
[image]
initial connection dauert... da ist quasi tcp Verbindungsaufbau drin und ssl

Mein Setup: cc firmware aus dem meshkit / VPN3 Server / schnelles inet per Kabel
Fragen: wer hat das auch? mit welchem setup?

Bisher probiert: auf dem Router die qdisc von codel auf fifo (tap0) -> keine Änderung

cs

glycoknob, 2015-10-08:
Problem hab ich auch gesehen - ich war direkt mit Kabel dran, aber diese längeren Hänger waren definitiv da - könntest mal probieren mit `mtr 8.8.8.8` zu schauen wo Packet-Loss ist... da war immer 20-30% Loss bei in-berlin - wo genau der herkommt und das wirklich an in-berlin liegt weiß ich nicht.

glycoknob, 2015-10-08:
Bzw. war immer die Verbindung komplett weg für einige Sekunden, hat man gut mit mtr gesehen.

glycoknob, 2015-10-08:
@andibraeu hatte am Server etwas verändert - dann lief es erstmal wieder. Vielleicht tritt das Problem wieder auf?

bittorf, 2015-10-08:
genau das ist doch das TCP in TCP problem. Bisher hat mir ja immer keiner geglaubt...

kaese77, 2015-10-08:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

> genau das ist doch das TCP in TCP problem. Bisher hat mir ja immer keiner geglaubt...

Isch hab Dir geglaubt. Ist bei openVPN ja das gleiche Problem, weshalb auch überall in den Dokus steht ein UDP-Tunnel für TCP-Traffic und ein
TCP-Tunnel für UDP-Traffic.

ACK

René

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWFln7AAoJEJuS/5F7Le+BZy0IAJR/R1IYceq3o1fYptk+23Ul
99zJSkjTnNNTq+2Z/yNlRlUCsewfvqI+3gKXyBncoufrAAk7v9QXq9gscg2jkhUu
GGcHcxXhFazQtlZs8o66FZbw5W5EDR5y4BiSBNitBuojzjrz2CyU8rdusl9oDCmB
1My/MA7Fhp6ojZc9XAQREn2K/94l5/5Qk8InFMjS+ohozLuFxvdpSl0LWqXskDAf
I5E5YeEXQqMm1X1KcxXhEEiPiYh1ia8kbAN0PvIWF7Bu3PRvlBs4aWw6fMTIYUc3
OKG+xwtcq7XCYG66TA5Iyb5WU3tqvbWR50pzC9LN45SRNRpL7N62ukpbqZ3D/Hg=
=afN5
-----END PGP SIGNATURE-----

storchi, 2015-10-08:
> -----BEGIN PGP SIGNED MESSAGE-----
> Hash: SHA256
>
>  > genau das ist doch das TCP in TCP problem. Bisher hat mir ja immer
> keiner geglaubt...
>  >
>
>
> Isch hab Dir geglaubt. Ist bei openVPN ja das gleiche Problem, weshalb
> auch überall in den Dokus steht ein UDP-Tunnel für TCP-Traffic und ein
> TCP-Tunnel für UDP-Traffic.

Das ging wahrscheinlich in einer Auto Update Diskussion unter
cs

andibraeu, 2015-10-08:
nein, es ist nicht das TCP in TCP-Problem. Auf dem Server vpn3 geht http://localhost:2006 nicht, das txtinfo-plugin liefert nichts

das führt dazu, dass kalua irgendwelche dateien nicht anlegen kann

das führt dazu, dass kalua freidreht und jede minute olsr neu startet, obwohl eigentlich alles läuft

andibraeu, 2015-10-08:
hab jetzt das txtinfo-plugin explizit konfiguriert mit listen und accept auf "0.0.0.0" und jetzt scheint es zu gehen

@bittorf das solltest du vielleicht in dein serverscript mit aufnehmen

storchi, 2015-10-08:
mtr - das ist ja genau was man braucht. hat sich das aufstehen wieder gelohnt

cc / vpn3 / tcp in tcp tunnel
[image]

ich weiss ja schon wo die reise hingeht

andibraeu, 2015-10-08:
nach Berlin? :-)

Hop3 in deinem Bild ist nicht in unserer Hand. Und die Berliner haben mir gesagt, dass es da momentan Probleme bei In-Berlin gibt, weil die so viel Traffic von FF-Comunities aus ganz D haben.

probier mal vpn2 aus...

andibraeu, 2015-10-08:
Ich würd mal sagen, das liegt nicht an unserem vpn... so sieht mtr direkt vom server aus

```
vpn3 (0.0.0.0)                                        Thu Oct  8 14:33:07 2015
Keys:  Help   Display mode   Restart statistics   Order of fields   quit      
                                      Packets               Pings             
 Host                               Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. bgp01.berlin.freifunk.net       63.1%   206    0.4   0.4   0.3   4.1   0.4
 2. vlan935.nautilus.in-berlin.de    1.5%   206    0.6   1.5   0.3  39.2   3.7
 3. vlan84.octalus.in-berlin.de      2.9%   206    0.8   2.0   0.5  45.6   4.7
 4. google.ber.ecix.net              4.9%   206    1.1   2.0   0.9  18.7   2.0
 5. 72.14.239.111                    2.9%   205    1.6   2.2   1.4  16.5   1.7
 6. google-public-dns-a.google.com   4.9%   205  152.5   2.8   1.1 152.5  11.1
```

andibraeu, 2015-10-08:
Zum Vergleich auf vpn2:

```
vpn2 (0.0.0.0)                                           Thu Oct  8 14:36:54 2015
Keys:  Help   Display mode   Restart statistics   Order of fields   quit
                                         Packets               Pings
 Host                                  Loss%   Snt   Last   Avg  Best  Wrst StDev
 1. 192.168.122.1                       0.0%    32    0.3   0.2   0.1   0.5   0.1
 2. static.1.46.9.176.clients.your-ser  0.0%    32    1.2   1.8   0.9   4.5   1.2
 3. hos-tr1-juniper1.rz15.hetzner.de    0.0%    32    0.5   1.2   0.3  16.3   2.8
 4. core21.hetzner.de                   0.0%    32    1.8   1.0   0.4   4.9   0.9
 5. core1.hetzner.de                    0.0%    31    5.3   5.9   5.1  13.5   1.9
 6. juniper1.ffm.hetzner.de             0.0%    31    6.8   5.6   5.1   9.1   0.9
 7. de-cix20.net.google.com             0.0%    31    5.5  11.1   5.5  40.1   8.4
 8. 72.14.238.225                       0.0%    31    5.8  12.7   5.6  88.0  19.2
 9. 72.14.238.203                       0.0%    31    6.4   6.7   6.0   9.8   1.0
10. google-public-dns-a.google.com      0.0%    31    6.6  14.3   5.6  45.3  11.0
```

storchi, 2015-10-08:
und schon ist die auswahl des "besten" vpn servers wichtig...

storchi, 2015-10-08:
Am 08.10.2015 um 13:20 schrieb Bastian Bittorf:
> genau das ist doch das TCP in TCP problem. Bisher hat mir ja immer
> keiner geglaubt...

so. hab nachgedacht.

mein testrouter macht überhaupt kein "tcp in tcp" (kein ppoe oder irgend 
sowas), also kann das auch nicht der grund fuer die lags sein.

die lags treten aber auch bei nur einem tcp tunnel auf, benutze ich 
einen udp tunnel sind die lags weg

cs

storchi, 2015-10-08:
warum tcp in tcp doof ist:
http://sites.inka.de/bigred/devel/tcp-tcp.html
zusammengefasst: alles schick bis es zu packetloss kommt und/oder einer der beiden tunnel in die sättigung geht.

glycoknob, 2015-10-08:
@storchi: Yup - wollte mal auf vpn6 und vpn7 (work in progress) UDP testen, es hat dann aber nicht geklappt sich überhaupt zu verbinden, können wir vielleicht nächsten Dienstag mal debuggen. Ansonsten gibt es wohl mittlerweile auch in TCP ein paar Features, die das Problem nicht so groß werden lassen: Selective Ack (SACK) und wohl noch was anderes... hab ich zumindest gelesen - die Timeouts könnten durch den minütlichen OLSR Neustart verursacht worden sein - TCP in TCP sollte auch bei schlechter Verbindung nicht zu 20s Timeouts führen /gefährliches Halbwissen

bittorf, 2015-10-08:
@andibraeu das txtinfo-plugin ist schon (immer?) im cronscript drinne:
https://github.com/bittorf/kalua/blob/master/openwrt-addons/etc/kalua/cron#L47

wie auch immer: man siehts ja im syslog und im monitoring das OLSR neugestartet wird...

andibraeu, 2015-10-08:
@bittorf das stimmt, allerdings ist das nicht konfiguriert, auf dem server funktionierte das unter olsr 0.9.0.3 erst, als ich explizit listen und accept konfiguriert habe. Selbiges für jsoninfo: http://3.v.weimarnetz.de:9090/plugins

bittorf, 2015-10-08:
@andibraeu verstehe ich nicht. das cron-script _erzeugt_ doch die config fuer olsr wo das drinsteht und olsr wird gebaut mit 'make build_all' und da gibts dann alle plugins. jsoninfo nehme ich mal mit rein...

andibraeu, 2015-10-08:
ja, ist ja alles richtig.

die optionen sollten am ende so aussehen:

```
LoadPlugin "olsrd_txtinfo.so.0.1"                                           │                $ch = curl_init();
{                                                                           │                $timeout = 5; // 0 wenn kein Timeout
        PlParam "accept" "0.0.0.0"                                          │                curl_setopt($ch, CURLOPT_URL, "http://localhost:9090/links");
        PlParam "listen" "0.0.0.0"                                          │                curl_setopt($ch, CURLOPT_RETURNTRANSFER, 1);
        PlParam "port" "2006"                                               │                curl_setopt($ch, CURLOPT_CONNECTTIMEOUT, $timeout);
}
```

oder je nachdem kann man es auch auf 127.0.0.1 einschränken

bittorf, 2015-10-09:
@andibraeu ok, habs drinne - obwohl das doch sowieso die standard-werte sind, also eigentlich reicht ein: https://github.com/bittorf/kalua/commit/2acf0d9e2e09a866062e21d6a4aa3fd0e4488cbe
```
LoadPlugin "olsrd_txtinfo.so.0.1"
{
}

LoadPlugin "olsrd_jsoninfo.so.0.1"
{
}
```

andibraeu, 2015-10-16:
#129 macht vtun mit udp.. muss nur am server eingestellt sein

bittorf, 2015-10-16:
ich habs mal vereinigt, wir beobachten ein paar tage und testen und werden sehen...danke!

storchi, 2015-10-16:
schön dass es jetzt mit udp geht. TESTENTESTENTESTEN
mit nat kann es allerdings nichts zu tun haben, da von 122 zu vpn3 kein nat da ist.

storchi, 2015-10-21:
> cs@TP-T420s:~$ curl http://speedtest.netcologne.de/test_1gb.bin > /dev/null
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
 11 1024M   11  113M    0     0  2330k      0  0:07:29  0:00:49  0:06:40 3021k^C

am arbeitsplatz netz per kabel am 3600er über vpn3 udp

andibraeu, 2015-12-01:
ist gelöst mit vpn per udp

closed 2015-12-01 by andibraeu