weimarnetz/weimarnetz #128: cc / wlan client kommt nicht nach 10.63.x.x

storchi, 2015-10-13:
der router selbst kommt dorthin
...

storchi, 2015-10-13:
komisch: ich hab auf dem wlan client eine hostroute zu 10.63.122.33
ping ich auf 10.63.59.1 dann bekomme ich per tcpdump: 
10:29:44.809304 ARP, Request who-has 10.63.122.33 tell 10.63.122.17, length 28
???

glycoknob, 2015-10-13:
Interessant - kannst du sagen welche Firmware? Die aktuelle aus dem git/ja.is.halt.so Meshkit?

storchi, 2015-10-13:
jup. liegt wahrscheinlich and dhcp option 16: list dhcp_option '121,10.63.0.0/16,10.63.122.33'
was soll das sein?

storchi, 2015-10-13:
ist so, nehme ich "16,10.63.122.33" raus, gehts

glycoknob, 2015-10-13:
Bei mir funktionierts (zumindest für direkte Nachbarn) - die Option haben wir auch noch drinnen: https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/apply_profile.code#L1169 

Die Idee war mal eine separate Route zu haben um noch auf Weimarnetz-Knoten zu kommen, wenn es einen anderen Default-Gateway gibt, der nicht der Router ist. Diese Route wird per DHCP mitgegeben. Die IP ist immer die LAN-IP - hier scheint also irgendwas kaputt zu sein, wenn man darüber nicht zu anderen Knoten kommt. Könnte natürlich sein, dass da - wenn es WLAN-Clients sind - die LAN-IP nicht funktioniert aus irgendwelchen Gründen. 

Workaround wäre die Zeile aus `/etc/config/dhcp` rauszunehmen - aber das klingt so, als ob da mehr kaputt wäre.

glycoknob, 2015-10-13:
(das funktioniert hab ich nur mit Android getestet, vielleicht ignoriert der die Route auch einfach)

storchi, 2015-10-13:
> "Die Idee war mal eine separate Route zu haben um noch auf Weimarnetz-Knoten zu kommen, wenn es einen anderen Default-Gateway gibt, der nicht der Router ist"
unter der voraussetzung man bekommt noch per dhcp vom wn router ne ip? was solln das fuer ein szenario sein?
halte ich fuer sonderdroesel und kann von mir aus raus
cs

glycoknob, 2015-10-13:
Huch, bei mir crasht gleich der Router... und grad komm ich nichtmal ins VPN... ich update mal auf die Meshkit-Version und teste später.

bittorf, 2015-10-13:
das war irgendwie noetig wenn man selbst internet anbietet:
```
commit e2ce8f06b9139caaab60ada01a82107264639ed1
Author: L337hium <eddy@sdf.org>
Date:   Tue Feb 25 23:34:51 2014 +0100

    Add Classless Static Route Option for DHCP
    
    See: https://tools.ietf.org/html/rfc3442
    
    For now a router sends a static route in the 10.63.0.0/16 network to the client.
    
    This helps to reach the weimarnetz-network even if a user/client has two or more network interfaces running.
```

glycoknob, 2015-10-13:
> unter der voraussetzung man bekommt noch per dhcp vom wn router ne ip? was solln das fuer ein szenario sein?

Idee war, wenn man schon Internet am Rechner hat und mit dem Router auf's Weimarnetz will diese Route hat... 

> halte ich fuer sonderdroesel und kann/soll/muss von mir aus raus

Von mir aus kann das auch raus, macht viele Probleme und wenn man will, kann man die Route ja auch von Hand setzen. 

Trotzdem sollte es auch mit der Route funktionieren - frage ist warum das nicht klappt.

storchi, 2015-10-13:
> To accomplish this, we use DHCP option 121 (which replaced option 33) and allows you to use the netmask:
dhcp-option=121,x.x.x.x/yy,z.z.z.z
Where x.x.x.x is the destination LAN, yy is the CIDR notation (usually /24), and z.z.z.z is the host which will do the routing.

z.z.z.z ist aber per arp nicht auflösbar und damit nutzlos

bittorf, 2015-10-13:
@storchi zeig mal bitte die routen auf deinem rechner, und 'ip neigh' und ein traceroute ins 10.63.x.x netz

storchi, 2015-10-13:
z.z.z.z oist 10.63.122.33

storchi, 2015-10-13:
cs@TP-T420s:~$ ip r
default via 10.63.122.2 dev wlan0  proto static 
10.63.0.0/16 via 10.63.122.33 dev wlan0  proto static 
10.63.122.0/27 dev wlan0  proto kernel  scope link  src 10.63.122.17  metric 9 
10.63.122.33 dev wlan0  proto static  scope link 
192.168.122.0/24 dev virbr0  proto kernel  scope link  src 192.168.122.1

storchi, 2015-10-13:
cs@TP-T420s:~$ ip neigh
10.63.122.2 dev wlan0 lladdr 66:66:b3:4f:fa:95 REACHABLE

storchi, 2015-10-13:
traceroute to 10.63.59.1 (10.63.59.1), 30 hops max, 60 byte packets
 1  wardriver-wifi16-n122.olsr (10.63.122.17)  2999.333 ms !H  2999.231 ms !H  2999.209 ms !H
cs@TP-T420s:~$

bittorf, 2015-10-13:
interessant. das probiere mal die gateway-adresse (z.z.z.z) auch 10.63.122.2 zu setzen (und nicht .33)

storchi, 2015-10-13:
dann gehts natürlich. 
und schon sind wir wieder bei sogenannten nützlichen features. braucht keine sau und beschäftigt horden von leuten die was sinnvolles tun könnten
cs

bittorf, 2015-10-13:
es bedeutet also, das abhaengig vom interface und hybrid oder normal ne andere IP rausgegeben werden muss. ich denke drueber nach.

glycoknob, 2015-10-13:
Ok, dann raus damit - wieso hat das überhaupt jemals so funktioniert? Das würde für alle die im WLAN sind, den Zugriff auf den Router kaputt machen!? Alternative wäre Ziel für WLAN auf die .2 zu setzen. Ich würd es aber ganz rauswerfen.

Unter meinem Linux hat es übrigens funktioniert weil sich die Route garnicht hinzufügen lässt: 

    [~]$ sudo ip route add 10.63.0.0/16 dev wlan0 via 10.63.188.33                                       
    RTNETLINK answers: Network is unreachable

    [~]$ arping -I wlan0 10.63.188.2                                                                     
    ARPING 10.63.188.2 from 10.63.188.14 wlan0
    Unicast reply from 10.63.188.2 [12:FE:ED:2B:91:30]  2.119ms
    Unicast reply from 10.63.188.2 [12:FE:ED:2B:91:30]  13.063ms
    Sent 2 probes (1 broadcast(s))
    Received 2 response(s)
    [~]$ arping -I wlan0 10.63.188.33                                                                    
    ARPING 10.63.188.33 from 10.63.188.14 wlan0
   Sent 5 probes (5 broadcast(s))
   Received 0 response(s)
  

Fieser Bug. Danke @storchi fürs melden!

bittorf, 2015-10-13:
ich glaube das ging deswegen, weil es fuers LAN konzipiert war. ich halte das feature wir nicht so doof, aber wegen mir kicks raus...

glycoknob, 2015-10-13:
@bittorf: Wenn der dnsmasq es erlaubt dann kann es ja für LAN drinbleiben - ich finds auch sinnvoll, aber es hat schon oft Probleme gemacht - jetzt wissen wir ja warum.

closed 2015-10-13 by bittorf

andibraeu, 2015-10-13:
...und schon wieder ohne pull request...

bittorf, 2015-10-14:
ja, ohne pullrequest habe ich auch jedes mal einen Stich in der Brust - ich wollte es aber noch fix vom Tisch haben bevor sich wieder ewig keiner drum kuemmert. der konsens war ja da...

commit ac997f22 (2015-10-13): build: apply_profile: disable DHCP-option 121 for now this we fix the issues. closes #128