weimarnetz/weimarnetz #130: wlan client im adhoc...

storchi, 2015-10-19:
> cs@TP-T420s:~$ ip r
default via 10.63.122.1 dev wlan0  proto static 
10.63.0.0/16 via 10.63.122.2 dev wlan0  proto static 
10.63.122.0/27 dev wlan0  proto kernel  scope link  src 10.63.122.17  metric 9 
192.168.122.0/24 dev virbr0  proto kernel  scope link  src 192.168.122.1 
cs@TP-T420s:~$ 
cs@TP-T420s:~$ ping 10.63.122.1
PING 10.63.122.1 (10.63.122.1) 56(84) bytes of data.
^C
--- 10.63.122.1 ping statistics ---
7 packets transmitted, 0 received, 100% packet loss, time 5999ms
cs@TP-T420s:~$ 

kann das wer nachvollziehen?
cs

bittorf, 2015-10-19:
aber der ping auf die .122.2 geht?

storchi, 2015-10-19:
nein, weder 122.1 noch 122.2 noch 122.33 noch eine startseitenumleitung

storchi, 2015-10-19:
> cs@TP-T420s:~$ ip n
10.63.122.1 dev wlan0 lladdr 64:66:b3:4f:fa:95 REACHABLE
10.63.122.2 dev wlan0  FAILED
cs@TP-T420s:~$

glycoknob, 2015-10-19:
Hilft ein 

    ip route del 10.63.0.0/16 via 10.63.122.2 dev wlan0 proto static  

Auf dem Notebook? Dann ist es die schon entfernte Route via DHCP 

PS: Ich bin immernoch dafür das wir das IP-Schema mal fixen :)

storchi, 2015-10-19:
nein, bringt nix.
meine wlan karte schmeisst mir hier auch immer mal eine firmware error im adhoc netz. ich muss dann die wlan treiber neu laden oder der Rechner steht ganz.
Also evtl. ist das nur bei mir ein Problem.
cs

storchi, 2015-10-19:
wer kann im adhoc netzwerk irgendwas anstellen?

bittorf, 2015-10-19:
also meiner meinung nach bekommt man im adhoc-netzwerk garnkeine IP, oder?

storchi, 2015-10-19:
doch, siehte doch

bittorf, 2015-10-19:
dann ist das wohl eher ein wlan-fehler auf laptop seite. ist das ein kernel <4.1? wenn ja, dann mal hochziehen. atheros karte oder sogar intel gurke?

storchi, 2015-10-20:
laptop ist auszuschliessen, gleiches verhalten an einem anderen windows laptop.
auf dem router: 
> root@cs-test:~ ip r g 10.63.122.29
10.63.122.29 dev wlan0-1  src 10.63.122.2 

obwohl die 29 an wlan1 hängt.
hab schonmal die ip von wlan1 auf 1 und 2 gesetzt - bringt aber das gleiche Ergebnis.
ich behaupte mit diesem verfrickelten ip schema ist das nicht zu machen.
also: dhcp auf adhoc wlan aus? geht das überhaupt? dann würde ich das adhoc wlan aber auch auf unsichtbar stellen, damit sich da keiner sinnlos zu verbinden versucht.
cs

bittorf, 2015-10-20:
auf jeden fall dhcp auf adhoc aus. ich dachte das waere schon laengst

storchi, 2015-10-20:
und wie geht das?
und dann sehn die laptops ein weimarnetz adhoc und wolln sich verbinden und nix geht -> wn kann nix
also adhoc wlan unsichtbar...
cs

kaese77, 2015-10-20:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

> und wie geht das? und dann sehn die laptops ein weimarnetz adhoc und wolln sich verbinden und nix geht -> wn kann nix also adhoc wlan
> unsichtbar... cs

Sollte das nicht eh "ffintern" heißen?

Und ein Laptop verbindet sich doch nicht selbst mit einem adhoc-Netzwerk!?

René

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWJgpHAAoJEJuS/5F7Le+BvSIH/1eKb0TPGaXsWwaM4UgYJYIH
eEjp+Vpu9SznMNa2XcVBuNSlOtFIhlmIzOsIO85jy2m9NeKVnsFwKWX48KbR/jQJ
Izx8HGDP45fQhZTy3cUiqgl7m73s37Q3e+nRKv23ogTf6eAjIMt1CdupmA9hM3iL
koLozxRnz+ZHTU7G7wY/zq6CCzYwGsNhjfzEC5ts9bP1alUVuvMWc/pRurlNcq00
s2vQZpfRuU5M/WNpCFtwimWMzEogUryg80ivbvTcU7wJ4bVwhFDKvHQlFaFKKoOf
0HnvTyhfRsPfNJ9q4IAhCzdAgcDWbUR+j1LSO8lu+Zuuf/Z2rs2J0x8BmDfuz2A=
=hc9g
-----END PGP SIGNATURE-----

andibraeu, 2015-10-20:
unsichtbar müsste doch ausreichen, bin mir unsicher, ob man dnsmasq so einstellen kann, dass er nur auf das ap-device hört.

andibraeu, 2015-10-20:
damals™ als wir noch nüscht hatten, waren wir froh, dass es adhoc gab

glycoknob, 2015-10-20:
Hi, unsichtbar klappt gut - hab das bei mir laufen `option hidden 1` fur
das adhoc dev

Adhoc IP ist im Subnetz vom AP, DHCP macht ja broadcast, keine Ahnung was
da passiert.

storchi, 2015-10-20:
> Adhoc IP ist im Subnetz vom AP, DHCP macht ja broadcast, keine Ahnung was
da passiert.

die 1 anfrage ist broadcast, die antwort schon unicast, die erneuerung auch unicast.
hab schon gesehn, dass broadcast anfragen über das eine und unicast antworten über das andere(falsche) if rausgingen

storchi, 2015-10-20:
hab jetzt:
> config wifi-iface
	option device 'radio0'
	option network 'wlanadhoc'
	option mode 'adhoc'
	option bssid '02:ca:ff:ee:ba:be'
	option ssid 'weimarnetz'
	option mcast_rate '1000'
	option hidden '1'
config wifi-iface
	option device 'radio0'
	option network 'wlan'
	option mode 'ap'
	option ssid 'Weimarnetz - Knoten 122'
config wifi-device 'radio1'
	option type 'mac80211'
	option country 'US		# see hacked /usr/lib/crda/regulatory.bin'
	option channel '40'
	option frag 'off'
	option txpower '20'
	option beacon_int '1000'
	option path 'pci0000:00/0000:00:00.0'
	option hwmode '11a'
	option disabled '0'
config wifi-iface
	option device 'radio1'
	option network 'wlanRADIO1'
	option mode 'adhoc'
	option bssid '02:ca:ff:ee:00:40'
	option ssid 'weimarnetz'
        option hidden '1'
	option mcast_rate '6000'

aber sehe weimarnetz adhoc immernoch

bittorf, 2015-10-20:
schaut mal bitte genau mit "horst" und nicht mit dem network-manager.

bittorf, 2015-10-20:
habe mal intensiv probiert. adhoc-ssid-beaconing auszuschalten ist einfach nicht unterstuetzt scheint mir.
dann besser
```
ffintern.2GHz
ffintern.5GHz
```

glycoknob, 2015-10-20:
Grad mal getestet: Ich bekomme auch eine IP via DHCP über Adhoc und kann mich nicht verbinden. 

hidden=1 scheint tatsächlich nicht zu klappen.

storchi, 2015-10-20:
Oder und das beacon interval auf 65000 stellen

glycoknob, 2015-10-20:
```
[~]$ ip n                                                                                            
10.63.188.1 dev wlan0 lladdr 10:fe:ed:2b:91:30 REACHABLE
``` 

ping und alles andere klappt nicht.

bittorf, 2015-10-20:
das beacon-interval ist fuer beide interfaces zusammen, weil es bei der radio-section gesetzt wird.
betrofft dann also auch das AP-interface und da wuerde ich eigentlich nicht langsamer als 250 werden
```
wireless.radio1.beacon_int='250'
```

glycoknob, 2015-10-20:
Wenn ich die IP-Konfiguration so ändere: https://github.com/weimarnetz/weimarnetz/issues/36#issuecomment-75922262

Und DHCP für Adhoc explizit anknipse, dann klappt auch alles. 

```
Tue Oct 20 16:29:52 2015 daemon.info dnsmasq-dhcp[28019]: DHCP, IP range 10.63.188.52 -- 10.63.188.54, lease time 1h
Tue Oct 20 16:29:52 2015 daemon.info dnsmasq-dhcp[28019]: DHCP, IP range 10.63.188.4 -- 10.63.188.30, lease time 1h
Tue Oct 20 16:29:52 2015 daemon.info dnsmasq-dhcp[28019]: DHCP, IP range 10.63.188.36 -- 10.63.188.46, lease time 2h
``` 

Ich wäre dafür das IP-System zu fixen.

storchi, 2015-10-21:
zeig mal die dhcp conf

bittorf, 2015-10-21:
```
root@EG-labor-AP:~ uci show dhcp | grep -v ^'dhcp.@'
dhcp.lan=dhcp
dhcp.lan.interface='lan'
dhcp.lan.start='34'
dhcp.lan.limit='13'
dhcp.lan.leasetime='2h'
dhcp.lan.force='1'
dhcp.lan.ignore='0'
dhcp.wlan=dhcp
dhcp.wlan.force='1'
dhcp.wlan.ignore='0'
dhcp.wlan.interface='mastergate'
dhcp.wlan.dhcp_option='option:router,192.168.0.1,192.168.0.1'
dhcp.wlan.leasetime='12h'
dhcp.wlan.start='192.168.2.2'
dhcp.wlan.limit='253'
dhcp.wan=dhcp
dhcp.wan.interface='wan'
dhcp.wan.ignore='1'
```

erstmal nicht an den IP's stoeren. es gibt halt nur noch ein "mastergate" und da sind alle AP's drinne:

hier die wichtigen sagen aus /etc/config/network:
```
network.wlan=interface
network.wlan.proto='static'
network.wlan.ipaddr='10.63.2.2'
network.wlan.netmask='255.255.255.224'

network.wlanadhoc=interface
network.wlanadhoc.proto='static'
network.wlanadhoc.ipaddr='10.63.2.1'
network.wlanadhoc.netmask='255.0.0.0'

network.mastergate=interface
network.mastergate.type='bridge'
network.mastergate.proto='static'
network.mastergate.ipaddr='192.168.0.1'
network.mastergate.netmask='255.255.0.0'
network.mastergate.ifname='roaming1'

network.roaming1=device
network.roaming1.type='macvlan'
network.roaming1.name='roaming1'
network.roaming1.ifname='@wlan'
network.roaming1.macaddr='02:ff:ff:ff:00:00'
```

glycoknob, 2017-01-19:
- beacon_int ist jetzt wieder default nach den Problemen
- AdHoc SSID ist jetzt `ff.intern.<knotennr>.ch5` oder `ff.intern.<knotennummer>.ch40` 

- DHCP könnte für AdHoc laufen und kaputt sein. Müsste man nochmal testen. 
- Die Probleme mit den Clients kenn ich auch, der Intel Treiber spinnt, wenn er was mit AdHoc machen muss

Ich schließe erstmal.

closed 2017-01-19 by glycoknob