weimarnetz/weimarnetz #131: VPN mit UDP - probleme auf Server bei >8 Nachbarn

bittorf, 2015-10-21:
wenn viele Clients sich zu dem Server verbinden, geht es irgendwann nicht mehr.
Verbindung geht immer an/aus fuer die neuen tap-interfaces. Bisher nur auf http://4.v.weimarnetz.de/
beobachtet mit >8 Nachbarn. Laeuft erstmal wieder auf TCP, bis ich weiter testen kann.

storchi, 2015-10-21:
ist das so ein - ich hab nur ein paar ports - vm server?

glycoknob, 2015-10-21:
Das müsste egal sein, vtun nimmt immer den Port in der Konfiguration - wo das nicht geklappt hat, war es ein Bug.

storchi, 2015-10-21:
anderen server testen mit >8 verbindungen...

bittorf, 2015-10-21:
@storchi nein, das ist ein "echter" server mit allen ports frei, 2gig ram, 50gig HDD - root-login wie immer

kaese77, 2015-10-21:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

> @storchi <https://github.com/storchi> nein, das ist ein "echter" server mit allen ports frei, 2gig ram, 50gig HDD - root-login wie immer

2GB RAM? Kannst Du mal auf die RAM-Auslastung schauen? Wie gross ist die SWAP-Partition?

René

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWJ/nwAAoJEJuS/5F7Le+BmfIH/1fK5o/6wf3+XuMVeEKXiX5e
EwY3lJ4mxhxQgqPII3h0pmocZqYJw1HDfWDUU3FHetgSf0S9Y/kDaTS1CeRzPbNq
Dvzfth9osdtSbA19Gvd7nqGWXMamgE198XdCbk5+mPlTVvDCbJBNf3PKgIsa2vfr
Cph9B8YYkKbAqZ3oCMHHWcrHcLVY2pamGiiar2TWqT+u2ReIqp+2dmRmyUE6IpOt
fz1SP2YQfbOvePB9LQehTbl8HyUvQyQJxe/UCGzhDFo4f8m7Z1DHtN4KCp0gO9on
0dcB15eSe6gIxQqYLMebXto+fnnKyIAox2a3gUkYjjQGySxL8+ozAP9Z3kgvy+A=
=ID9w
-----END PGP SIGNATURE-----

bittorf, 2015-10-22:
@kaese77 ram-Auslastung ist minimal, meist so um die 100 Megabyte.
```
root@server1:~ free
             total       used       free     shared    buffers     cached
Mem:       2097152      88312    2008840          0          0      56796
-/+ buffers/cache:      31516    2065636
Swap:            0          0          0
```

kaese77, 2015-10-22:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

vtund "stracen"?

> @kaese77 <https://github.com/kaese77> ram-Auslastung ist minimal, meist so um die 100 Megabyte.
> 
> |root@server1:~ free total used free shared buffers cached Mem: 2097152 88312 2008840 0 0 56796 -/+ buffers/cache: 31516 2065636 Swap: 0 0
> 0 |

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWKIG7AAoJEJuS/5F7Le+BT6AH/jS9KRfey4w8ylWmxirjZBE9
9wG59m65XXhfNVewTDjlr63nIORv43ukjreDAKVK6yK/eQPc+KkR3GL+Ei/h0Dyd
HlUkx9j/MAY4g3Cb/KuH/MFRw+hNcCXkwGGLcKNd4UMbzvVMS8vc66xZP6w6kSHQ
R+r1YhGhRl5TGv8CgQjd9olZ+1/HsFQ4HDIuz4N6ptp5P5om/464MG8WRnzxPWVr
YMiACwUXnmr/jnGZtB50LbfL8U7ZTxLp01GXhwHPKAz/Z960RI/nZWspzLcamYWE
RWl5VrzXAuAMUySB3Pq8QnFFyZfnIMfjaLoxSnDWcJ+j0U84ZL8xRDRd/i+DG2M=
=nF49
-----END PGP SIGNATURE-----

storchi, 2015-10-22:
ich würde trotzdem mal beim anbieter fragen ob da was an der davor liegenden firewall sein könnte.
cs

bittorf, 2015-10-22:
@storchi : der support hat keine Ahnung, alles Inder. Ich probiere erstmal einen anderen Server.

andibraeu, 2015-11-17:
hast du hier schon mehr in erfahrung bringen können?

bittorf, 2015-11-17:
noch nicht, habe ne anderen server aufgesetzt, aber noch nicht umgeschaltet.
gibt mir noch ne woche zeit.

andibraeu, 2015-12-10:
heute sind auf vpn2 11 olsr-nachbarn verbunden, davon 9 router und 2 server

Es sieht bisher gut aus und läuft

glycoknob, 2015-12-10:
bekomme grad: 

```
Thu Dec 10 15:12:02 2015 daemon.info vtund[2810]: Session Node188[2.v.weimarnetz.de] opened
Thu Dec 10 15:12:02 2015 daemon.info vtund[2810]: UDP connection initialized
Thu Dec 10 15:12:02 2015 daemon.info vtund[2810]: Connection refused (146)
```

relatiert?

andibraeu, 2015-12-10:
was heißt relatiert?

so sieht das grad auf vpn2 aus:
```
Dec 10 15:45:04 vpn2 vtund[3801]: Session Node319[95.90.213.199:47404] opened
Dec 10 15:45:04 vpn2 vtund[3801]: Can't allocate tap device tap319. Device or resource busy(16)
Dec 10 15:45:04 vpn2 vtund[3801]: Session Node319 closed
Dec 10 15:45:10 vpn2 vtund[3970]: Session Node319[95.90.213.199:8782] opened
Dec 10 15:45:10 vpn2 vtund[3970]: Can't allocate tap device tap319. Device or resource busy(16)
Dec 10 15:45:10 vpn2 vtund[3970]: Session Node319 closed
Dec 10 15:45:13 vpn2 kalua: sh: vpn_innercity_master_iface() locked: up/tap170 must wait for 'action: down dev: tap319'
Dec 10 15:45:15 vpn2 vtund[4118]: Session Node319[95.90.213.199:50433] opened
Dec 10 15:45:15 vpn2 vtund[4118]: Can't allocate tap device tap319. Device or resource busy(16)
Dec 10 15:45:15 vpn2 vtund[4118]: Session Node319 closed
Dec 10 15:45:15 vpn2 kalua: sh: vpn_innercity_master_iface() locked: down/tap170 must wait for 'action: down dev: tap319'
Dec 10 15:45:16 vpn2 vtund[4162]: Session Node188[217.253.208.66:33092] opened
Dec 10 15:45:16 vpn2 vtund[4162]: Can't allocate tap device tap188. Device or resource busy(16)
Dec 10 15:45:16 vpn2 vtund[4162]: Session Node188 closed
Dec 10 15:45:18 vpn2 kalua: sh: vpn_innercity_master_iface() locked: down/tap188 must wait for 'action: down dev: tap319'
Dec 10 15:45:20 vpn2 kalua: sh: vpn_innercity_master_iface() locked: up/tap319 must wait for 'action: down dev: tap319'
Dec 10 15:45:21 vpn2 vtund[4297]: Session Node319[95.90.213.199:21793] opened
Dec 10 15:45:21 vpn2 vtund[4297]: Can't allocate tap device tap319. Device or resource busy(16)
Dec 10 15:45:21 vpn2 vtund[4297]: Session Node319 closed
Dec 10 15:45:21 vpn2 kalua: sh: vpn_innercity_master_iface() locked: down/tap319 must wait for 'action: down dev: tap319'
Dec 10 15:45:22 vpn2 kalua: sh: vpn_innercity_master_iface() locked: up/tap441 must wait for 'action: down dev: tap319'
Dec 10 15:45:26 vpn2 vtund[4489]: Session Node319[95.90.213.199:50431] opened
Dec 10 15:45:26 vpn2 vtund[4489]: Can't allocate tap device tap319. Device or resource busy(16)
Dec 10 15:45:26 vpn2 vtund[4489]: Session Node319 closed
```
Node 319 von @kaese77 macht da schon den ganzen Tag Ärger, deswegen kann sich die 188 auch nicht verbinden, weil das script immer auf das tap up/down event wartet

glycoknob, 2015-12-10:
Ich hätte noch einen vtun client anzubieten der sich auf einen anderen Port verbinden will :/ 
Okay.. müssen wir nochmal in Ruhe testen. 

[image]

glycoknob, 2015-12-10:
> Dec 10 15:45:18 vpn2 kalua: sh: vpn_innercity_master_iface() locked: down/tap188 must wait for 'action: down dev: tap319'

Das macht keinen Sinn und ist doch überhaupt nicht notwendig? Das ist doch alles unabängig voneinander?

andibraeu, 2015-12-10:
mit welcher Version testest du?

glycoknob, 2015-12-10:
3.0.3 vom vtun CVS - ich glaube OpenWRT hat 3.0.2 oder?

andibraeu, 2015-12-10:
ja, ist 3.0.2...

ich meinte eher, mit welcher Firmware du testest

glycoknob, 2015-12-10:
aktuelles CC mit aktuellen GebrannteMandeln auf dem Router, Wireshark ist vom Ubuntu mit selbstgebauten vTUN

glycoknob, 2015-12-10:
Hab nochmal in den Source geschaut (netlib.c) - ich sehe keinen Grund warum das nicht klappen sollte, einiges Problem wäre, wenn viele Router hinter einem NAT sich mit dem gleichen VPN verbinden wollen, dann könnte der gleiche Source-Port ein Problem sein. Aber soweit ich weiß wird der Zufällig gewählt.

bittorf, 2015-12-10:
bekommt vpn2 alle kalua-aktualisierungen? wenn nicht, dann bitte mal hochziehen:
```
echo 'system.@fwupdate[0].mode=testing' >/etc/kalua_uci
_firmware update_pmu
```
die statusseite scheint auch nicht verlinkt zu sein:
http://2.v.weimarnetz.de/

am besten mal so machen:
```
touch /var/www/cgi-bin-status.html && ln -s /var/www/cgi-bin-status.html /var/www/index.html
```

andibraeu, 2015-12-11:
nunja, ich bin ja nicht verrückt und geb den Server als testing-Instanz her...

bittorf, 2015-12-11:
@andibraeu du kannst auch manuell updaten.
ich wollte nur sichergehen, das die locking probleme nicht durch ein altes script kommen und vielleicht schon lange behoben sind...

andibraeu, 2015-12-23:
VPN2 hat heute 18 Nachbarn und läuft. Ich schließe das erstmal

closed 2015-12-23 by andibraeu