weimarnetz/weimarnetz #144: splashpage iphone6 <-> 37711

storchi, 2015-11-11:
ein Benutzer berichtet, dass die splashpage auf dem iphone6 nicht kommt.
kann das wer nachprüfen?
andere Endgeräte gehen.
cs

bittorf, 2015-11-11:
mit version stable = r36817 und r37890

storchi, 2015-11-12:
das sind ja total krumme versionen - wer macht denn sowas?
mit 37711 wäre das nicht passiert

bittorf, 2015-11-12:
schwer zu sagen...bin gerade unsicher wer damals die VHS geraete vorbereitet hat.

andibraeu, 2015-11-12:
mit iphone 6 und ios9 hab ich keine Probleme mit den Splashpages, getestet am Mon Ami, Musikschule, Maschinenraum, zu Hause

bittorf, 2015-11-13:
kann es sein, das es mit gestörtem VPN zusammenhing?

storchi, 2015-11-13:
laut wuschel gingen andere geräte an den betreffenden mon ami knoten.

storchi, 2015-11-13:
weitere aussage megadave: aktuellstes iphone 6plus mit AA-37711 geht!

bittorf, 2015-11-14:
arrrg. evtl. muessen wir es an einer iOS-version festzurren - ich frage mal wuschel.
es gibt halt im alten ath9k ein paar fiese bugs, die mit dem stromsparmodus / timing
von iPhones nicht klar kommen. ich befuerchte das es sowas ist (und das tritt dann
auch nicht immer auch...)

storchi, 2015-11-15:
laut wuschel tritt es immer auf. diese ominösen fiesen bugs können es auch nicht sein, da das iphone von wuschel wlan und ip mässig verbunden ist. meiner meinung nach, haben die doofen 6er iphones eh alle die gleiche ios version. 
lass es dir mal vorführen, basti.
cs

storchi, 2015-11-16:
so, ganze aufregung umsonst.
ein neustart vom iphone hat den "fehler" behoben.
mannmannmann

closed 2015-11-16 by storchi

kaese77, 2015-11-16:
-----BEGIN PGP SIGNED MESSAGE-----
Hash: SHA256

Da griegsch ´nen iFön!

> so, ganze aufregung umsonst. ein neustart vom iphone hat den "fehler" behoben. mannmannmann

-----BEGIN PGP SIGNATURE-----
Version: GnuPG v2

iQEcBAEBCAAGBQJWSclpAAoJEJuS/5F7Le+BMhMH/RwYranNNkEZu5Ez5wCrQN/H
ioqPeuWIXOkB+OKWKAXeYveTtSElNJieiZc7FSvprqwyyrnJFWBTOvQo0eSxqrHX
m4ZC0ZXWE7GwGzTZ/LzKF5TyofP+ykCHRDJuiOkbX2V2daUqlFURCPyvt28h8nEd
XB/PrJvGrWBJW5DmIWY/Y7tXV0vg3FC3gjaMJHtie8WeMAbjCpJLK0D0EqDCO+r5
QC29NtjQovfl6m3FSbpF0BeWx/Mtkh/mgHCDDo/tWdkZFvC0ItymwlnMNEyi/ulH
KbCHaGNCJ9WCBfv2jlPcWbwoFvDbNx+rLjxHitgXPpE/9Jirlx7gGWk8MPu9rXg=
=K3cY
-----END PGP SIGNATURE-----