weimarnetz/weimarnetz #146: CC: landing page bringt "Bad gateway"

storchi, 2015-11-11:
user werden trotzdem freigeschaltet
fehler kommt offenbar nicht immer
cs

bittorf, 2015-11-11:
ich glaube das haengt einfach mit der umleitung zusammen, die dann falsch im cache ist.
ist das wirklich mit CC15.05 ?
und warum bleibt das ueberhaupt im cache?

andibraeu, 2015-11-12:
ich hab das auch schon gesehen. passiert nur auf der neuen Firmware. Der Server liefert offenbar gar nichts. Muss ich nochmal analysieren: https://github.com/weimarnetz/openwrt-packages/blob/master/utils/luci-weimar-splash/luasrc/view/weimarnetz-splash/infopage.htm#L28

bittorf, 2015-11-12:
hmmm, das ist ja in 'lua'.
können wir einen test haben, der alle codepfade durchlaeuft und jeweils den output
durch tidy (von "http://www.html-tidy.org/") jagt?

es wuerde auch reichen wenn man alle code-pfade im browser nachstellen kann.

storchi, 2015-11-12:
@basti, wie meinst du das mit dem cache?

bittorf, 2015-11-12:
@storchi es ist eigenartig, das es ueberhaupt kommt und dann auch immer wieder
beim eingeben der internet-adresse. aber das internet geht / man ist freigeschaltet.
evtl. ist das auch ein nebeneffekt, das es im cache bleibt aufgrund dieses fehlers.

andibraeu, 2015-11-12:
ein Problem mit dem Cache habe ich noch nicht gesehen. Kannst du bitte genauer beschreiben, was passiert?

storchi, 2015-11-12:
Idee: evtl liegt es am Paketverlust bei In.Berlin?

bittorf, 2015-11-13:
@andibraeu die fehlerseite kommt immer und immer wieder, wenn man die ursprunglich internetadresse eingibt zu der man wollte. gibt man eine andere ein, gehts ganz normal. um das nochmal zu sagen, die infoseite geht - der fehler tritt erst auf, wenn man auf den knopf drueckt.

@storchi mit paketverlust haengt das bestimmt nicht zusammen. der fehler kam ja flott und ohne haengen/warten.

storchi, 2015-11-13:
das was du meinst basti ist ein caching problem deines browsers. das ist nicht das was ich meine (lösch mal den browsercache, dann kannste deine ursprungsseite wieder laden). Was ich meine ist eine "bad gateway" fehlerseite anstelle der landingpage (weimarnetz.de). dieser fehler tritt nur selten auf und auch nur bei CC, evtl. hat es was mit den vpn servern 2 und 3 zu tun?

bittorf, 2015-11-14:
ja, ich meine das gleiche. "bad gateway" kommt vom internen webserver, wenn (z.b.) man header falsch ausliefert. ich wollte nur sagen, das es nix mit VPN zu tun hat.

andibraeu, 2015-11-14:
das passiert doch nur nach dem Klick auf den Knopf? In dem Fall sollte ein redirect passieren (Zeile 33). Vielleicht wirft die redirect-Funktion einen Fehler, weil die URL falsch zusammengebaut ist. Komisch nur, dass es nicht immer passiert. Ich guck mir das an, wenn ich wieder zu Hause bin.

bittorf, 2015-11-16:
Die Zeile 33: schick mal ne Link dafuer, ich finde das nicht

glycoknob, 2015-11-16:
Heute auch auf dem Café-Router im s140 gesehen. 

Das steht im Logfile (vermutlich loggt luci aber nicht nach `/tmp/messages`): 
```
Nov 16 15:20:22 ' [up 34737s 4259840 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript s140_dhcp-old() [ERR] _weblogin login_check 4c:3c:16:d2:fa:8b/s140
Nov 16 15:20:27 ' [up 34743s 4317184 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() DHCP-mode, profile is ffweimar_hybrid - ignoring
Nov 16 15:20:28 ' [up 34743s 4321280 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() query local db for ''
Nov 16 15:20:28 ' [up 34743s 4325376 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() ANSWER: '' (1=ok) FREE: '' URL: 'locally_asked:?LOGIN=&MAC=4c%3a3c%3a16%3ad2%3afa%3a8b&SIGNAL=s140_dhcp-old&KEYWORD=&USER_LANG=de'
Nov 16 15:20:28 ' [up 34743s 4317184 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() DHCP-mode, profile is ffweimar_hybrid - ignoring
Nov 16 15:20:28 ' [up 34743s 4354048 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() query local db for ''
Nov 16 15:20:28 ' [up 34743s 4325376 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() ANSWER: '' (1=ok) FREE: '' URL: 'locally_asked:?LOGIN=&MAC=4c%3a3c%3a16%3ad2%3afa%3a8b&SIGNAL=s140_dhcp-old&KEYWORD=&USER_LANG=de'
Nov 16 15:20:28 ' [up 34743s 4308992 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript s140_dhcp-old() [ERR] _weblogin login_check 4c:3c:16:d2:fa:8b/s140
Nov 16 15:20:51 ' [up 34766s 4554752 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript dhcp_del() '90:68:c3:aa:b4:80' is from 'wlan0-1' = unknown-DEV
Nov 16 15:21:11 ' [up 34786s 4411392 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() DHCP-mode, profile is ffweimar_hybrid - ignoring
Nov 16 15:21:11 ' [up 34786s 4415488 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() query local db for ''
Nov 16 15:21:11 ' [up 34786s 4411392 bytes free] daemon.info fff+ /etc/dhcp-script.d/10dhcpscript weblogin_login_check() ANSWER: '' (1=ok) FREE: '' URL: 'locally_asked:?LOGIN=&MAC=20%3a64%3a32%3a02%3a82%3a29&SIGNAL=s140_dhcp-old&KEYWORD=&USER_LANG=de'
```

bittorf, 2015-11-16:
@glycoknob hat das was mit dem ticket zu tun?

glycoknob, 2015-11-16:
@bittorf: Also hab die "Bad Gateway" Meldung beim Splash freischalten auf dem Router gesehen, ob die Logfiles direkt damit zusammenhängen weiß ich nicht (vermute eher nicht), aber die Meldung irritiert mich dennoch, nur so debugmäßig pfostiert daher.

andibraeu, 2015-11-16:
> @bittorf: Also eigentlich nur: Ja, ist der interne Webserver, auf dem s140 Router lässt es sich reproduzieren und ich weiß auch nicht woran es liegt..
> 
Da kommt der Fehler immer? Guck ich mir nachher mal an

glycoknob, 2015-11-16:
Am 16. November 2015 um 22:02 schrieb Andi Bräu <notifications@github.com>:

>
> > @bittorf: Also eigentlich nur: Ja, ist der interne Webserver, auf dem
> s140 Router lässt es sich reproduzieren und ich weiß auch nicht woran es
> liegt..
>
> Da kommt der Fehler immer? Guck ich mir nachher mal an

Gute Frage - heute Nachmittag hab ich den Fehler gesehen via Telefon -
Router hat sich grad rebootet, Login und Passwort hab ich dir geschickt
 Seltsam.

andibraeu, 2015-11-17:
ist nun gefixt: https://github.com/weimarnetz/openwrt-packages/commit/5a8383fe109830e2cf95e02827b28a83dc9c1a21

Problem war, dass /tmp/CHECK_RECENT_GATEWAY anfangs leer ist und da in eine Variable unsinn geschrieben wurde. Das führt dazu, dass die URL zu lang wurde und der Server beim Versuch des Redirects kapitulierte.

bittorf, 2015-11-17:
wunderbar!

closed 2015-11-17 by andibraeu

storchi, 2015-11-19:
die 43 hat gerade das bad gateway problem.
wer mal schauen möchte...
cs

root@WeimarnetzMaschinenraum:~ df
Filesystem           1K-blocks      Used Available Use% Mounted on
rootfs                     448       260       188  58% /
/dev/root                 3072      3072         0 100% /rom
tmpfs                    14608      1020     13588   7% /tmp
/dev/mtdblock3             448       260       188  58% /overlay
overlayfs:/overlay         448       260       188  58% /
tmpfs                      512         4       508   1% /dev
root@WeimarnetzMaschinenraum:~ free
              total         used         free       shared      buffers
Mem:         29220        25380         3840          912         1596
-/+ buffers:              23784         5436
Swap:        14332          924        13408
root@WeimarnetzMaschinenraum:~ ps
   PID USER       VSZ STAT COMMAND
     1 root      1408 S    /sbin/procd
     2 root         0 SW   [kthreadd]
     3 root         0 SW   [ksoftirqd/0]
     5 root         0 SW<  [kworker/0:0H]
     6 root         0 SW   [kworker/u2:0]
     7 root         0 SW<  [khelper]
    28 root         0 SW<  [writeback]
    30 root         0 SW<  [bioset]
    31 root         0 SW<  [kblockd]
    32 root         0 SW   [kswapd0]
    95 root         0 SW   [fsnotify_mark]
   148 root         0 SW   [spi0]
   244 root         0 SW<  [deferwq]
   267 root         0 SW   [kworker/u2:2]
   296 root         0 SW   [kworker/0:2]
   333 root         0 SWN  [jffs2_gcd_mtd3]
   386 root       908 S    /sbin/ubusd
   387 root       772 S    /sbin/askfirst /bin/ash --login
   976 root         0 SW<  [ipv6_addrconf]
  1105 root         0 SW<  [cfg80211]
  1253 root      1040 S    /sbin/logd -S 16
  1262 root      1532 S    /sbin/rpcd
  1306 root      1564 S    /sbin/netifd
  1326 root      1168 S    /usr/sbin/odhcpd
  1801 root      1508 S    /usr/sbin/hostapd -P /var/run/wifi-phy0.pid 
-B /var/run/hostapd-phy0.conf
  1852 root      1360 S    udhcpc -p /var/run/udhcpc-eth1.pid -s 
/lib/netifd/dhcp.script -f -t 0 -i eth1 -C
  1855 root       800 S    odhcp6c -s /lib/netifd/dhcpv6.script -P0 
-t120 eth1
  2118 nobody    1000 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k 
-x /var/run/dnsmasq/dnsmasq.pid
  2119 root       924 S    /usr/sbin/dnsmasq -C /var/etc/dnsmasq.conf -k 
-x /var/run/dnsmasq/dnsmasq.pid
  2210 root      1152 S    /usr/sbin/dropbear -F -P 
/var/run/dropbear.1.pid -p 22 -K 300
  2277 root      1168 S    /usr/sbin/uhttpd -f -h /www -r 
WeimarnetzMaschinenraum -c /etc/httpd.conf -x /cgi-bin -t 60 -T 30 -k 20 
-A 1 -E /weimar-splash-index.sh -n 5 -N 100
  2316 root      1368 S    /usr/sbin/crond -L /dev/null
  2341 root      1364 S    sh /tmp/cron_watchdog.sh
  2754 root      1352 S    lock -n /tmp/VPN_LOCK
  3419 root       832 S <  vtund[c]: Node43 ether tap0
  3528 root      4300 S    /usr/sbin/olsrd -f /var/etc/olsrd.conf -nofork
  5756 root      1364 S    /usr/sbin/ntpd -n -S /usr/sbin/ntpd-hotplug 
-p 0.openwrt.pool.ntp.org -p 1.openwrt.pool.ntp.org -p 
2.openwrt.pool.ntp.org -p 3.openwrt.pool.ntp.org
17257 root         0 SW   [kworker/0:0]
18667 root      1220 R    /usr/sbin/dropbear -F -P 
/var/run/dropbear.1.pid -p 22 -K 300
18668 root      1384 S    -ash
18900 root      1352 S    sleep 60
18901 root         0 SW   [kworker/0:1]
19026 root      1360 R    ps

bittorf, 2015-11-19:
@storchi hat denn die 43 schon den bugfix von andi drinne? wenn nicht, brauchen wir erstmal nicht weiter zu suchen...

storchi, 2015-11-20:
Am 19.11.2015 um 20:24 schrieb Bastian Bittorf:
> @storchi <https://github.com/storchi> hat denn die 43 schon den bugfix
> von andi drinne? wenn nicht, brauchen wir erstmal nicht weiter zu suchen...

da haste recht. ich war verwirrt und dachte dieses tippmannsche lock 
dingens hatte was damit zu tun, und das hatte ich ja ausgeschlossen.
da machmer die 43 am di neu.
cs