weimarnetz/weimarnetz #147: CC: Knopf "Internet stellt dir dieser freundliche Nachbar zur Verfügung" linkt falsch

storchi, 2015-11-13:
ziel ist http://10.63.122.1/cgi-bin/luci/freifunk/contact
ziel sollte sein: http://10.63.122.2/cgi-bin/luci/freifunk/contact

liegt daran, dass das ap netz auf dem router jetzt die ip 10.63.122.2 hat.

Bitte nicht mit ip umstellung kommen...

cs

bittorf, 2015-11-14:
bitte einfach (und generell) die LANIP nehmen, die geht immer, also:
_ipsystem getvar LANADR 122 -> 10.63.122.33

zeige mal auf die stelle im code, dann bau ich das ein

andibraeu, 2015-11-14:
der nimmt die OLSR mainip... und die wird auf die LAN-IP gesetzt: https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/apply_profile.code#L1282

@storchi wie sieht /etc/config/olsr bei dir aus?

storchi, 2015-11-15:
weiss nicht. es is die 122.
cs

andibraeu, 2015-11-15:
linkt jetzt richtig... Problem war gar nicht der Router, sondern die Gatewayadresse. Die wird aus /tmp/GATEWAY_CHECK_RECENT ausgelesen und steht da so drin. Das funktioniert auch, wenn du mind. einen Hop weg bist. Hab das nun auf der Seite geändert: https://github.com/weimarnetz/web/commit/966a2b99a55da1b97dda9b1a950b567674fab62e

closed 2015-11-15 by andibraeu

bittorf, 2015-11-16:
die datei sieht z.b. so aus:
```
DATE=2015nov16_20:31; HOSTNAME=fonera2n; COST_INT=4070; GATEWAY=10.63.21.97; NEXTHOP=10.63.49.225; METRIC=3
```

bittorf, 2015-11-16:
@andibraeu  du hast was in einer....PHP datei? geaendert? wann wird die denn ausgeführt...?

andibraeu, 2015-11-17:
ja, die liegt auf unserem webserver und wird beim Aufruf von weimarnetz.de ausgeführt ;-)

bittorf, 2015-11-17:
wieso ist das nicht auf dem router selbst? waere das wegen der dezentralität viel besser?