weimarnetz/weimarnetz #150: Keypair für Signed Packages erzeugen

andibraeu, 2015-11-17:
Hintergrund: Packages sind signiert und mit opkg update kann man nur Pakete mit der richtigen Signatur installieren/updaten. Standardmäßig wird diese Signatur beim Bauen neu erzeugt und ist im OpenWRT-Verzeichnis in key-build und key-build.pub abgelegt.

Für unsere offiziellen Releases müssen wir dann alles einmal mit dem offiziellen Schlüsselpaar bauen.

bittorf, 2015-11-17:
dazu gehört auch gleich als erinnerung, das wir das Pärchen sichern um es im Fehlerfall wieder herstellen zu können.

andibraeu, 2015-11-23:
hab dafür mal ein private repository hier beantragt...

andibraeu, 2015-11-24:
Keys werde ich in unser privates repo packen: https://github.com/weimarnetz/keys

closed 2015-11-24 by andibraeu