weimarnetz/weimarnetz #154: GM: wlan ist mir zu unsichtbar

storchi, 2015-11-19:
es dauert teilweise bis zu 10/20s bis ein client (linux mac os) das ap wlan sieht
das beacon interval steht bei 1000 -> ich würde das gerne auf 500 stellen
cs

bittorf, 2015-11-19:
bitte auf 250 - das benutze ich auch in allen installationen. Es ist ein guter Kompromiss auf dem man sich intern auf dem Battlemesh geeinigt hat. Mann kann natuerlich noch schneller beacons aussenden, aber das geht auf den Akku von Mobilgeraeten die das dann empfangen, daher:

4 beacons/Sekunde = 250

storchi, 2015-11-19:
Am 19.11.2015 um 14:49 schrieb Bastian Bittorf:
> bitte auf 250 - das benutze ich auch in allen installationen. Es ist ein
> guter Kompromiss auf dem man sich intern auf dem Battlemesh geeinigt
> hat. Mann kann natuerlich noch schneller beacons [...]

das ist ja 4x mehr als bisher
das belegt zu viel airtime
cs

bittorf, 2015-11-19:
jedes normale WLAN steht auf 10 oder 20 mal pro sekunde (also beacon_int = 100 oder 50)

storchi, 2015-11-19:
Am 19.11.2015 um 15:31 schrieb Bastian Bittorf:
> jedes normale WLAN steht auf 10 oder 20 mal pro sekunde (also beacon_int
> = 100 oder 50)

stimmt nicht, hab hier 6 gegenbeispiele die ich empfange
cs

bittorf, 2015-11-19:
@storchi  was genau stimmt nicht?

storchi, 2015-11-19:
mhh. 100 hab ich bei nem nachbarnetz gesehn. der rest kommt wahrscheinlich nicht durch.
egal, 1000 ist zuviel. ich stells mal auf 500.

bittorf, 2015-11-19:
argumente sind sinnlos oder was? ich erklaere mir nen wolf und es wird einfach ignoriert...

storchi, 2015-11-20:
wir senden durch hybrid doppelt beacons auf dem physischen device -> macht auch irgendwie ein interval von 250
cs

bittorf, 2015-11-20:
@storchi neeeee. das beacon wird NUR vom radio ausgesandt (also phy0) und nicht von den virtuellen devices. d.h. auch wenn du 4 wlan's hast (wlan0, wlan0-1, wlan0-2, wlan0-3) wird nur ein beacon ausgesandt. deswegen steht die option auch bei "radio" und nicht bei "wifi-iface"

storchi, 2015-11-20:
das sieht mit horst aber anders aus.

bittorf, 2015-11-20:
danke - ich pruefe, komme aber erst naechste woche dazu. ich vermute horst zerlegt ein beacon in die anzahl der interfaces...

storchi, 2015-11-20:
Am 20.11.2015 um 11:07 schrieb Bastian Bittorf:
> danke - ich pruefe, komme aber erst naechste woche dazu. ich vermute
> horst zerlegt ein beacon in die anzahl der interfaces...

kann ich verneinen, denn es gab bei mir eine unterschiedliche anzahl von 
beacons auf wn ap/adhoc

andibraeu, 2015-11-20:
ist das schon auf einen niedrigeren Wert gestellt?

bittorf, 2015-11-20:
@storchi also kamen vom *gleichen* router eine unterschiedliche anzahl von beacons vom AP und vom ADHOC interface im selben Zeitraum?

storchi, 2015-11-21:
so stand es bei horst

storchi, 2015-11-21:
is ja auch wurst - 1000 ist zu wenig

andibraeu, 2015-12-01:
lasst uns das mit der neuen einstellung testen und ggf. später nochmal anpassen

closed 2015-12-01 by andibraeu

storchi, 2015-12-10:
nochmal fuers protokoll:
cs@TP-T420s:~/src/horst$ sudo tcpdump -ni mon0 type mgt subtype beacon | grep -i weimarnetz
tcpdump: WARNING: mon0: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 65535 bytes
11:08:41.966321 25369729082us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:42.012923 25369776218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:42.349880 25370112217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:42.397026 25370160217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:42.733460 25370496216us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:42.780940 25370544218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:43.117457 25370880217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:43.164942 25370928218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:43.502298 25371265085us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:43.549047 25371312219us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:43.885465 25371648218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:43.933037 25371696218us tsft 1.0 Mb/s 2432 MHz 11b -21dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:44.269353 25372032217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:44.316991 25372080218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:44.653460 25372416216us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:44.700959 25372464217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:45.038350 25372801090us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:45.084966 25372848218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:45.421399 25373184217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:45.469059 25373232218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:45.805432 25373568218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:45.853065 25373616218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:46.189485 25373952217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:46.236957 25374000218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:46.574261 25374337091us tsft 1.0 Mb/s 2432 MHz 11b -21dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:46.621062 25374384218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:46.957494 25374720217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:47.005062 25374768218us tsft 1.0 Mb/s 2432 MHz 11b -23dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:47.341502 25375104217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:47.388989 25375152218us tsft 1.0 Mb/s 2432 MHz 11b -21dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
11:08:47.725387 25375488217us tsft 1.0 Mb/s 2432 MHz 11b -21dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
[... cut ...]
11:08:49.261516 25377024217us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (Weimarnetz - Knoten 122) [1.0* 2.0* 5.5* 11.0* 6.0 9.0 12.0 18.0 Mbit] ESS CH: 5
11:08:49.309093 25377072218us tsft 1.0 Mb/s 2432 MHz 11b -22dB signal antenna 1 Beacon (weimarnetz) [1.0* 2.0 5.5 11.0 6.0 9.0 12.0 18.0 Mbit] IBSS CH: 5
^C184 packets captured
191 packets received by filter
0 packets dropped by kernel

wer lust hat kann sich es ausrechnen. bei beacon interval 375 und hybrid mit 2 wlans werden ca alle 0,184s beacons versendet, laut beacon interval sollte der abstand eher doppelt so gross sein
cs

bittorf, 2015-12-10:
ja, ich sehe (das gleiche): 41 beacons in ~7.3 sekunden, ergo: 5.6 beacons/sekunde = alle 0.18 Sekunden 1 beacon. Die frage ist aber: sind das einzelne pakete oder gehoeren die zusammen? (also 1 beacon mit 2 feldern)

storchi, 2015-12-10:
laut tcpdump timestamp nicht

storchi, 2015-12-10:
1 beacon 2 felder?
die frage stellt sich nicht.
http://mrncciew.com/2014/10/08/802-11-mgmt-beacon-frame/