weimarnetz/weimarnetz #157: GM: splashpage fordert zum Downgrade

storchi, 2015-11-20:
ich finds ja gut!
[image]

glycoknob, 2015-11-20:
Hier ist der Code: 

https://github.com/weimarnetz/openwrt-packages/blob/master/utils/luci-weimar-splash/luasrc/view/weimarnetz-splash/infopage.htm#L79

In dem Fall ist es so, dass es mit ODER verknüpft ist... also OpenWRT ist neuer ODER kalua - hier ist kalua neuer...

glycoknob, 2015-11-20:
Ich glaub aber, dass das so grad eher verwirrt als hilft - oder?

andibraeu, 2015-11-20:
Ja, das liegt an der falschen Versionsnummer für kalua. Ist in Arbeit.

andibraeu, 2015-11-27:
ist nun auch für den Bau als OpenWRT-Package gelöst: https://github.com/weimarnetz/openwrt-packages/commit/442629cf131e12665f281552bae1cf0f3845869e

closed 2015-11-27 by andibraeu