weimarnetz/weimarnetz #158: Splash ist kaputt - s140

glycoknob, 2015-11-20:
Status: 500 Internal Server Error /usr/lib/lua/luci/dispatcher.lua:433: Failed to execute template dispatcher target for entry '/freifunk/captive'. The called action terminated with an exception: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'weimarnetz-splash/infopage'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/bootstrap/header'. A runtime error occured: /usr/lib/lua/luci/util.lua:587: Unable to establish ubus connection stack traceback: [C]: in function 'assert' /usr/lib/lua/luci/dispatcher.lua:433: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:168: in function

andibraeu, 2015-11-20:
war das evtl ganz kurz nach dem neustart? jetzt geht sie nämlich...

glycoknob, 2015-11-20:
Heute gegen 14.45

glycoknob, 2015-11-28:
ich schließ das mal - wenn das mit dem Release auch passiert, kann man das ja wieder öffnen.

closed 2015-11-28 by glycoknob