weimarnetz/weimarnetz #160: Liste aller Router erstellen

andibraeu, 2015-11-21:
Fürs Update und um Nutzer zu informieren. Nötige Felder:

* Name
* Knotennummer
* Typ
* kann unsere neue Firmware installiert werden?
* kann etx_ffeth eingestellt werden?
* kennen wir jemanden, der Zugang zum Router hat?

bittorf, 2015-11-21:
sehr gut! kannst du mir man einen link geben oder einen tipp wie man moeglichst minimal ein monitoring JSON-update schickt?

andibraeu, 2015-11-22:
da isses: https://github.com/weimarnetz/openwifimap-api/blob/master/API.md

Wir benutzen das Package auf den Routern: https://github.com/weimarnetz/openwrt-packages/tree/master/utils/luci-app-owm

glycoknob, 2015-11-22:
Hier mal ne Möglichkeit zu filtern: https://gist.github.com/glycoknob/7c124b5f6be26fa27532

bittorf, 2015-11-22:
@andibraeu das sieht gut aus. wie ist "unsere" APIURL um einen knoten zu updaten?

curl -X PUT APIURL/update_node/myid -d '{...mydata...}'

bittorf, 2015-11-22:
@glycoknob das ist ein guter anfang. wir brauchen natuerlich eine excel tabelle 8-)

andibraeu, 2015-11-22:
mapapi.weimarnetz.de

glycoknob, 2015-11-22:
Hier noch mit Datum des letzten Kontakts... https://gist.github.com/glycoknob/366786299db235349c4f 

    $ cat data.json | jq '.rows | .[].doc | { node: .hostname, mtime: .mtime, contact: .freifunk.contact, weimarnetz: .weimarnetz, hardware: .system.sysinfo[1], firmware: .firmware }'

Excel geht auch irgendwie... sortieren wäre auch cool, mal sehen...

glycoknob, 2015-11-22:
Gibt übrigens nur 69 Knoten, die sich heute gemeldet haben, diesen Monat nur 89 Knoten... sind das alle?

andibraeu, 2015-11-22:
dort melden sich nur die, die auch den den entsprechenden Client drauf haben

glycoknob, 2015-11-22:
okay, bei reg.weimarnetz.de sind es 92 Knoten, die sich heute gemeldet haben.

glycoknob, 2015-11-22:
quick&dirty:

    % curl "http://reg.weimarnetz.de/ffweimar/knoten" | grep last_seen | awk '{ print $2 }' | sort > nodes
    % cat nodes | cut -b1-10 | xargs -I{} -n1 date -d @{} | grep "22. Nov" | wc -l 

Aber aus der Knotennummer können wir die IP ausrechnen dann haben wir auch ne Liste...

andibraeu, 2015-11-22:
Clients ohne Geokoordinaten melden sich auf älteren Firmwareversionen noch nicht, das hab ich erst später nachgerüstet...

bittorf, 2015-11-22:
einen einzelnen Datensatz bekommt man uebrigens so:
```
APIURL=http://mapapi.weimarnetz.de
curl $APIURL/db/Stadtpark1.olsr
```

andibraeu, 2015-11-28:
hab das mal verexcelt: https://docs.google.com/spreadsheets/d/1BExBgtmey05wtY8fbXR27gkhtyJelE9LyMJJDoH7p3A/edit?usp=sharing

bittorf, 2015-11-30:
Das Tabellenkalkulation ist erstmal nicht schlecht, ich habs aber immer noch nicht geschafft einen datensatz so zu senden, des es im monitoring aktiv wird:

```
APIURL='http://mapapi.weimarnetz.de'
URL="$APIURL/update_node/myid"
# HOSTNAME/OPENWRT_REV/NODENUMBER kommen vom loader

cat >FILE <<EOF
{
"id":"$HOSTNAME","_id":"$HOSTNAME","type":"node",
"longitude":50.97389,"latitude":11.31875,
"updateInterval":86400,"hostname":"$HOSTNAME",
"fffversion":"333333","revision":"$OPENWRT_REV","nodenumber":$NODENUMBER
}
EOF

curl -X PUT "$URL" -d "$( cat FILE )"
curl $APIURL/db/$HOSTNAME.olsr
```

die frage ist: was muss minimal gesendet werden, damit es akzeptiert wird?

glycoknob, 2015-11-30:
Kann ich auch nicht beantworten, aber owm.lua schickt folgendes raus: 

```
[~]$ sudo nc -l 80                                                                                   
PUT /update_node/netbox.olsr HTTP/1.1
Connection: close
Content-Type: application/json
User-Agent: LuCI httpclient 0.1
Content-Length: 5057
Host: 192.168.2.100

{"freifunk":{"contact":{"name":"mt","mail":"Email hidden","nickname":"mt"},"community":{"owm_api":"http://192.168.2.100","name":"weimar","mapserver":"http://map.weimarnetz.de","mail":"Email hidden","homepage":"http://freifunk.net"}},"api_rev":"1.0","hostname":"netbox","hardware":"Atheros AR9341 rev 3","latitude":50.978942840127,"firmware":{"distname":"","luciname":"LuCI Master","fffversion":"402430","luciversion":"git-15.322.41240-4e44955","revision":"OpenWrt (Weimarnetz edition) Chaos Calmer 15.05","distversion":"OpenWrt (Weimarnetz edition) Chaos Calmer 15.05","name":""},"ipv4defaultGateway":{"gateway":"10.63.188.57","dest":"0.0.0.0/0","dev":"tap0"},"type":"node","script":"luci-app-owm","interfaces":[{"netmask":"255.255.255.240","ifname":"eth1","ipaddr":"10.63.188.33","up":"1","device":"eth1","neighbors":[],"name":"lan","wifi":[],"physicalType":"ethernet","ipv4Addresses":["10.63.188.33"],"proto":"static"},{"accessNote":"everyone is welcome!","proto":"static","ipaddr":"10.63.188.2","up":"1","netmask":"255.255.255.224","neighbors":[],"ifname":"wlan0-1","access":"free","ssid":"weimar.freifunk.net","txpower":"20","name":"wlan","channel":"5","wifi":[{"wirelessdevice":{"country":"US","beacon_int":"375","path":"platform/ar934x_wmac","frag":"off","hwmode":"11g","name":"radio0","channel":"5","txpower":"20","htmode":"HT20","type":"mac80211"},"network":"wlan","device":"radio0","mode":"ap","assoclist":[],"ssid":"weimar.freifunk.net"}],"mode":"ap","ipv4Addresses":["10.63.188.2"],"physicalType":"wifi"},{"ifname":"wlan0","access":"free","netmask":"255.0.0.0","bssid":"02:ca:ff:ee:ba:be","wifi":[{"assoclist":[],"wirelessdevice":{"country":"US","beacon_int":"375","path":"platform/ar934x_wmac","frag":"off","hwmode":"11g","name":"radio0","channel":"5","txpower":"20","htmode":"HT20","type":"mac80211"},"ssid":"weimarnetz","network":"wlanadhoc","mcast_rate":"1000","mode":"adhoc","device":"radio0","bssid":"02:ca:ff:ee:ba:be"}],"physicalType":"wifi","proto":"static","ssid":"weimarnetz","up":"1","neighbors":[],"txpower":"20","name":"wlanadhoc","channel":"5","accessNote":"everyone is welcome!","ipaddr":"10.63.188.1","ipv4Addresses":["10.63.188.1"],"mode":"adhoc"},{"neighbors":{"3a:1f:XX:XX:33:76":{"ip4":["10.63.188.57"]}},"ifname":"tap0","name":"tap0","up":"1","device":"tap0","physicalType":"ethernet","ipv4Addresses":[],"wifi":[]}],"links":[],"weimarnetz":{"gateway":"10.63.188.1","nodenumber":188,"nexthop":"10.63.188.1","gatewaycost":1},"system":{"sysinfo":["Atheros AR9341 rev 3","TP-Link TL-WR841N/ND v8",{"totalhigh":0,"freeram":1044480,"mem_unit":1,"sharedram":552960,"bufferram":1642496,"uptime":394,"totalswap":0,"procs":45,"totalram":29667328,"freehigh":0,"loads":[0.06884765625,0.2177734375,0.13671875],"freeswap":0}],"uptime":[394],"loadavg":[0.06884765625,0.2177734375,0.13671875]},"longitude":11.321186997224,"olsr":{"ipv4Config":{"routeNetlinkSocket":4,"useSourceIpRoutes":false,"linkQualityFisheye":true,"defaultTcValidityTime":300,"defaultMidEmissionInterval":5,"maxSendMessageJitter":2,"defaultHnaEmissionInterval":5,"startTime":1448883009,"defaultIpv4Broadcast":"auto","linkQualityAging":0.05,"hostEmulation":false,"willingness":7,"olsrdVersion":"olsr.org - 0.9.0.3-git_121e044-hash_0baeac4dd2358cb60e21dff4e80abd4a","tcRedundancy":2,"rtProto":3,"deleteInternetGatewaysAtStartup":false,"rtTableDefault":254,"brokenLinkCost":4194304,"defaultTcEmissionInterval":5,"useNiit":false,"noFork":true,"lockFile":"/var/run/olsrd-ipv4.lock","willingnessAuto":false,"ipSize":4,"pollRate":50,"defaultAutoDetectChanges":true,"typeOfService":192,"exitValue":0,"defaultHelloEmissionInterval":2,"maxPrefixLength":32,"rtTable":254,"debugLevel":0,"unicastSourceIpAddress":"10.63.188.33","ipcAllowedAddresses":[],"linkQualityAlgorithm":"etx_ff","defaultMidValidityTime":300,"olsrdBuildDate":"2015-11-30 03:31:32","totalIpcConnectionsAllowed":0,"fibMetrics":"flat","rtTableTunnel":254,"defaultHnaValidityTime":300,"minTcValidTime":0,"olsrPort":698,"niit4to6InterfaceIndex":0,"defaultIpv6Multicast":"ff02::6d","os":"Undefined","olsrdBuildHost":"zeno","willingnessUpdateInterval":20000,"ipVersion":2,"hasIpv6Gateway":false,"linkQualityNatThreshold":1,"routeMonitorSocket":5,"rtTableDefaultPriority":-1,"ioctlSocket":3,"rtTablePriority":-1,"hasIpv4Gateway":true,"defaultInterfaceMode":"mesh","rtTableDefauiltOlsrPriority":-1,"mainIpAddress":"10.63.188.33","setIpForward":true,"hna":[{"gateway":"10.63.188.33","destination":"0.0.0.0","genmask":0},{"gateway":"10.63.188.33","destination":"10.63.188.0","genmask":26}],"niit6to4InterfaceIndex":0,"rtTableTunnelPriority":-1,"allowNoInterfaces":true,"nicChangePollInterval":2500,"maxTcValidTime":2000,"defaultLinkQualityMultipliers":[],"defaultHelloValidityTime":20,"mprCoverage":7,"clearScreen":false,"brokenRouteCost":-1,"smartGateway":false,"linkQualityLevel":2},"links":[{"localIP":"10.63.188.58","sourceAddr":"10.63.188.58","linkQuality":1,"validityTime":186914,"remoteIP":"10.63.188.57","destAddr":"10.63.188.57","neighborLinkQuality":1,"linkCost":1024}]},"updateInterval":3600,"location":"hvf5"}
```

In hübsch:
```
{
  "freifunk": {
    "contact": {
      "name": "mt",
[... cut ...]
      "mprCoverage": 7,
      "clearScreen": false,
      "brokenRouteCost": -1,
      "smartGateway": false,
      "linkQualityLevel": 2
    },
    "links": [
      {
        "localIP": "10.63.188.58",
        "sourceAddr": "10.63.188.58",
        "linkQuality": 1,
        "validityTime": 186914,
        "remoteIP": "10.63.188.57",
        "destAddr": "10.63.188.57",
        "neighborLinkQuality": 1,
        "linkCost": 1024
      }
    ]
  },
  "updateInterval": 3600,
  "location": "hvf5"
}
```

andibraeu, 2015-11-30:
da stehen die Checks für ein minimales Set: https://github.com/freifunk/openwifimap-api/blob/master/validate_doc_update.js#L44

Minimal wäre demzufolge das da:
```
{
  "api_rev": "1.0",
  "hostname": "netbox",
  "latitude": 50.978942840127,
  "type": "node",
  "longitude": 11.321186997224,
  "updateInterval": 3600,
  "location": "hvf5"
}
```

Initial wäre vielleicht das auch gut, um ein paar Firmware- und Knoteninfos zu haben
```
{
  "freifunk": {
    "contact": {
      "name": "mt",
      "mail": "Email hidden",
      "nickname": "mt"
    },
    "community": {
      "owm_api": "http://192.168.2.100",
      "name": "weimar",
      "mapserver": "http://map.weimarnetz.de",
      "mail": "Email hidden",
      "homepage": "http://freifunk.net"
    }
  },
  "api_rev": "1.0",
  "hostname": "netbox",
  "latitude": 50.978942840127,
  "firmware": {
    "distname": "",
    "luciname": "LuCI Master",
    "fffversion": "402430",
    "luciversion": "git-15.322.41240-4e44955",
    "revision": "OpenWrt (Weimarnetz edition) Chaos Calmer 15.05",
    "distversion": "OpenWrt (Weimarnetz edition) Chaos Calmer 15.05",
    "name": ""
  },
  "type": "node",
  "script": "luci-app-owm",
  "weimarnetz": {
    "gateway": "10.63.188.1",
    "nodenumber": 188,
    "nexthop": "10.63.188.1",
    "gatewaycost": 1
  },
  "longitude": 11.321186997224,
  "updateInterval": 3600,
  "location": "hvf5"
}
```

glycoknob, 2015-11-30:
Der `content-type` Header könnte auch entscheidend sein bei solchen Sachen.

andibraeu, 2015-11-30:
also das da:
```
curl -X PUT http://mapapi.weimarnetz.de/update_node/bastitest.olsr -d '{
  "api_rev": "1.0",
  "hostname": "bastitest",
  "latitude": 50,
  "type": "node",
  "longitude": 12,
  "updateInterval": 3600,
  "location": "bastistest"
}'
doc updated% 
```
funktioniert und die Knoten tauchen auch in mapapi.weimarnetz.de/db/_all_docs?include_docs=true auf. Im Moment wird auf der Monitoringseite aber alles rausgefoltert, was keine Firmwaresection hat

mit sollte es dann gehen:

```
curl -X PUT http://mapapi.weimarnetz.de/update_node/bastitest.olsr -d '{
  "api_rev": "1.0",
  "hostname": "bastitest",
  "latitude": 50,
  "type": "node",
  "longitude": 12,
  "updateInterval": 3600,
  "location": "bastistest",

"firmware": {
    "distname": "",
    "luciname": "LuCI Master",
    "fffversion": "402430",
    "luciversion": "git-15.322.41240-4e44955",
    "revision": "OpenWrt (Weimarnetz edition) Chaos Calmer 15.05",
    "distversion": "OpenWrt (Weimarnetz edition) Chaos Calmer 15.05",
    "name": ""
  }
}'
```

bittorf, 2015-11-30:
wow! ich werde heute abend hacken!

bittorf, 2015-12-05:
ok, habe gehakt. es tauch nun bald ein paar neue knoten im monitoring auf. problem: die knotennummer wird nicht genommen. mein JSON sieht so aus:

```
{
"api_rev": "1.0",
"hostname": "Exkesselsaal-GarageDodge",
"latitude": 50,
"type": "node",
"longitude": 12,
"updateInterval": 3600,
"location": "Exkesselsaal-GarageDodge",
"system": {
        "sysinfo": [
                "Qualcomm Atheros QCA9558 rev 0",
                "TP-LINK TL-WR1043ND v2"
        ],
        "uptime": [
                51021
        ]
        },
"weimarnetz": {
        "gateway": "10.63.21.97",
        "nodenumber": 602,
        "nexthop": "10.63.42.65",
        "gatewaycost": 1
        },
"firmware": {
        "distname": "",
        "luciname": "no LuCI",
        "fffversion": "402593",
        "luciversion": "0",
        "revision": "OpenWrt (kalua) r41353",
        "distversion": "OpenWrt (kalua) r41353",
        "name": ""
        }
}
```
was mache ich falsch?

andibraeu, 2015-12-05:
nix, ist ein Anzeigeproblem...

andibraeu, 2015-12-05:
hab es im monitoring ausgebessert.

bittorf, 2015-12-06:
ok, schonmal sehr gut. noch eine kleinigkeit: wenn man auf einen router "draufdrueckt", sieht man "Geraet" und "Hardware". eins ist leer, ich sende es aber mit...auch ein anzeigeprob?

andibraeu, 2015-12-06:
schreib auch mal reale geodaten rein... weimarnetz hat inzwischen Knoten in Wunsiedel

andibraeu, 2015-12-06:
Die Hardware schreibt OWM direkt in Hardware, auf oberster Ebene, siehe Beispiel von Martin: https://github.com/weimarnetz/weimarnetz/issues/160#issuecomment-160606845

bittorf, 2015-12-06:
also naechstes kuemmere ich mich um die koordinaten

bittorf, 2016-01-12:
ich habe gestern fuer fast alle die koordinaten hinterlegt. es klappt aber nicht, wenn ich den datensatz abfrage stehen die "falschen" koordinaten drinne:

```
curl http://mapapi.weimarnetz.de/db/EG-labor-AP.olsr
#
"latitude":50.973889999999997258,"longitude":11.318749999999999645,
```

aber uebertragen wird:
```
"latitude": 50,989873,
"longitude": 11,336828,
```
ich vermute es liegt am komma Vs. punkt:
```
root@EG-labor-AP:~ _watch monitoring force
Jan 12 08:06:52 ' [up 14159s 8676 bytes free] TERM:screen -ash [daemon.debug] curl_it() max 30s, 125 bytes, wget  -qO -  'http://intercity-vpn.de/networks/liszt28/meshrdf/?refresh=true&mac=f4ec389d8182&up=3&h2=8884&load=28&version=403484&r4=0&r5=3'
Jan 12 08:06:53 ' [up 14160s 8792 bytes free] TERM:screen -ash [daemon.info] heartbeat_mapapi() netcat'ing file '/tmp/kalua/mapapi.cache' to 'mapapi.weimarnetz.de' port 80 / update_node/EG-labor-AP.olsr
HTTP/1.1 500 Internal Server Error
Access-Control-Allow-Origin: *
Server: CouchDB/1.3.0 (Erlang OTP/R15B01)
Date: Tue, 12 Jan 2016 07:06:53 GMT
Content-Type: text/plain; charset=utf-8
Content-Length: 1323
Cache-Control: must-revalidate

{"error":"render_error","reason":"function raised error: (new SyntaxError(\"JSON.parse\", \"updates.node\", 2)) \nstacktrace: ([object Object],[object Object])@updates.node:2\nrunUpdate(function (doc, req) {var newdoc = JSON.parse(req.body);if (newdoc.type != \"node\" && newdoc.type != \"node_no_loc\") {return [null, \"Error: type != node\"];}if (doc) {newdoc._id = doc._id;newdoc._rev = doc._rev;}if (!newdoc._id && req.id) {newdoc._id = req.id;}var date = (new Date).toISOString();newdoc.ctime = doc ? doc.ctime : date;newdoc.mtime = date;return [newdoc, \"doc updated\"];},[object Object],[object Array])@/usr/local/share/couchdb/server/main.js:961\n(function (doc, req) {var newdoc = JSON.parse(req.body);if (newdoc.type != \"node\" && newdoc.type != \"node_no_loc\") {return [null, \"Error: type != node\"];}if (doc) {newdoc._id = doc._id;newdoc._rev = doc._rev;}if (!newdoc._id && req.id) {newdoc._id = req.id;}var date = (new Date).toISOString();newdoc.ctime = doc ? doc.ctime : date;newdoc.mtime = date;return [newdoc, \"doc updated\"];},[object Object],[object Array])@/usr/local/share/couchdb/server/main.js:1031\n(\"_design/owm-api\",[object Array],[object Array])@/usr/local/share/couchdb/server/main.js:1505\n()@/usr/local/share/couchdb/server/main.js:1550\n@/usr/local/share/couchdb/server/main.js:1561\n"}
```

bittorf, 2016-01-12:
komischerweise erscheinen die knoten nicht auf der Live-Karte:
http://www.weimarnetz.de/maplive/

wenn ich mir die Daten von anderen Knoten anschauen, werden lat/long auch an voellig unterschiedlichen Stellen abgelegt:
```
curl "http://mapapi.weimarnetz.de/db/fh331.olsr" | grep itude
```

bei mir sieht das so aus:
```
root@EG-labor-AP:~ head -n20 /tmp/kalua/mapapi.cache
 {
"api_rev": "1.0",
"hostname": "EG-labor-AP",
"hardware": "Atheros AR9132 rev 2",
"latitude": 50.989873,
"longitude": 11.336828,
"type": "node",
"updateInterval": 3600,
"location": "EG-labor-AP",
...
```

glycoknob, 2016-01-12:
Hmm, hier seh ich nix: 
http://www.weimarnetz.de/maplive/

Auf http://www.weimarnetz.de/map2/ tauchen die Knoten auf.

glycoknob, 2016-01-12:
> wenn ich mir die Daten von anderen Knoten anschauen, werden lat/long auch an voellig unterschiedlichen Stellen abgelegt:

Das müsste egal sein, Objekte sind  ungeordnet in JSON - Position ist egal, solange wie es die gleiche Verschachtelungsebene ist.

bittorf, 2016-01-12:
Stimmt, sieht gut aus. Evtl. synct die andere Karte die Daten nicht so oft.
Ausserdem: wir sollten fuer die karte alle  verbindungen zum VPN-server ausblenden.

andibraeu, 2016-01-12:
maplive ist die karte, die auch auf den routern liegt und die informationen aus olsr gewinnt. Deswegen heißt die "live": http://home.andi95.de:3380/cgi-bin/luci/freifunk/map

das muss im nameservice plugin konfiguriert sein:

```
config 'LoadPlugin'                                                                            
        option 'library' 'olsrd_nameservice.so.0.3'
        option 'ignore' '0'                  
        option 'hosts_file' '/var/etc/hosts'                                         
        option 'suffix' '.olsr'
        option 'interval' '180'
        option 'timeout' '3600'            
        option 'latlon_file' '/var/run/latlon.js'                                
        option 'latlon_infile' '/var/run/latlon.txt'
```

andibraeu, 2016-01-12:
@bittorf in deinem Beispiel stehen doch auch die Werte drin, die auf der OpenWifiMap ankommen

andibraeu, 2016-01-17:
hier sind wir schön weit abgedriftet.. bitte ein neues Ticket aufmachen, wenn es hier noch was gibt

closed 2016-01-17 by andibraeu