weimarnetz/weimarnetz #163: VPN: Verbindung kommt nicht zu stande

andibraeu, 2015-11-24:
Beispiel: WeimarnetzMaschinenraum

lock ist gesetzt, und nichts passiert

andibraeu, 2015-11-25:
_net local_inet_offer sagt, es handele sich um eine PPPoE-Verbindung, ist aber dhcp auf wan

@Bittorf sagt dir das was?

bittorf, 2015-11-25:
interessant. ich pruefe.

andibraeu, 2015-11-25:
doch, ist das allgemein bekannte standardpasswort...

Ausgabe _net local_inet_offer:
```
Nov 25 10:43:10 ' [up 23858s 4501504 bytes free] TERM:screen -ash [daemon.debug] net_local_inet_offer() PHYDEV: 'eth1' METRIC: '' N: 'static' D: '192.168.254.1'
Nov 25 10:43:10 ' [up 23858s 4468736 bytes free] TERM:screen -ash [daemon.info] net_local_inet_offer() found local default-gateway over device 'eth1' = 'pppoe'
```

andibraeu, 2015-11-25:
hab da gerade mal wan eingetragen: https://github.com/weimarnetz/weimarnetz/blob/GebrannteMandeln/openwrt-addons/etc/kalua/net#L409

zusammen mit #165 startet vpn zumindest

bittorf, 2015-11-25:
ok, komme nun drauf. ursache gefunden, bereite patch vor.

bittorf, 2015-11-25:
ok, dies ist die ursache allen uebels:
https://github.com/weimarnetz/weimarnetz/pull/166
https://github.com/weimarnetz/weimarnetz/pull/167

die aenderung in vpn_needed() wuerde ich ja rueckgaengig machen, oder?

andibraeu, 2015-11-25:
warum? ist ein return 0 nicht relativ eindeutig und lesbar?

andibraeu, 2015-11-25:
mach den PR mal bitte in Gebrannte Mandeln

bittorf, 2015-11-25:
du hast recht. das return 0 muss bleiben.
ich dacht PR167 ist in Gebrannte Mandeln?

andibraeu, 2015-11-25:
nein, da wolltest du zurück in master mergen

bittorf, 2015-11-25:
puh...ich versuchs nochmal

andibraeu, 2015-11-26:
das funktioniert nun

closed 2015-11-26 by andibraeu