weimarnetz/weimarnetz #164: _switch show funktioniert nicht

andibraeu, 2015-11-24:
liefert z.b. bei 841n v7 u.uuuuu

a) /etc/HARDWARE enthält TP-Link 841N/ND v7, in /etc/kalua/switch steht TP-LINK 841N/ND v7
b) selbst wenn /etc/HARDWARE gesetzt ist liefert u.uuuu

Ideen?

bittorf, 2015-11-25:
wow! ich repariere das,,,

andibraeu, 2015-11-25:
hm, ja.. ich hatte es auch geändert und es hat trotzdem nicht funktioniert

bittorf, 2015-11-25:
die variable HARDWARE wird aus /tmp/loader genommen.
wenn der loader generiert wird, kommt es aus /etc/HARDWARE.
wenn man die datei aendert, muss der loader neu gebaut werden: "regen"

bittorf, 2015-11-25:
die frage ist: wie kann HARDWARE "falsch" gesetzt sein?

andibraeu, 2015-11-25:
hm, wie wird /etc/HARDWARE gesetzt? liegt da evtl. der Hase im Pfeffer?

bittorf, 2015-11-25:
hier:
https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/apply_profile.code#L212

bei mir wird diese datei aber schon seit ner weile beim build gesetzt. (jede hardware krieg einen eigenen build-run)

andibraeu, 2015-11-26:
Öbeltäter gefonden: Die Hardware wird von dort geladen: 

```/etc/rc.d/S15set_etc_hardware:        read HARDWARE < /tmp/sysinfo/model```

Und da steht ```TP-Link``` statt ```TP-LINK```

Gibts bei OpenWRT ne Übersicht, wie die Geräte bezeichnet sind?

Für jede Hardware ein eigener Build lässt sich im Meshkit schlecht realisieren

glycoknob, 2015-11-26:
nicht so wirklich - `/tmp/sysinfo/model ` kommt hierher: http://git.openwrt.org/?p=openwrt.git;a=blob;f=target/linux/ar71xx/base-files/lib/ar71xx.sh;hb=HEAD 

soweit ich das sehe gibt es das für jede Architektur. 

Ansonsten gibt's im OpenWRT-Build System Variablen, die man auch ins Image-Schreiben könnte: http://git.openwrt.org/?p=openwrt.git;a=blob;f=target/linux/ar71xx/generic/profiles/tp-link.mk;h=2875290e9901ee4f5772aaa6e20ba5691f1e4612;hb=HEAD

Der Hack wegen `/tmp/sysinfo/model` kommt nur daher, dass wir `/etc/HARDWARE` nicht beim boot haben... IMHO ist `/tmp/sysinfo/model` gut genug.

andibraeu, 2015-11-26:
dann müssten wir die Bezeichnungen, wo sie verglichen werden ergänzen/ändern...

glycoknob, 2015-11-26:
Hmm, ja das ist dann ein generelles Problem... müssten wir mal überlegen was wir machen - so schreibt OpenWRT die UCI-Config für switche: https://github.com/openwrt/openwrt/blob/master/target/linux/ar71xx/base-files/etc/uci-defaults/02_network - ist also eigentlich auch schon alles da.

bittorf, 2015-11-26:
im prinzip muss es einen Wrapper gaben, openwrt-name -> echter name.
Problem aus meiner Sicht ist, das OpenWrt da einiges falsch macht und versucht Modelle
zusammenzufassen. z.b. sind alle Ubiquiti-Bullet-Modelle 'bullet'. In wahrheit gibt es aber:

Bullet 2M
Bullet 5M
Bullet Titanium 2M
Bullet Titanium 5M
Picostation 2M
Picostation 5M

es ist halt nur ein build (gleiche firmware) aber ich wuerde das im Monitoring schon gerne sehen.
Der gleiche Name wird/soll ja auch fuer den Download gelten. Und da ist es völlig verwirrend 'bullet' runterzulassen, wenn man eine Picostation hat.
Es gibt noch viel krassere Beispiele, das ist nur eins von vielen.

Also bitte beim build im Meshkit einbauen lassen.

glycoknob, 2015-11-26:
Ich glaub die Erkennung für die Ubiquity-Modell ist noch in apply_profile.code enthalten und sollte auch funktionieren, nur für andere Typen wird `/tmp/sysinfo/model` genommen. 

Am einfachsten wäre es doch, einfach das TP-Link zu TP-LINK im set_hardware Script zu machen - oder _switch eben anpassen. Meshkit könnte kompliziert werden.

bittorf, 2015-11-26:
ich wuerde vorschlagen fuer die bekannten modelle muss es in 'S15set_etc_hardware' einen uebersetzer geben. die bekannten modelle sind. die kritischen kann man da erstmal machen...
```
UML
x86_64
Soekris net5501
PC Engines WRAP
PC Engines ALIX.2
Buffalo WZR-HP-AG300H
TP-LINK TL-WR703N v1
TP-LINK TL-MR3020
TP-LINK TL-WR741ND v1
TP-LINK TL-WR741ND v2
TP-LINK TL-WR741ND v4
TP-LINK TL-WR841N/ND v1.5
TP-LINK TL-WR841N/ND v3
TP-LINK TL-WR841N/ND v5
TP-LINK TL-WR841N/ND v7
TP-LINK TL-WR841N/ND v8
TP-LINK TL-WR841N/ND v9
TP-LINK TL-WR847N v8
TP-LINK TL-WDR3600
TP-LINK TL-WDR4300
TP-LINK TL-WDR4900 v1
TP-LINK TL-WR940N
TP-LINK TL-WR941ND v4
TP-LINK TL-WR1043ND
TP-LINK TL-WR1043ND v2
TP-LINK TL-WDR7500
TP-LINK Archer C7
TP-LINK Archer C7 v2
Mercury MAC1200R
D-Link DIR-615-E4
Ubiquiti Nanostation2
Ubiquiti Picostation2
Ubiquiti Bullet2
Ubiquiti Nanostation5
Ubiquiti Picostation5
Ubiquiti WispStation5
Ubiquiti Bullet5
Ubiquiti Bullet M2
Ubiquiti Bullet M5
Ubiquiti Picostation M2
Ubiquiti Picostation M5
Ubiquiti Nanostation M2
Ubiquiti Nanostation M5
Ubiquiti Nanostation M2 XW
Ubiquiti Nanostation M5 XW
Ubiquiti Nanostation Loco M2 XW
Ubiquiti Nanostation Loco M5 XW
Targa WR-500-VoIP
Speedport W500V
Linksys WRT54G/GS/GL
Buffalo WHR-HP-G54
Dell TrueMobile 2300
ASUS WL-500g Premium
ASUS WL-500g Premium v2
ASUS WL-HDD 2.5
T-Mobile InternetBox TMD SB1-S
4G Systems MTX-1 Board
La Fonera 2.0N
D-Link DIR-300-B1
Nexx WT3020A
Nexx WT3020H
Nexx WT3020F
Nexx WT3020AD
Mikrotik Routerboard 532
Seagate GoFlex Home
Seagate GoFlex Net
Pandaboard
Beagleboard
FritzBox 7170
```

andibraeu, 2015-11-26:
stimme @glycoknob zu, warum nicht einfach TP-Link statt TP-LINK? wird das von woanders her in UPPERCASE ausgelesen?

bittorf, 2015-11-26:
stimmt denn ansonsten der rest? dann koennte man einen einfachen convert machen. bedenkt auch, das dies zwar fuer das wort "TP Link" hinhaut, aber ansonsten nicht:
```
root@EG-labor-AP:~ cat /tmp/sysinfo/model
TP-Link TL-WR1043N/ND v1
root@EG-labor-AP:~ cat /etc/HARDWARE 
TP-LINK TL-WR1043ND

root@EG-bastianHQ:~ cat /etc/HARDWARE 
TP-LINK TL-WDR4300
root@EG-bastianHQ:~ cat /tmp/sysinfo/model
TP-Link TL-WDR4300 v1
```

andibraeu, 2015-11-26:
woher stammen eigentlich die guten™ Namen? Mir scheint, als hätte /tmü/sysinfo/model die detaillierteren Namen

glycoknob, 2015-11-26:
Die werden - wenn nicht beim Build angegeben direkt hier abgefragt: https://github.com/bittorf/kalua/blob/master/openwrt-build/apply_profile.code 

Ich hatte das mal für Weimarnetz rausoperiert, daher kommt auch das Start-Script - aber im Prinzip ist es egal, ich denk mir halt - Die OpenWRT Daten genügen für 90% und wenn die falsch sind können wir nen Bug gegen OpenWRT filen und solange selbst in apply_profile.code patchen..

Mir ists egal, macht das womit wir glücklich werden - wir können das Script auch wegwerfen und die Erkennung von kalua wieder einbauen - ich weiß halt nicht ob sich das lohnt. 

Bin grad mal mit grep -ri über den Code... die Stellen wo $HARDWARE genutzt sind überschaubar, ist wirklich größtenteils nur `_switch` 

- für die Ubiquity Geräte ist die genauere Erkennung in apply_profile.code enthalten, da fällt nichts weg, das Script schreibt auch nichts in die Datei, wenn dort etwas anderes steht. 
- Für alles andere müssten doch die OpenWRT-Daten taugen? 

Das Problem ist, dass Meshkit dafür anpassen glaub ich 10x mehr Aufwand ist, als das Script als Notlösung zu behalten und mit sed die Namen anpassen. Eigene Änderungen können ja immer noch in apply_profile.code 

Hier ist die Liste: http://git.openwrt.org/?p=openwrt.git;a=blob;f=target/linux/ar71xx/base-files/lib/ar71xx.sh;h=df77040bdd8fac3494543b3f9bf72e1e577e6a02;hb=HEAD für die anderen Architekturen gibt es jeweils auch ein Script in base-files/lib z.B. ramips.sh

andibraeu, 2015-11-26:
ich bin für so nah wie möglich am OpenWRT.. 

evtl unterscheiden sich ubiquity-Geräte gar nicht so stark von Revision zu Revision wie die TP-Link-Rouer? Ich mein, die brauchen genauso das korrekte Modell, um den Switch richtig zu konfigurieren...

Ich plädiere dafür, die OpenWRTErkennung zu nutzen und ```_switch``` entsprechend anzupassen

glycoknob, 2015-11-27:
Hab nochmal in apply_profile.code geschaut... da ist doch alles drinnen und die Namen sollten richtig sein. 

Der Grund warum das nicht klappt ist folgender:

Wir haben  `CONFIG_KERNEL_PRINTK` für Kernel-Messages ausgeschaltet, https://github.com/weimarnetz/builder/blob/master/openwrt-config/config_shrink.txt

Also klappt das hier nicht mehr: 
https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-build/apply_profile.code#L150

Bleibt eigentlich nur das wieder anzuknipsen

bittorf, 2015-11-27:
Die "guten" Namen stammen von mir. ich habe schon ausführlich erklärt warum OpenWrt es falsch macht. Aus ihrer Sicht ist es aber kein Bug, daher wird es auch nie gefixt werden. OpenWrt findet es gut möglichst viele gleichartige Geraete unter einem Namen zusammenzufassen um die Makefiles kleinzuhalten. Ist aber aus Usersicht (der sich ein image fuer *sein* Modell runterladen will) kagge. Ich bin erstmal fuer einen Wrapper, bis das entschieden ist. Tragt mal die Namen zusammen, die es betrifft.
(das sind ja laut christian nur 2-3 geraete die wir unterstuetzen)
das "TP LINK" vs. "TP Link" ist leicht zu loesen, andere Sachen sind aber handarbeit.

gedanke:
man koennte jetzt nur fuer die switch-sache die namen aus "/tmp/sysinfo/model" nehmen. ich muss mal pruefen seit wann es diese datei gibt, weil ich noch nen paar aeltere modelle in der landschaft habe.

bittorf, 2015-11-27:
@glycoknob du hast recht, der groesste Nutzer von $HARDWARE ist die switch geschichte und eben das monitoring.

andibraeu, 2015-11-27:
Hm, die Liste in tmp hat doch wenig mit dem Buildprozess zu tun und ist detaillierter - 1043 vs 1043 v1. Die sind auch nicht alle zu einem Typ wie im Build beim Beispiel bullet. 

Printk ist aus wegen Speicher sparen. Lasst uns einfach mal prüfen, welche Geräte bei uns vorkommen. Ich denke fast, openwrt macht das an der Stelle richtig. Die setzen damit ja auch die Switcheinstellungen.

bittorf, 2015-11-27:
@andibraeu wie gesagt: fuer die switcheinstellungen ist das vollkommen ok. aber nicht fuer das gesamtkonzept. (und dann will ich im monitoring schon stehen haben, ob das eine Picostation2 oder Bullet Titanium M5 ist - fuer OpenWrt ist es einfach das gleiche, weil gleiche firmware...mähhh)

andibraeu, 2015-11-27:
dann spricht ja nichts dagegen das in _switch zu nutzen und evtl. deine Erkennung an einer 2. Stelle abzulegen und beides im Monitoring zu verwenden

bittorf, 2015-11-27:
@andibraeu OpenWrt ist da nicht genauer:
bei mir heisst es momentan

TP-LINK TL-WR1043ND
TP-LINK TL-WR1043ND v2

und bei OpenWrt:
TP-Link TL-WR1043N/ND v1

also zwei verschiedene Modelle in einem Namen.
Das ist doch schmutz. Anderes Beispiel:

Targa WR-500-VoIP und
Speedport W500V

sind zwei voellig unterschiedliche Modelle. Unter OpenWrt haben die die gleiche Firmware und heissen: SPW500V - geil, oder?

glycoknob, 2015-11-27:
@andibraeu: ohne CONFIG_PRINTK kein dmesg und keine Erkennung :/  - daher auch die Probleme der Code macht schons genau das was der von kalua macht, nur bis zum apply_profile.code wird /tmp/sysinfo/model genommen... 

Ich wäre dafür CONFIG_PRINTK einfach wieder einzubauen, Logfiles sind schon schön

glycoknob, 2015-11-27:
Alternative wäre eben sich komplett auf OpenWRT zu verlassen und _switch fragt nur uci ab, weil ohne dmesg haben wir keine weiteren Infos.

andibraeu, 2015-11-27:
für _switch reicht es doch erstmal vollkommen aus, sich auf OpenWRT zu verlassen

print_k kostet halt relativ viel Speicher und wir sollten uns schon ein paar Reserven auf den Geräten lassen

bittorf, 2015-11-27:
ok, ich werde einen mini-wrapper fuer "switch" bauen. tragt mal bitte die liste der geraete zusammen

glycoknob, 2015-11-27:
Das Problem ist gerade: Mit CONFIG_PRINK -> alles läuft, weil apply_profile.code die richtigen Namen nutzt... durch größtenteils greppen in dmesg 

ohne CONFIG_PRINTK -> wir haben überall für $HARDWARE `/tmp/sysinfo/model` 

Die Frage wäre jetzt wie man das schön löst, ich wäre ja fast dafür erstmal CONFIG_PRINTK anzuschalten und dann läuft erstmal alles ohne Änderungen. 

Ich teste mal wie groß der Unterschied da ist und ob wir nicht noch was sparen können, wir haben glaub ich auch zu viele tc Pakete, die wir alle gar nicht benötigen.

andibraeu, 2015-11-27:
@glycoknob ehrlich gesagt möchte ich an dieser Stelle gerade nicht so viel umwerfen.

andibraeu, 2015-11-27:
Eine Liste, eine Liste! Hab mal alle eingetragen, die ich kenne

| Kalua                           | OpenWRT                  |
|---------------------------------|--------------------------|
| 4G Systems MTX-1 Board          |                          |
| ASUS WL-500g Premium            |                          |
| ASUS WL-500g Premium v2         |                          |
| ASUS WL-HDD 2.5                 |                          |
| Beagleboard                     |                          |
| Buffalo WHR-HP-G54              |                          |
| Buffalo WZR-HP-AG300H           |                          |
| Dell TrueMobile 2300            |                          |
| D-Link DIR-300-B1               |                          |
| D-Link DIR-615-E4               |                          |
| FritzBox 7170                   |                          |
| La Fonera 2.0N                  |                          |
| Linksys WRT54G/GS/GL            |                          |
| Mercury MAC1200R                |                          |
| Mikrotik Routerboard 532        |                          |
| Nexx WT3020A                    |                          |
| Nexx WT3020AD                   |                          |
| Nexx WT3020F                    |                          |
| Nexx WT3020H                    |                          |
| Pandaboard                      |                          |
| PC Engines ALIX.2               |                          |
| PC Engines WRAP                 |                          |
| Seagate GoFlex Home             |                          |
| Seagate GoFlex Net              |                          |
| Soekris net5501                 |                          |
| Speedport W500V                 |                          |
| Targa WR-500-VoIP               |                          |
| T-Mobile InternetBox TMD SB1-S  |                          |
| TP-LINK Archer C7               | TP-Link Archer C7 v1     |
| TP-LINK Archer C7 v2            |                          |
| TP-LINK TL-MR3020               | TP-Link TL-MR3020 v1     |
| TP-LINK TL-WDR3600              | TP-Link TL-WDR3600 v1    |
| TP-LINK TL-WDR4300              | TP-Link TL-WDR4300 v1    |
| TP-LINK TL-WDR4900 v1           |                          |
| TP-LINK TL-WDR7500              |                          |
| TP-LINK TL-WR1043ND             | TP-Link TL-WR1043N/ND v1 |
| TP-LINK TL-WR1043ND v2          |                          |
| TP-LINK TL-WR703N v1            |                          |
| TP-LINK TL-WR741ND v1           |                          |
| TP-LINK TL-WR741ND v2           |                          |
| TP-LINK TL-WR741ND v4           |                          |
| TP-LINK TL-WR841N/ND v1.5       |                         |
| TP-LINK TL-WR841N/ND v3         |                          |
| TP-LINK TL-WR841N/ND v5         |                          |
| TP-LINK TL-WR841N/ND v7         | TP-Link TL-WR841N/ND v7  |
| TP-LINK TL-WR841N/ND v8         | TP-Link TL-WR841N/ND v8  |
| TP-LINK TL-WR841N/ND v9         | TP-Link TL-WR841N/ND v9  |
|   | TP-Link TL-WR842N/ND v1 |
| TP-LINK TL-WR847N v8            |                          |
| TP-LINK TL-WR940N               |                          |
| TP-LINK TL-WR941ND v4           |                          |
| Ubiquiti Bullet2                |                          |
| Ubiquiti Bullet5                |                          |
| Ubiquiti Bullet M2              | Ubiquiti Bullet M        |
| Ubiquiti Bullet M5              |                          |
| Ubiquiti Nanostation2           |                          |
| Ubiquiti Nanostation5           |                          |
| Ubiquiti Nanostation Loco M2 XW |                          |
| Ubiquiti Nanostation Loco M5 XW |                          |
| Ubiquiti Nanostation M2         | Ubiquiti Nanostation M   |
| Ubiquiti Nanostation M2 XW      |                          |
| Ubiquiti Nanostation M5         |                          |
| Ubiquiti Nanostation M5 XW      |                          |
| Ubiquiti Picostation2           |                          |
| Ubiquiti Picostation5           |                          |
| Ubiquiti Picostation M2         |                          |
| Ubiquiti Picostation M5         |                          |
| Ubiquiti WispStation5           |                          |
| UML                             |                          |
| x86_64                          |                          |

bittorf, 2015-11-27:
bitte teste, siehe:
https://github.com/weimarnetz/weimarnetz/pull/182
https://github.com/weimarnetz/weimarnetz/pull/183

andibraeu, 2015-11-27:
ja, funktioniert nun deutlich besser auf dem MR-Router

closed 2015-11-27 by andibraeu

bittorf, 2015-11-27:
es ist zugegebenermassen keine schoene loesung und ich denke darueber nach.