weimarnetz/weimarnetz #172: Firmware Update versucht alle 10s Infos zu holen, auch wenn keine da sind.

glycoknob, 2015-11-26:
Ist jetzt für x86, also nicht so relevant für das Release, aber vielleicht könnte man das auch lockerer angehen - vor allem wenn 404s kommen - ständig aller 10s zu prüfen bringt nichts, oder? Dann lieber maximal 5x am Tag im abstand von 5m oder so... 

```
Thu Nov 26 19:33:06 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:33:16 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:33:26 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:33:36 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:33:47 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:33:57 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:07 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:17 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:27 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:37 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:47 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:34:57 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:35:08 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:35:18 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:35:28 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Thu Nov 26 19:35:38 2015 daemon.info fff+ /bin/sh curl_it(): [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
```

bittorf, 2015-11-26:
das problem ist, das eben kein 404 kommt. (das wird naemlich abgefangen) sondern eine scheinbar kaputte datei...er probiert aber "nur" 100 mal.

andibraeu, 2015-11-26:
doch, da kommt ein 404:

```
curl -IL http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt
HTTP/1.1 404 Not Found
Access-Control-Allow-Origin: *
Content-type: text/html
Date: Thu, 26 Nov 2015 19:19:12 GMT
Server: lighttpd
```

glycoknob, 2015-11-26:
(ich seh grad mein KVM build iss nich so aktuell wie ich dachte.. mach grad ein make clean - ist vielleicht schon gefixt - wobei die Version auch höchstens 1-2 Wochen alt ist.

glycoknob, 2015-11-26:
Problem ist noch da - mit frischester GebrannteMandeln Revision von github:

```
root@ffweimar-hybrid-dhcp-989:~# _firmware_update_pmu
Nov 26 23:09:52 ' [up 3707s 7933952 bytes free] TERM:linux /bin/ash [daemon.debug] curl_it() max 15s, 65 bytes, wget  -qO -  'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:09:53 ' [up 3708s 7909376 bytes free] TERM:linux /bin/ash [daemon.info] curl_it() [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:09:58 ' [up 3713s 7766016 bytes free] TERM:linux /bin/ash [daemon.debug] curl_it() max 15s, 65 bytes, wget  -qO -  'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:09:59 ' [up 3714s 7786496 bytes free] TERM:linux /bin/ash [daemon.info] curl_it() [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:10:04 ' [up 3720s 7786496 bytes free] TERM:linux /bin/ash [daemon.debug] curl_it() max 15s, 65 bytes, wget  -qO -  'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:10:06 ' [up 3721s 7770112 bytes free] TERM:linux /bin/ash [daemon.info] curl_it() [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:10:11 ' [up 3726s 7815168 bytes free] TERM:linux /bin/ash [daemon.debug] curl_it() max 15s, 65 bytes, wget  -qO -  'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
Nov 26 23:10:12 ' [up 3727s 7868416 bytes free] TERM:linux /bin/ash [daemon.info] curl_it() [ERR] returning 1 after fetching 'http://weimarnetz.de/freifunk/firmware/x86/images/stable/info.txt'
```

andibraeu, 2015-11-26:
Ich bin dafür, das Abfrageintervall einfach zu reduzieren

PR #180 (2015-11-27): kalua: curl: it() workaround a race condition when having a 404 with …
…a large response - where error output is very late - fixes #172
commit 8191a360 (2015-11-27): kalua: curl: it() workaround a race condition when having a 404 with a large response - where error output is very late - fixes #172
PR #178 (2015-11-27): kalua: curl: it() workaround a race condition when having a 404 with …
…a large response - where error output is very late - fixes #172
closed 2015-11-27 by andibraeu