weimarnetz/weimarnetz #174: Usability Weimarnetz Konfigurationsseite

glycoknob, 2015-11-26:
Bin grad am testen (läuft soweit prima!), aber eine Sache ist wirklich verwirrend: 

Wenn ich die VPN-Einstellungen will, dann muss ich erst ganz unten die Änderungen aktivieren und _dann_ erst auf den Knopf drücken... 

Siehe diesen hochgradig professionellen Screenshot: 

[image]

Könnten wir die VPN-Änderungen da auch mit dem Apply erledigen? 

Dann bekomme ich nach einige Zeit des nichts tuns Gateway-Timeouts auf der Seite - bin mir nicht so sicher woran das liegt..  laut Chrome wird da ein POST gesendet.. ich weiß aber nicht wo der herkommt.

andibraeu, 2015-11-26:
wenns einfach wär, hätte ich das da eingebaut :-) Man müsste glaub ich ein init-script bauen und könnte darüber dann vpn auch mit apply neu starten

glycoknob, 2015-11-29:
okay, das neuladen macht wohl [luci-reload](https://github.com/openwrt/luci/blob/master/modules/luci-base/root/sbin/luci-reload) zusammen mit `/etc/config/ucitrack` - entweder ein init-script oder dieses exec was da steht... da müssten wir was bauen können. gibt wohl auch noch hooks, die man dafür nutzen könnte: https://github.com/elationfoundation/luci_tutorials/blob/master/04-model-cbi.md

andibraeu, 2015-12-01:
hab mal an der richtigen Stelle ein Issue angelegt: https://github.com/weimarnetz/openwrt-packages/issues/21

closed 2015-12-01 by andibraeu