weimarnetz/weimarnetz #177: Besseres Default Passwort

glycoknob, 2015-11-27:
die Aufforderung das Passwort zu wechseln sollte dennoch bleiben, aber damit könnten wir zumindest SSH-Bots oder vergessene Router mit Default-Passwort _etwas_ aus der ganz offensichtlichen Schusslinie nehmen - irgendwas leicht zu merkendendes, nicht zu kurzes fällt uns schon ein. 

Für 12 Zeichen muss jemand schon ein paar Tage die GPU brutzeln lassen, wenn es nur als Hash im Repository ist: http://blog.codinghorror.com/speed-hashing/ 

Sowas hier https://xkpasswd.net/s/ - 3 bis 4 leicht zu merkende Worte

bittorf, 2015-11-27:
fuer bots wuerde "weimarnetz" reichen. jemand mit genug insiderwissen wuerde auch andere kennen, oder?

andibraeu, 2015-11-27:
Man muss aber alle Funktionen anpassen, die auf das Standardpasswort hinweisen?

Ist vielleicht kein pw besser?

bittorf, 2015-11-27:
ich denke auch, das dies unnoetig ist. die hinweise zum setzen reichen doch. wer das ignoriert will es nicht anders.

glycoknob, 2015-11-27:
@bittorf: stimmt. Ist ja auch im Meshkit, wäre z.B: eine Idee.

andibraeu, 2015-11-27:
kann das wieder zu?

bittorf, 2015-11-27:
ich denke ja.

closed 2015-11-27 by bittorf