weimarnetz/weimarnetz #184: luci-theme-material evaluieren

glycoknob, 2015-11-27:
schönes neues Theme basierend auf http://www.getmdl.io/ 
https://github.com/LuttyYang/luci-theme-material

Ich teste mal wie viel Platz das frisst, würd das gerne per Default haben wollen!

glycoknob, 2015-11-27:
```
-rw-r--r--  1 mt mt  14K Nov 27 18:16 luci-theme-bootstrap_git-15.330.80595-c123fa8-1_all.ipk
-rw-r--r--  1 mt mt  50K Nov 27 18:16 luci-theme-material_git-15.330.80595-c123fa8-1_all.ipk
``` 

Mal sehen ob sich das lohnt.

glycoknob, 2015-11-27:
fazit nach 5 minuten: nope, nope, nope...

closed 2015-11-27 by glycoknob

andibraeu, 2015-11-27:
ganz schön groß... die haben da auch jquery eingebaut...

andibraeu, 2015-11-27:
ist das im luci-feed drin? man könnte es als optionales paket bauen und anbieten

glycoknob, 2015-11-27:
ist zumindest im trunk, aber ist momentan noch mehr kaputt als sinnvoll.