weimarnetz/weimarnetz #185: root: /etc/kalua_init: [ERR] iproute2 not installed: ip

glycoknob, 2015-11-28:
Nach einen sysupgrade (ohne -n): 

ip ist aber da: 

```
root@netbox:~ ip 
root: /etc/kalua_init: [ERR] iproute2 not installed: ip 
root@netbox:~ which ip 
/usr/bin/ip
root@netbox:~ /usr/bin/ip 
Usage: ip [ OPTIONS ] OBJECT { COMMAND | help }
       ip [ -force ] -batch filename
where  OBJECT := { link | addr | addrlabel | route | rule | neigh | ntable |
                   tunnel | tuntap | maddr | mroute | mrule | monitor | xfrm |
                   netns | l2tp | fou | tcp_metrics | token | netconf }
       OPTIONS := { -V[ersion] | -s[tatistics] | -d[etails] | -r[esolve] |
                    -h[uman-readable] | -iec |
                    -f[amily] { inet | inet6 | ipx | dnet | bridge | link } |
                    -4 | -6 | -I | -D | -B | -0 |
                    -l[oops] { maximum-addr-flush-attempts } |
                    -o[neline] | -t[imestamp] | -ts[hort] | -b[atch] [filename] |
                    -rc[vbuf] [size] | -n[etns] name | -a[ll] }
```

glycoknob, 2015-11-28:
Problem ist wohl hier: https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/kalua_init.user_fake_bad_iproute2#L4

bittorf, 2015-11-28:
ok, das wir ein dummer fehler...gefixt.

closed 2015-11-28 by bittorf

glycoknob, 2015-11-28:
danke fürs fixen!

commit 215b34ac (2015-11-28): kalua: init/loader: sync with upstream - do the 'iproute2'-check more generic - fixes #185
PR #188 (2015-11-28): kalua: init/loader: sync with upstream - do the 'iproute2'-check more…
… generic - fixes #185