weimarnetz/weimarnetz #186: /usr/lib/lua/luci/controller/weimarnetz/splash.lua: bad header in precompiled chunk

glycoknob, 2015-11-28:
Ebenfalls nach sysupgrade - und build mit `./build_release.sh meta.ffweimar-4MBtrunk`

Auf x86 läufts.. kann das jemand reproduzieren? 

```
/usr/lib/lua/luci/ccache.lua:70: error loading module 'luci.controller.weimarnetz.splash' from file '/usr/lib/lua/luci/controller/weimarnetz/splash.lua':
	/usr/lib/lua/luci/controller/weimarnetz/splash.lua: bad header in precompiled chunk
stack traceback:
	[C]: in function 'loader'
	/usr/lib/lua/luci/ccache.lua:70: in function </usr/lib/lua/luci/ccache.lua:61>
	[C]: in function 'require'
	/usr/lib/lua/luci/dispatcher.lua:509: in function 'createindex'
	/usr/lib/lua/luci/dispatcher.lua:536: in function 'createtree'
	/usr/lib/lua/luci/dispatcher.lua:216: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

andibraeu, 2015-11-28:
Meine Beobachtung bisher: in den Minuten nach dem flashen kommt das vor, danach nicht mehr. 

Geht der der Fehler bei dir irgendwann weg?

glycoknob, 2015-11-28:
Ging jetzt nicht weg - zumindest nicht in 20-30m - ersetze ich die kompilierte Datei durch den Source, dann klappt es.

Ich glaube die make-Zeile hier könnte man rausnehmen, dann wird nicht kompiliert? 
 https://github.com/weimarnetz/openwrt-packages/blob/master/utils/luci-weimar-splash/Makefile#L31 

Ist vielleicht auch ein Bug in trunk..

andibraeu, 2015-11-29:
ja, jetzt wo du trunk sagst, da hatte ich auch immer mal bei zwischenversionen probleme mit lua

glycoknob, 2015-11-29:
Ich teste heute nochmal mit ChaosCalmer - Trunk scheint da an enigen stellen Probleme zuhaben.

glycoknob, 2015-11-29:
Problem tritt nur im trunk auf - hier auch die Frage: Schließen?

andibraeu, 2015-11-29:
schreib vielleicht eher mal an die luci-liste, oder kannst du daraus einen qualifizierten Bug machen?

glycoknob, 2015-12-10:
jetzt auch unter CC :/

andibraeu, 2015-12-11:
das scheint mir aber kein Problem auf unserer Seite zu sein, sondern eher etwas von luci/lua

andibraeu, 2015-12-11:
wie kompilierst du eigentlich? vielleicht kommen die Fehler auch daher. Versuch es doch bitte nochmal auf einer "sauberen" Umgebung

glycoknob, 2015-12-11:
Ich versuch mal nen Bug-Report zu erstellen, bau mit dem builder - also müsste alles in Ordnung sein, kann sein, dass es am neuen Ubuntu oder so liegt.

glycoknob, 2015-12-11:
Mit Ubuntu 14.04 und CC läuft alles.

andibraeu, 2015-12-12:
hab heute auch nochmal durchkompiliert und die Seite funktioniert... Ich mach erstmal zu

closed 2015-12-12 by andibraeu