weimarnetz/weimarnetz #187: Test auf schwaches Passwort (Admin) funktioniert nicht.

glycoknob, 2015-11-28:
Passwort ist gleich (admin), aber Hash ist immer unterschiedlich - gibt noch ein Salt wie es aussieht. 

Der Test in https://github.com/weimarnetz/weimarnetz/blob/master/openwrt-addons/etc/profile.d/kalua.sh#L16 funktioniert so nicht. 

```
root@netbox:~ cat /etc/shadow
root:$1$FLuGYCpI$rxucSh633a9SCLqdAoFXK/:16767:0:99999:7:::
daemon:*:0:0:99999:7:::
ftp:*:0:0:99999:7:::
network:*:0:0:99999:7:::
nobody:*:0:0:99999:7:::
root@netbox:~ passwd
Changing password for root
New password: 
Bad password: too short
Retype password: 
Password for root changed by root
root@netbox:~ cat /etc/shadow
root:$1$s4Wags/y$M5Z5sMF/PGwD3djsaeBSq0:16767:0:99999:7:::
daemon:*:0:0:99999:7:::
ftp:*:0:0:99999:7:::
network:*:0:0:99999:7:::
nobody:*:0:0:99999:7:::
```

bittorf, 2015-11-28:
danke, kuemmere mich darum.

andibraeu, 2015-12-08:
gibts hier ideen?

glycoknob, 2015-12-10:
zum vergleichen müssten wir den hash mit dem salt selbst berechnen...  das hier könnte klappen: http://comments.gmane.org/gmane.linux.busybox/3664 

wäre allerdings ne busybox config-option, platzmäßig kostet das wohl nichts - ich würd mal testen obs klappt

glycoknob, 2015-12-10:
mkpasswd klappt. 
awk (sorry :) 

    salt=$(awk -F"$" '/^root/ { print $3}' < /etc/shadow)
    pass=admin
    weakhash=$(mkpasswd -5 -S $salt $pass) 
    hash=$(cat /etc/shadow | awk -F":" '/^root/ { print $2}')
    [ "$weakhash" = "$hash" ]] && echo "change passwd plz!"

bittorf, 2015-12-11:
ja, ist klar. wir muessen dann halt mkpasswd mit ins image nehmen.
jetzt stellt sich die frage: machen? oder das feature ("schwaches passwort erkennen") einfach rausstreichen?

andibraeu, 2015-12-11:
ich würde das auf später verschieben

Wir können in der Doku nochmal gesondert darauf hinweisen

bittorf, 2015-12-11:
ok, ich werde trotzdem den code erstmal in profile.d drinlassen und spaeter anpassen auf mkpasswd (wenn installiert) - ticket erstmal offen lassen

andibraeu, 2015-12-11:
Hab das in #215 verschoben

closed 2015-12-11 by andibraeu