weimarnetz/weimarnetz #190: luci: A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:78: attempt to index local 'boardinfo' (a nil value)

glycoknob, 2015-11-28:
```
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute function dispatcher target for entry '/'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute function dispatcher target for entry '/freifunk'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/dispatcher.lua:452: Failed to execute template dispatcher target for entry '/freifunk/index'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: The called action terminated with an exception:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'freifunk/index'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/bootstrap/header'.
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:78: attempt to index local 'boardinfo' (a nil value)
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]: stack traceback:
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       [C]: in function 'assert'
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       /usr/lib/lua/luci/dispatcher.lua:452: in function 'dispatch'
Sun Nov 29 00:35:04 2015 daemon.err uhttpd[1693]:       /usr/lib/lua/luci/dispatcher.lua:141: in function </usr/lib/lua/luci/dispatcher.lua:140>
```

Code dafür ist hier: 
https://github.com/openwrt/luci/blob/222284578bb61c591ee7e3afabae2952682081f3/themes/luci-theme-bootstrap/luasrc/view/themes/bootstrap/header.htm#L14

ubus läuft allerdings: 

```
ubus call system board
{
        "kernel": "4.1.11",
        "hostname": "netbox",
        "system": "Atheros AR9341 rev 3",
        "model": "TP-Link TL-WR841N\/ND v8",
        "release": {
                "distribution": "OpenWrt (Weimarnetz edition)",
                "version": "Bleeding Edge",
                "revision": "r47665",
                "codename": "designated_driver",
                "target": "ar71xx\/generic",
                "description": "OpenWrt (Weimarnetz edition) Designated Driver r47665"
        }
}
```

Ist jetzt im trunk, aber irgendwas ist da kaputt :/

glycoknob, 2015-11-29:
ok, problem ist real - luci ändert den user auf `nobody:nogroup` und irgendwie klappt das dann mit dem ubus nicht mehr. 

wenn man das im controller von freifunk.lua abschaltet, dann klappt es.

glycoknob, 2015-11-29:
Problem tritt auch im trunk auf, mal sehen ob CC da besser läuft.

andibraeu, 2015-11-29:
welche seite rufst du da eigentlich auf?

glycoknob, 2015-11-29:
Eine von den Freifunk-Seiten - ich glaube überblick, wenn man eingeloggt ist, oder ubus mehr Rechte gibt, dann klappt es auch.

glycoknob, 2015-11-29:
Okay Problem tritt _nur_ im trunk auf - Schließen? Oder Bug Report senden?

bittorf, 2015-11-30:
bug senden -> issue aufmachen -> github/luci

glycoknob, 2015-12-09:
hab nen issue aufgemacht, gibt wohl gleich mehrere kaputte Sachen im trunk - das olsr modul nutzt eine alte api die jetzt nicht mehr zugänglich ist, hier fehlen wohl Rechte für die anonyme ubus-session über die boardinfo geholt wird, hab aber noch keine antwort bekommen. 

Die precompiled header Probleme muss ich nochmal testen, hab ich zu wenig Daten für ein Ticket.

glycoknob, 2015-12-09:
Das konkrete Problem war ein OpenWRT Bug: https://dev.openwrt.org/changeset/47816

closed 2015-12-09 by glycoknob