weimarnetz/weimarnetz #191: http://<routerip>/cgi-bin/luci//freifunk/olsr/routes crashed 32MB Router - riesige Routingtable

glycoknob, 2015-11-29:
Aufruf produziert reproduzierbar einen Reboot

Wieso haben wir bei nichtmal 200 Routern eigentlich so eine fette Routingtabelle? 

```
root@netbox:~ ip r | wc -l 
812
```

andibraeu, 2015-11-29:
weil da auch die routen in die anderen freifunknetze drin sind

glycoknob, 2015-11-29:
Ok, mal sehen warum es crasht - die Daten werden über jsoninfo olsr-Plugin geholt... kann eigentlich auch bei 800 Routen nicht so schlimm sein, wer weiß. Ich teste hier auch nochmal mit CC

glycoknob, 2015-11-29:
Tritt auch unter Chaos Calmer auf.

glycoknob, 2015-11-29:
olsrd jsoninfo läuft prima: 

    echo /status | nc 127.0.0.1 9090

Problem ist wohl irgendwo bei luci...

andibraeu, 2015-11-29:
das ist ein kandidat für einen bugreport bei luci

glycoknob, 2015-11-29:
Ja, mysterious - mit CONFIG_PRINTK und logread -f | nc meinrechner seh ich nur folgendes, beim Aufruf der Seite: 

```
Mon Nov 30 00:45:22 2015 daemon.err procd: Rebooting as procd has crashed
Mon Nov 30 00:45:22 2015 daemon.info procd: reboot
``` 

irgendwas seltsames passiert da...

glycoknob, 2015-11-30:
Voll das Rätsel... also wenn ich panic on oom ausschalte und manuell den oom killer starte _UND_ zram auschalte, dann bekomm ich auch sinnvolle Log Einträge 

mit zram ist irgendwas kaputt:

root@netbox:~ logread 
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found
Failed to find log object: Not found
Thu Jan  1 01:00:00 1970 kern.emerg kernel: 
root@netbox:~ logread 
Thu Jan  1 01:00:00 1970 kern.emerg kernel: 
root@netbox:~ logread 
Thu Jan  1 01:00:00 1970 kern.emerg kernel: 

ohne zram wird luci gekillt - wegen ungefähr 6MB RAM: 

```
Mon Nov 30 02:16:48 2015 kern.warn kernel: hostapd invoked oom-killer: gfp_mask=0x201da, order=0, oom_score_adj=0
Mon Nov 30 02:16:48 2015 kern.warn kernel: CPU: 0 PID: 1525 Comm: hostapd Not tainted 3.18.23 #1
Mon Nov 30 02:16:48 2015 kern.warn kernel: Stack : 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Nov 30 02:16:48 2015 kern.warn kernel:        00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Nov 30 02:16:48 2015 kern.warn kernel:        00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Nov 30 02:16:48 2015 kern.warn kernel:        00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Nov 30 02:16:48 2015 kern.warn kernel:        00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mon Nov 30 02:16:48 2015 kern.warn kernel:        ...
Mon Nov 30 02:16:48 2015 kern.warn kernel: Call Trace:[<80070b88>] 0x80070b88
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<80070b88>] 0x80070b88
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800c8f20>] 0x800c8f20
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800c9288>] 0x800c9288
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800c9a24>] 0x800c9a24
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800ccc44>] 0x800ccc44
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800cfa00>] 0x800cfa00
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800c5730>] 0x800c5730
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800c8330>] 0x800c8330
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800e2908>] 0x800e2908
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800e543c>] 0x800e543c
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<8021b684>] 0x8021b684
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800e5bb8>] 0x800e5bb8
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800751a0>] 0x800751a0
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800ab01c>] 0x800ab01c
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<8010d720>] 0x8010d720
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<8010e5b0>] 0x8010e5b0
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<800ae53c>] 0x800ae53c
Mon Nov 30 02:16:48 2015 kern.warn kernel: [<80060820>] 0x80060820
Mon Nov 30 02:16:48 2015 kern.warn kernel: 
Mon Nov 30 02:16:48 2015 kern.warn kernel: Mem-Info:
Mon Nov 30 02:16:48 2015 kern.warn kernel: Normal per-cpu:
Mon Nov 30 02:16:48 2015 kern.warn kernel: CPU    0: hi:    0, btch:   1 usd:   0
Mon Nov 30 02:16:48 2015 kern.warn kernel: active_anon:2032 inactive_anon:452 isolated_anon:0
Mon Nov 30 02:16:48 2015 kern.warn kernel:  active_file:46 inactive_file:67 isolated_file:32
Mon Nov 30 02:16:48 2015 kern.warn kernel:  unevictable:0 dirty:0 writeback:0 unstable:0
Mon Nov 30 02:16:48 2015 kern.warn kernel:  free:169 slab_reclaimable:222 slab_unreclaimable:2526
Mon Nov 30 02:16:48 2015 kern.warn kernel:  mapped:1 shmem:167 pagetables:112 bounce:0
Mon Nov 30 02:16:48 2015 kern.warn kernel:  free_cma:0
Mon Nov 30 02:16:48 2015 kern.warn kernel: Normal free:676kB min:676kB low:844kB high:1012kB active_anon:8128kB inactive_anon:1808kB active_file:184kB inactive_file:268kB unevictable:0kB isolated(anon):0kB isolated(file):128kB present:32768kB managed:28896kB mlocked:0kB dirty:0kB writeback:0kB mapped:4kB shmem:668kB slab_reclaimable:888kB slab_unreclaimable:10104kB kernel_stack:392kB pagetables:448kB unstable:0kB bounce:0kB free_cma:0kB writeback_tmp:0kB pages_scanned:20004 all_unreclaimable? yes
Mon Nov 30 02:16:48 2015 kern.warn kernel: lowmem_reserve[]: 0 0
Mon Nov 30 02:16:48 2015 kern.warn kernel: Normal: 1*4kB (R) 0*8kB 10*16kB (R) 4*32kB (R) 2*64kB (R) 0*128kB 1*256kB (R) 0*512kB 0*1024kB 0*2048kB 0*4096kB = 676kB
Mon Nov 30 02:16:48 2015 kern.warn kernel: 312 total pagecache pages
Mon Nov 30 02:16:48 2015 kern.warn kernel: 0 pages in swap cache
Mon Nov 30 02:16:48 2015 kern.warn kernel: Swap cache stats: add 362, delete 362, find 4/14
Mon Nov 30 02:16:48 2015 kern.warn kernel: Free swap  = 0kB
Mon Nov 30 02:16:48 2015 kern.warn kernel: Total swap = 0kB
Mon Nov 30 02:16:48 2015 kern.warn kernel: 8192 pages RAM
Mon Nov 30 02:16:48 2015 kern.warn kernel: 0 pages HighMem/MovableOnly
Mon Nov 30 02:16:48 2015 kern.warn kernel: 968 pages reserved
Mon Nov 30 02:16:48 2015 kern.info kernel: [ pid ]   uid  tgid total_vm      rss nr_ptes swapents oom_score_adj name
Mon Nov 30 02:16:48 2015 kern.info kernel: [  411]     0   411      186       17       4        0             0 ubusd
Mon Nov 30 02:16:48 2015 kern.info kernel: [  412]     0   412      155       12       4        0             0 askfirst
Mon Nov 30 02:16:48 2015 kern.info kernel: [ 1111]     0  1111      223       36       3        0             0 logd
[... 124 lines cut ...]
Mon Nov 30 02:16:59 2015 kern.warn kernel: [<8009c00c>] 0x8009c00c
Mon Nov 30 02:16:59 2015 kern.warn kernel: [<800e9fb8>] 0x800e9fb8
Mon Nov 30 02:16:59 2015 kern.warn kernel: [<80060820>] 0x80060820
[... cut ...]
Mon Nov 30 02:16:59 2015 kern.warn kernel: 255 total pagecache pages
Mon Nov 30 02:16:59 2015 kern.warn kernel: 0 pages in swap cache
Mon Nov 30 02:16:59 2015 kern.warn kernel: Swap cache stats: add 362, delete 362, find 4/14
Mon Nov 30 02:16:59 2015 kern.warn kernel: Free swap  = 0kB
Mon Nov 30 02:16:59 2015 kern.warn kernel: Total swap = 0kB
Mon Nov 30 02:16:59 2015 kern.warn kernel: 8192 pages RAM
Mon Nov 30 02:16:59 2015 kern.warn kernel: 0 pages HighMem/MovableOnly
Mon Nov 30 02:16:59 2015 kern.warn kernel: 968 pages reserved
```

glycoknob, 2015-11-30:
Stellt sich raus wir haben tatsächlich von 32MB nur knapp 6MB freies RAM zur Verfügung... 

zwischen 9 und 12MB ist Kernel...  /proc/meminfo müsste halbwegs recht haben: https://git.kernel.org/cgit/linux/kernel/git/torvalds/linux.git/commit/?id=34e431b0ae398fc54ea69ff85ec700722c9da773

bittorf, 2015-11-30:
schönes ding, danke. habe mit https://github.com/weimarnetz/weimarnetz/pull/192 mal support für "MemAvailable" drinne.

bittorf, 2015-11-30:
das luci crasht beim abholen von so wenig routen (alles unter 1 million ist wenig), sollte gemeldet werden.

glycoknob, 2015-11-30:
Ja, seltsam - der Lua-Code macht auch nichts außergewöhnliches, eigentlich nur die JSON-Daten durchreichen - Gibt wohl einen Debugger der Memory-Verbrauch misst: https://github.com/elationfoundation/luci_tutorials/blob/master/06-debugging.md 

Muss das mal unter KVM testen, dass muss ja eh noch in den Builder...

glycoknob, 2015-12-04:
> alles unter 1 million ist wenig

Okay, musste mal eben procrastinieren, kann man tatsächlich rausbekommen: 

```
/proc/slabinfo: 
ip_fib_trie          821    867     80   51
ip_fib_alias         830    840     72   56
``` 
Da müssten die Routern gespeichert werden

80Byte+72Byte=152Byte/Route 

Für 1Mio Routen bräuchte man etwa 150Mbyte nur für den Platz im Kernel :)

bittorf, 2015-12-04:
@glycoknob das ist fuer einen minirouter natuerlich viel, aber auch die sollten sich mit 100.000 routen nicht schwertun. (und soviel haben wir lange nicht, eher 1000 = 150 kilobyte)

glycoknob, 2016-01-09:
Aus dem Luci Memory Tracer werd ich noch nicht so recht klug... der sagt etwa 3MB - allerdings traced der erst recht spät, wenn die View (also die Seite) gerendert wird... 

Während die Seite noch lädt wird noch ein 2. Request via Ajax abgefeuert -> 2x olsr Routes - das dürfte die 32MB Kisten crashen - siehe hier Chrome-Netzwerktools - die erste blaue Leiste lädt noch, während unten schon der 2. Request reinkommt: 

[image]

Das ist es aber auch nicht, bekomme auch keine Antwort ohne JavaScript - mal weiter testen, vielleicht zickt auch zram rum.

bittorf, 2016-01-09:
luci gehört weg 8-)

glycoknob, 2016-08-12:
Lässt sich nix machen, luci2, rpcd etc.pp helfen da.

closed 2016-08-12 by glycoknob