weimarnetz/weimarnetz #194: Splashpage überarbeiten

andibraeu, 2015-12-01:
Aktuelle aus Wiki entfernen

storchi, 2015-12-02:
woher sollen dann "wichtige" infos für die splashpage kommen?

bittorf, 2015-12-02:
momentan ist es doch so, das die "neuigkeiten" vom wiki runtergeparst werden, oder? koennte man das nicht vom server selbst in ein JSON werfen lassen und nur das abholen?

andibraeu, 2015-12-02:
Von da: http://www.weimarnetz.de/inc/feed/feed.php

Das liest unsere fb-Posts aus. Wir könnten auch die Mailingliste einbinden. Das ersetzt dann 2 Punkte

bittorf, 2015-12-03:
komisch, für mich *sieht* das wie die mailingliste aus. ein convert nach JSON koennte ich machen, soll ich?

andibraeu, 2015-12-03:
> komisch, für mich sieht das wie die mailingliste aus. ein convert nach JSON koennte ich machen, soll ich?
> 
Stimmt, Das ist unser rss Sammler, der beides einsammelt. 

Converter brauchst du nicht bauen. Das Ding kann in der neuen version auch json. Ich bau das heute Abend ein.

andibraeu, 2015-12-03:
so, jetzt aber :-)

Facebook:
http://weimarnetz.de/inc/feed/feed.php?source=fbweimarnetz&format=json

ML:
http://weimarnetz.de/inc/feed/feed.php?source=mlweimarnetz&format=json

beides:
http://weimarnetz.de/inc/feed/feed.php?format=json

andibraeu, 2015-12-03:
achso, und jsonp geht auch: http://weimarnetz.de/inc/feed/feed.php?source=fbweimarnetz&format=json&callback=function

andibraeu, 2015-12-03:
erledigt: https://github.com/weimarnetz/openwrt-packages/commit/5aba6563bdf4746cb4372e6ebc4a81faf58d3868

Läuft auf dem MR-Router und kann per opkg update installiert werden und wird ins meshkit gebacken

bittorf, 2015-12-04:
wunderbar.

kaese77, 2015-12-04:
> erledigt: weimarnetz/openwrt-packages@5aba656

1 mit Bienchen!

closed 2015-12-04 by andibraeu