weimarnetz/weimarnetz #195: /tmp/messages

glycoknob, 2015-12-04:
tmpfs füllt sich langsam mit /tmp/messages - mal schneller oder langsamer...  dürfte bei Routern die häufig benutzt werden schnell ein paar MB werden... genügt es nicht den logbuffer von 16kb etwas hochzustellen? Dann ist garantiert, dass die länger der Logfiles begrenzt ist.

bittorf, 2015-12-04:
es wird sowieso nur ein "circular buffer" mit 16k genommen. die datei /tmp/messages kommt daher, das nicht in in den permanent-storage geschrieben werden kann, weil voll. (z.b. logprio 'alert'). da dort aber sowieso nur wenige meldungen landen und die router sowieso jeden tag neustarten und das sowieso super (per zram) komprimiert werden kann ist das nicht so wild. sollte die datei schnell wachsen, hat man ganz andere sorgen: was steht denn da so wichtiges drinn bei dir?

glycoknob, 2015-12-05:
Das ist glaub ich auch kein größeres Problem, bin mir bei der "wo ist der verdammte RAM hin?" Suche drüber gestolpert. 

Zumindest in der Weimarnetz Firmware wächst das schon auf fast 500-600KB auch wenn der Router unbenutzt ist und da wird auch nichts komprimiert oder komprimiert tmpfs jetzt? tmps via zram mit ext4 haben wir nicht - bringt das was? 

```
root@netbox:/tmp ls -l messages 
-rw-r--r--    1 root     root        440256 Dec  5 22:33 messages
```

```
Dec 05 22:19:04 ' [up 65617s 3878912 bytes free] daemon.info fff+ /bin/sh watch_vpn() checking /tmp/OLSR/ALL for ip: 10\.63\.188\.57
Dec 05 22:20:04 ' [up 65677s 3829760 bytes free] daemon.info fff+ /bin/sh watch_vpn() checking /tmp/OLSR/ALL for ip: 10\.63\.188\.57
Dec 05 22:21:04 ' [up 65737s 3514368 bytes free] daemon.info fff+ /bin/sh watch_vpn() checking /tmp/OLSR/ALL for ip: 10\.63\.188\.57
```

Ist aber wohl wirklich eher egal. Soll ich schließen?

bittorf, 2015-12-06:
nee, das ist nicht egal: dort sollten NUR messages mit prio "alert" landen. warum ist da "info" drinne? ich pruefe das mal genauer...

bittorf, 2015-12-06:
fehler gefunden. ich werde die "log()" function mal backporten/updaten...

bittorf, 2015-12-06:
ok, bitte patch anschauen und vereinigen - dann schliessen

andibraeu, 2015-12-06:
yo

closed 2015-12-06 by andibraeu

glycoknob, 2015-12-06:
danke für den bugfix!

bittorf, 2015-12-07:
@glycoknob danke fuer die meldung!