weimarnetz/weimarnetz #197: Unnötige Pakete und Module aus Image rauswerfen

glycoknob, 2015-12-07:
Wir haben nicht so viel Platz auf den 4MB-Routern und könnte noch einige Bytes rauskratzen 

Konkret ist die tc konfiguration wohl zu bloated: 
https://github.com/weimarnetz/builder/blob/master/openwrt-config/config_tc.txt

    CONFIG_PACKAGE_kmod-sched-core=y
    CONFIG_PACKAGE_tc=y
    CONFIG_PACKAGE_kmod-lib-textsearch=y
    CONFIG_PACKAGE_kmod-ipt-compat-xtables=y

brauchen wir

    CONFIG_PACKAGE_kmod-ipt-condition=y
    CONFIG_PACKAGE_kmod-ipt-extra=y
    CONFIG_PACKAGE_kmod-ipt-filter=y

die brauchen wir nicht oder? siehe https://wiki.openwrt.org/doc/howto/netfilter 

    CONFIG_PACKAGE_kmod-ipt-iface=y

Das hat nichts mit dem Definieren des Interfaces zu tun sondern kann den Zustand des Interfaces abfragen.. brauchen wir das? Wir nutzen es nicht.  

    CONFIG_PACKAGE_kmod-ipt-u32=y

Wir nutzen nur `tc filter u32` müsste davon völlig unabhängig sein, oder? 

    CONFIG_PACKAGE_kmod-netem=y

Das kann definitiv weg. 

    CONFIG_PACKAGE_kmod-sched=y

Brauchen wir für die `prio qdisc` 

    CONFIG_PACKAGE_iptables-mod-condition=y
    CONFIG_PACKAGE_iptables-mod-extra=y
    CONFIG_PACKAGE_iptables-mod-filter=y
    CONFIG_PACKAGE_iptables-mod-iface=y
    CONFIG_PACKAGE_iptables-mod-u32=y

Brauchen wir nicht... 

Ich würde mal testen und dann ein Ziehbegehren erstellen.

glycoknob, 2015-12-08:
Less is more: https://github.com/weimarnetz/builder/pull/14

glycoknob, 2015-12-09:
Bei mir läuft das alles, die tc regeln werden erstellt und es gibt keine iptables Fehler - kmod-sched und kmod-sched-core noch kleiner machen bringt nicht viel, das sind unkomprimiert nur etwa 60kb

glycoknob, 2015-12-09:
ich stelle das mal zur diskussion: https://github.com/weimarnetz/builder/pull/15 

genaue zahlen kann ich gleich nachliefern.

andibraeu, 2015-12-09:
haste das alles mit CC kompiliert? Und auf welchen routern läuft das schon?

glycoknob, 2015-12-09:
der tc krams läuft bei meinem router hier ohne probleme, olsr prio läuft auch - wie groß der unterschied wegen den strip_symbols ist weiß ich gleich, wenn mein rechner das durchgekaut hat.

andibraeu, 2015-12-09:
aber es ist auch Chaos Calmer?

glycoknob, 2015-12-09:
@andibraeu: In der Theorie: Ja, ich teste gleich nochmal.

glycoknob, 2015-12-10:
hier die zahlen für CC: https://github.com/weimarnetz/builder/pull/15#issuecomment-163441767 

Ich würd mich freuen, wenn wir das anknipsen könnten und dafür printk und crashlogging drinlassen können. Zum einen sehen wir in logread/dmesg irgendwelche ooopses oder andere fehlermeldungen zur not über serielle, zum anderen klappt dann auch die Hardware-Erkennung von apply_profile.code - d.h. die Ubiquity-Hardware wird korrekt einsortiert.

glycoknob, 2015-12-10:
Ich glaub die Zeit zum eventuell extra Image bauen für benötigt kmods ist geringer als die Zeit die wir sinnlos irgendwas debuggen weil printk fehlt - immerhin soll das auf alle Router.

glycoknob, 2015-12-10:
Hmm, Router gebrickt :/ 

Kommando zurück.

glycoknob, 2015-12-10:
Hmm, liegt wohl nicht am Image sondern am Router, auf zweitem 841N läuft alles ohne Probleme.

glycoknob, 2015-12-10:
downgrade von trunk auf CC macht wohl probleme.

andibraeu, 2015-12-10:
Frage: Brauchen wir die Änderungen wirklich fürs nächste Release? Mir macht ein wenig Bauchschmerzen, dass wir das Basisimage, was bis jetzt ganz gut funktioniert noch einmal so stark umkrempeln.

andibraeu, 2015-12-10:
kannst du mir mal sysupgrade-images des Chaos-Calmer-Kompilats für diese Geräte irgendwo hochladen?

* 1043nd v1
* Bullet M
* 841n/nd v7 und v8
* wdr3600 v1
* wdr4300 v1

Dann installier ich mir die mal zu Hause

storchi, 2015-12-10:
Am 10.12.2015 um 08:48 schrieb Andi Bräu:
> Frage: Brauchen wir die Änderungen wirklich fürs nächste Release? Mir
> macht ein wenig Bauchschmerzen, dass wir das Basisimage, was bis jetzt
> ganz gut funktioniert noch einmal so stark umkrempeln.

ich finde die tippmannschen änderungen gut und würde die fuer post GM 
anwenden und testen usw.
cs

glycoknob, 2015-12-10:
CONFIG_EXPORT_SYMBOL würd ich erstmal sein lassen - hab jetzt 2 Router kaputt gespielt, weiß nicht woran das liegt - eventuell haben wir noch einen fiesen Bug woanders. 

Router verhält sich wie Switch nach Upgrade... die Änderungen mit "nur" tc laufen aber.

FreifunkUFO, 2015-12-10:
welche Pakete habt ihr denn derzeit noch dabei? Könnt Ihr mal bitte ein "opkg list-installed" posten oder so?

glycoknob, 2015-12-10:
@FreifunkFO: https://gist.github.com/glycoknob/5f489ac0693cbaeb3aaa

FreifunkUFO, 2015-12-10:
Ihr baut also gar keine zusätzlichen Pakete (zb Soundkarten..)?
Bitte poste mal ein "opkg list-installed" von einem tplink-841... 
da ist ja sicherlich kein USB installiert und selbst opkg braucht man da eigentlich auch gar nicht.

glycoknob, 2015-12-10:
Also Router ist nicht gebrickt, das sah gestern nur so aus - nach Reboot läuft alles, irgendwann leuchtet nur noch die Power-LED und er verhält sich wie ein Switch... mal debuggen. 

@FreifunkUFO USB haben wir drinnen u.A. wegen Ubiquity-Hardware die das wohl benötigt

Hier noch ein `opkg list` 

```
root@netbox:~ opkg list
base-files - 157-r47815
busybox - 1.23.2-1
community-profiles - 2
dnsmasq - 2.73-1
dropbear - 2015.67-1
freifunk-common - 1
fstools - 2015-09-15-cc63723d886fde2cd364a545b7cb05b70721b83f
hostapd-common - 2015-03-25-1
hostapd-mini - 2015-03-25-1
ip - 4.0.0-1
ip6tables - 1.4.21-1
iptables - 1.4.21-1
iptables-mod-ipopt - 1.4.21-1
iw - 3.17-1
jshn - 2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb
jsonfilter - 2014-06-19-cdc760c58077f44fc40adbbe41e1556a67c1b9a9
kalua - 1449705024-GebrannteMandeln
kernel - 3.18.23-1-1e025f64fa3e695fa5ae5e897297a459
kmod-ath - 3.18.23+2015-03-09-3
kmod-ath9k - 3.18.23+2015-03-09-3
kmod-ath9k-common - 3.18.23+2015-03-09-3
kmod-cfg80211 - 3.18.23+2015-03-09-3
kmod-crypto-aes - 3.18.23-1
kmod-crypto-arc4 - 3.18.23-1
kmod-crypto-core - 3.18.23-1
kmod-gpio-button-hotplug - 3.18.23-1
kmod-ip6tables - 3.18.23-1
kmod-ipt-conntrack - 3.18.23-1
kmod-ipt-core - 3.18.23-1
kmod-ipt-ipopt - 3.18.23-1
kmod-ipt-nat - 3.18.23-1
kmod-ipv6 - 3.18.23-1
kmod-ledtrig-usbdev - 3.18.23-1
kmod-lib-crc-ccitt - 3.18.23-1
kmod-lib-lz4 - 3.18.23-1
kmod-lib-lzo - 3.18.23-1
kmod-mac80211 - 3.18.23+2015-03-09-3
kmod-nf-conntrack - 3.18.23-1
kmod-nf-conntrack6 - 3.18.23-1
kmod-nf-ipt - 3.18.23-1
kmod-nf-ipt6 - 3.18.23-1
kmod-nf-nat - 3.18.23-1
kmod-nf-nathelper - 3.18.23-1
kmod-nls-base - 3.18.23-1
kmod-sched - 3.18.23-1
kmod-sched-core - 3.18.23-1
kmod-tun - 3.18.23-1
kmod-usb-core - 3.18.23-1
kmod-usb-ohci - 3.18.23-1
kmod-usb2 - 3.18.23-1
kmod-zram - 3.18.23-1
libblobmsg-json - 2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb
libc - 0.9.33.2-1
libgcc - 4.8-linaro-1
libip4tc - 1.4.21-1
libip6tc - 1.4.21-1
libiwinfo - 2015-06-01-ade8b1b299cbd5748db1acf80dd3e9f567938371
libiwinfo-lua - 2015-06-01-ade8b1b299cbd5748db1acf80dd3e9f567938371
libjson-c - 0.12-1
libjson-script - 2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb
liblua - 5.1.5-1
libnl-tiny - 0.1-4
libpthread - 0.9.33.2-1
libubox - 2015-11-08-10429bccd0dc5d204635e110a7a8fae7b80d16cb
libubus - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
libubus-lua - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
libuci - 2015-08-27.1-1
libuci-lua - 2015-08-27.1-1
libxtables - 1.4.21-1
lua - 5.1.5-1
luci-app-olsr - git-15.340.41458-ab31abd-1
luci-app-olsr-services - git-15.340.41458-ab31abd-1
luci-app-owm - 0.6.1
luci-app-owm-cmd - 0.6.1
luci-base - git-15.340.41458-ab31abd-1
luci-i18n-base-de - git-15.340.41458-ab31abd-1
luci-i18n-base-en - git-15.340.41458-ab31abd-1
luci-i18n-freifunk-de - git-15.340.41458-ab31abd-1
luci-i18n-freifunk-en - git-15.340.41458-ab31abd-1
luci-i18n-olsr-de - git-15.340.41458-ab31abd-1
luci-i18n-olsr-en - git-15.340.41458-ab31abd-1
luci-lib-httpclient - git-15.340.41458-ab31abd-1
luci-lib-ip - git-15.340.41458-ab31abd-1
luci-lib-json - git-15.340.41458-ab31abd-1
luci-lib-luaneightbl - git-15.340.41458-ab31abd-1
luci-lib-nixio - git-15.340.41458-ab31abd-1
luci-mod-admin-full - git-15.340.41458-ab31abd-1
luci-mod-freifunk - git-15.340.41458-ab31abd-1
luci-theme-bootstrap - git-15.340.41458-ab31abd-1
luci-weimar-splash - 0.4.4
mtd - 21
netifd - 2015-09-27-509ffb22475ebdd5291d510a098f996473951344
odhcp6c - 2015-07-13-024525798c5f6aba3af9b2ef7b3af2f3c14f1db8
odhcpd - 2015-11-19-01d3f9d64486ac1daa144848944e877e7f0cb762
olsrd - 0.9.0.3-1
olsrd-mod-arprefresh - 0.9.0.3-1
olsrd-mod-dyn-gw - 0.9.0.3-1
olsrd-mod-jsoninfo - 0.9.0.3-1
olsrd-mod-nameservice - 0.9.0.3-1
olsrd-mod-txtinfo - 0.9.0.3-1
olsrd-mod-watchdog - 0.9.0.3-1
opkg - 9c97d5ecd795709c8584e972bfdf3aee3a5b846d-9
procd - 2015-10-29-d5fddd91b966424bb63e943e789704d52382cc18
rpcd - 2015-01-10-f00890cd6eb47ad9bb5da0fb6c50aedc8406e7c5
swconfig - 10
tc - 4.0.0-1
uboot-envtools - 2014.10-2
ubox - 2015-11-22-c086167a0154745c677f8730a336ea9cf7d71031
ubus - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
ubusd - 2015-05-25-f361bfa5fcb2daadf3b160583ce665024f8d108e
uci - 2015-08-27.1-1
uhttpd - 2015-11-08-fe01ef3f52adae9da38ef47926cd50974af5d6b7
usign - 2015-05-08-cf8dcdb8a4e874c77f3e9a8e9b643e8c17b19131
vtun - 3.0.2-1
```

andibraeu, 2015-12-10:
@FreifunkUFO für die releases im meshkit bauen wir schon noch Pakete mit, die der geneigte User mit opkg installieren kann.

bittorf, 2015-12-10:
ich habe ja noch diesen auto-opkg-nachinstallierer bei mir drinne. spart nochmal 50k

glycoknob, 2015-12-10:
@bittorf wenn Andi nichts dagegen hat, wäre cool wenn das noch rein könnte.

andibraeu, 2015-12-10:
ich muss das neu komilieren und ins meshkit reinwerfen, werd ich die tage mal machen. Auch deshalb: http://git.openwrt.org/?p=15.05/openwrt.git;a=commit;h=6d9007ba67703dee913e643876a6b24bae9b16ef

andibraeu, 2015-12-11:
opkg werde ich nicht aus dem basisimage entfernen...

andibraeu, 2015-12-12:
die unnötigen tc-Pakete sind entfernt

closed 2015-12-12 by andibraeu

FreifunkUFO, 2016-12-30:
aufgeschoben ist nicht aufgehoben:
Bitte poste mal ein "opkg list-installed" von einem tplink-841...