weimarnetz/weimarnetz #198: Fehlermeldung nach Update: _sanitizer_shellsafe not found

andibraeu, 2015-12-07:
log läuft voll mit diesen Meldungen, danach rebootet sich der Router

```
Mon Dec  7 22:30:02 2015 user.emerg syslog: root: /etc/init.d/apply_profile.code: [OK] mode 'hybrid' is valid
Mon Dec  7 22:30:02 2015 user.notice root: /etc/init.d/apply_profile.code: [OK] mode 'hybrid' is valid
Mon Dec  7 22:30:02 2015 user.emerg syslog: root: /etc/init.d/apply_profile.code: password_set() ignore call, password already set
Mon Dec  7 22:30:02 2015 user.notice root: /etc/init.d/apply_profile.code: password_set() ignore call, password already set
Mon Dec  7 22:30:02 2015 user.emerg syslog: root: /etc/init.d/apply_profile.code: leaving hostname BraeuSys
Mon Dec  7 22:30:02 2015 user.notice root: /etc/init.d/apply_profile.code: leaving hostname BraeuSys
Mon Dec  7 22:30:02 2015 user.emerg syslog: root: /etc/init.d/apply_profile.code: [OK] building kalua-loader in ramdisk
Mon Dec  7 22:30:02 2015 user.notice root: /etc/init.d/apply_profile.code: [OK] building kalua-loader in ramdisk
Mon Dec  7 22:30:02 2015 user.emerg syslog: root: /etc/kalua_init: [OK] use $TMPDIR which is RAM-drive (tmpfs) and points to '/tmp/kalua'
Mon Dec  7 22:30:02 2015 user.notice root: /etc/kalua_init: [OK] use $TMPDIR which is RAM-drive (tmpfs) and points to '/tmp/kalua'
Mon Dec  7 22:30:03 2015 user.emerg syslog: root: /etc/kalua_init: [OK] shell interpreter: 'busybox' SHELL: ''
Mon Dec  7 22:30:03 2015 user.notice root: /etc/kalua_init: [OK] shell interpreter: 'busybox' SHELL: ''
Mon Dec  7 22:30:03 2015 user.emerg syslog: root: /etc/kalua_init: [OK] could not use '/NETPARAM' - maybe later
Mon Dec  7 22:30:03 2015 user.notice root: /etc/kalua_init: [OK] could not use '/NETPARAM' - maybe later
Mon Dec  7 22:30:04 2015 user.emerg syslog: root: /etc/kalua_init: [OK] generated '/tmp/loader' using files in '/tmp/kalua/kalua_pool'
Mon Dec  7 22:30:04 2015 user.notice root: /etc/kalua_init: [OK] generated '/tmp/loader' using files in '/tmp/kalua/kalua_pool'
Mon Dec  7 22:30:04 2015 user.emerg syslog: root: /etc/kalua_init: [OK] reload it with 'unset -f _; . /tmp/loader'
Mon Dec  7 22:30:04 2015 user.notice root: /etc/kalua_init: [OK] reload it with 'unset -f _; . /tmp/loader'
Mon Dec  7 22:30:04 2015 daemon.debug kalua: /etc/init.d/S41build_static_netparam: build() building '/tmp/NETPARAM' for an uci-device
Mon Dec  7 22:30:04 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 5: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:05 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:06 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:06 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:06 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:06 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:06 2015 user.notice root: /etc/rc.common: /etc/init.d/apply_profile.watch: watching for becoming '/etc/init.d/apply_profile' invisible, reboot in 29 min
Mon Dec  7 22:30:06 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:07 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:07 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:07 2015 authpriv.notice dropbear[1986]: Password auth succeeded for 'root' from 10.63.52.41:41628
Mon Dec  7 22:30:07 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:07 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:08 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:08 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:09 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
root@BraeuSys:~ lr -f
Mon Dec  7 22:30:11 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:11 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:12 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:12 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:12 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:13 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:13 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:14 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:14 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:15 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:15 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:16 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:16 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:17 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:17 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:18 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:18 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:19 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:20 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:20 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:21 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:21 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:22 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:23 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:24 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
Mon Dec  7 22:30:25 2015 user.emerg syslog: /etc/init.d/S41build_static_netparam: line 1: _sanitizer_shellsafe: not found
```

gab es sowas früher mal? Ich kann im source code nichts mit shellsafe finden...

glycoknob, 2015-12-07:
Hmm, ich auch nicht... das ist bei @bittorf  hier vorhanden: 

https://github.com/bittorf/kalua/blob/fbde398e0ab15b0da854f903294d4551c4363989/openwrt-addons/etc/kalua/sanitizer#L200

scheint irgendwo via /tmp/loader aufgerufen zu werden...

glycoknob, 2015-12-07:
too much magic :/

glycoknob, 2015-12-07:
einfachster workaround wäre _sanitizer aus kalua zu übernehmen.

glycoknob, 2015-12-07:
okay, ursache ist das hier: 

    ./openwrt-addons/etc/kalua/log:                         echo >>"$TMPDIR/ARCHIV_MONITORING" "log=$( _sanitizer run "$ARCHIV_NODE_MESSAGE" shellsafe urlvalue )&time=${UNIXTIME}"

jede logmessage wird nochmal validiert, der validator fehlt jedoch.

andibraeu, 2015-12-07:
komischerweise ist das in "unserem" log nicht drin

evtl. hat er hier beim update alte dateien stehen gelassen

andibraeu, 2015-12-07:
stimmt, das kam im letzten PR dazu...

glycoknob, 2015-12-07:
spricht imho nichts dagegen _sanitizer von kalua zu übernehmen.

andibraeu, 2015-12-07:
machste den mal fertig?  :-)

glycoknob, 2015-12-07:
quick & dirty: https://github.com/weimarnetz/weimarnetz/pull/200

bittorf, 2015-12-08:
wunderbar - das stand auch auf meiner todo-liste.

PR #200 (2015-12-08): sync _sanitizer with kalua
fixes #198
commit e43a2162 (2015-12-08): sync _sanitizer with kalua 

fixes #198
closed 2015-12-08 by bittorf